AutonomyLab/bebop_autonomy #8: Missing H264 decoder on Ubuntu 15.04

jakebruce, 2015-09-30:
I get the following build error on Ubuntu 15.04, even though I've fetched all the documented dependencies.

/home/jake/catkin_ws/src/bebop_autonomy/bebop_driver/src/bebop_video_decoder.cpp:
In member function bool bebop_driver::VideoDecoder::InitCodec(uint32_t,
uint32_t):
/home/jake/catkin_ws/src/bebop_autonomy/bebop_driver/src/bebop_video_decoder.cpp:79:39:
error: CODEC_ID_H264 was not declared in this scope
 codec_ptr_ = avcodec_find_decoder(CODEC_ID_H264);

jakebruce, 2015-10-01:
Works fine on Ubuntu 14.04. Incompatibility with 15.04 could be considered a feature rather than a bug, given the versions that ROS supports.

mani-monaj, 2015-10-01:
Thank you for reporting this. I think it's a bug since the driver should be compilable on all distros that support ROS _Jade_.

mani-monaj, 2016-04-01:
It has been fixed in 8da8e188d1c6dd210d1d55937f59be5b4dfd5e98 .

closed 2016-04-01 by mani-monaj