AutonomyLab/bebop_autonomy #15: Connecting Multiple Bebops

bellz121, 2015-10-14:
Is it possible to connect multiple bebops to a single computer following the same procedure for the ardrone_autonomy? Or is there some other way?

mani-monaj, 2015-10-15:
We haven't explored it yet. On the driver side, it should be trivial to [change the IP address of target Bebop](https://github.com/AutonomyLab/bebop_autonomy/blob/14252d39610ec9bbf83f86b2193a2f9908b2eced/bebop_driver/src/bebop.cpp#L227). Configuring Bebop to change it's IP address or make it connect to an infrastructure network needs investigation though.

mani-monaj, 2015-12-22:
The IP address configuration is now supported by the driver: #20

nicolasgallardo, 2016-07-22:
I have successfully connected the bebop to a secured network (linksys router). I will detail the process if ya'll are interested.

mani-monaj, 2016-07-22:
> I have successfully connected the bebop to a secured network (linksys router). I will detail the process if ya'll are interested.

It's definitely going to be useful.

nicolasgallardo, 2016-07-23:
Here you go. Any problems or improvements, please let me know.

https://github.com/nicolasgallardo/multiple_bebops

tnaegeli, 2016-08-20:
Hi mani, hi nicolas
i managed to connect two drones into our lab network. i can ping them separatly and everything works nicely. but when i try to start the bebop_autonomy node, i get the following error:

[ERROR] [1471663299.254993686]: [ARNETWORKAL_WifiNetwork] 23:21:39:254 | ARNETWORKAL_WifiNetwork_Bind:571 - [0x7fa73824c440] bind fd=12, addr='0.0.0.0', port=43210: error='Address already in use'
[ERROR] [1471663299.255159490]: [ARDISCOVERY_Device] 23:21:39:255 | ARDISCOVERY_Device_NewARNetworkAL:238 - error: Unknown generic error
[ERROR] [1471663299.255244766]: [ARCONTROLLER_Network] 23:21:39:255 | ARCONTROLLER_Network_New:191 - error: Wifi generic error
[ERROR] [1471663299.255348620]: [ARCONTROLLER_Network] 23:21:39:255 | ARCONTROLLER_Network_New:225 - error: Error during the getting of the ARNETWORKAL_Manager from the device
[ INFO] [1471663299.255544175]: [ARCONTROLLER_Device] 23:21:39:255 | ARCONTROLLER_Device_StartRun:3514 - Start failed or canceled.
[ERROR] [1471663299.255756633]: [ARCONTROLLER_Device] 23:21:39:255 | ARCONTROLLER_Device_StartRun:3522 - Start fail error :Error during the getting of the ARNETWORKAL_Manager from the device
[ INFO] [1471663299.255872429]: [BebopSDK] 23:21:39:255 | Cleanup:331 - Bebop Cleanup()
[FATAL] [1471663299.256158499]: Init failed: Waiting for device failed: No error
[ INFO] [1471663299.256317528]: Bebop Nodelet Dstr: 0
[ INFO] [1471663299.256356918]: Killing Camera Thread ...
[ INFO] [1471663299.256398893]: Killing Aux Thread ...
[q1/real/bebop_driver-1] process has died [pid 14062, exit code -11, cmd /home/naegelit/catkin_ws/devel/lib/bebop_driver/bebop_driver_node __name:=bebop_driver __log:=/home/naegelit/.ros/log/43de27c0-667f-11e6-98c4-989096ad3513/q1-real-bebop_driver-1.log].
log file: /home/naegelit/.ros/log/43de27c0-667f-11e6-98c4-989096ad3513/q1-real-bebop_driver-1*.log

It seems that somewhere the address is not set correctly? i use the following launchscript which seems to bi correct:

<?xml version="1.0"?>
<launch>

    <arg name="config_file" default="$(find bebop_driver)/config/defaults.yaml" />
    <arg name="camera_info_url" default="package://bebop_driver/data/bebop_camera_calib.yaml" />
     <group ns="q2/real">
        <node pkg="bebop_driver" name="bebop_driver" type="bebop_driver_node" output="screen">
            <param name="camera_info_url" value="$(arg camera_info_url)" />
            <param name="bebop_ip" value="192.168.1.14" />
            <rosparam command="load" file="$(arg config_file)" />
        </node>
        <include file="$(find bebop_description)/launch/description.launch" />
    </group>
  
    
    <group ns="q1/real">
        <node pkg="bebop_driver" name="bebop_driver" type="bebop_driver_node" output="screen">
            <param name="camera_info_url" value="$(arg camera_info_url)" />
            <param name="bebop_ip" value="192.168.1.13" />
            <rosparam command="load" file="$(arg config_file)" />
        </node>
        <include file="$(find bebop_description)/launch/description.launch" />
    </group>
    
   
    
</launch>

hoangtungdinh, 2016-09-23:
Hi all, I'm also trying to connect to multiple bebops from a single computer. I create two ros masters and run bebop-autonomy on each of them. Then I have exactly the same error as @tnaegeli  mentioned above. Is there any solution for that error yet?

tnaegeli, 2016-09-27:
Hi,
The problem is the parrot SDK. I have a patched version which works with multiple drones.. maybe i can put it online. As i heard from parrot, they will fix the issue in the newest sdk. 
e.g. you have to separately download the sdk and copy the files into the bebop autonomy package.

mhct, 2016-09-27:
Hi all,

I patched the Parrot SDK, to be able to have a server with multiple IPs connecting to multiple bebops, avoiding the binding problems mentioned by @tnaegeli and @hoangtungdinh. 

The target setup for this patch is when you have one computer with different network adapters, each adapter connecting to a different drone.

The patch simply consists in binding the connections to a particular IP, which is configured via a environment variable "LOCAL_DRONE_IP"

The patch to the SDK consists in changing any libraries which bind to all adapters to bind to only one IP. Thus, any libraries used in the SDK which have code like:
`recvSin.sin_addr.s_addr = htonl(INADDR_ANY);`

becomes something like:
`const char* local_drone_ip = getenv("LOCAL_DRONE_IP");
recvSin.sin_addr.s_addr = inet_addr (local_drone_ip);`

If you want to test, I uploaded the compiled binaries here: 
https://github.com/mhct/arsdk_multiple_bebops

You have to untar this file at the root folder of your bebop_autonomy workspace. The tar will overried the arsdk libs folder (devel/.private/bebop_driver/arsdk/out/arsdk-native/staging/usr/lib), with the patched version mentioned above.

**Note** that before loading the bebop driver you have to export the environment variable
`export LOCAL_DRONE_IP=192.168.42.xx` 
where 192.168.42.xx is the IP of you adapter connected to the bebop.

If there is interest, I can also put the modified source code online.

mani-monaj, 2016-10-02:
Thanks @mhct for your solution. I've been told that version 3.10.x of the SDK would fix the problem. I've updated the driver to support SDK 3.10.x (tested with firmware 3.9). Is it possible for you to verify that?

tnaegeli, 2016-10-08:
Hi All,
I can confirm that the latest SDK fixed all problems with multiple drones. You can follow these instructions:
https://github.com/tnaegeli/multiple_bebops
to use multiple bebops in the same network. It works fine with the autonomy package.

best,
Tobi

mani-monaj, 2016-10-12:
Thanks @tnaegeli for verifying that SDK 3.10.x and the updated version of `bebop_autonomy` work fine with the `multiple_bebops` patch. I will add the link to that repo to the documentation. In your opinion is it possible to achieve the _connect-to-access-point_ procedure simpler or more streamlined?

SlouchyCat, 2017-02-17:
Hi all,
Is it possible to connect using 5GHz band instead of 2.4GHz? I am not able to do that by simply adding "bcmwl band a" to the script.

Best,
Sam

dannyjose, 2017-06-30:
Hi all, I have a problem @tnaegeli , please I need help, I did everything, but I got this message: "telnet: could not resolve 192.168.43.1]/ telnet: name or service not 
know"
someone knows why I got that message?

mhct, 2017-06-30:
Hi,  @dannyjose 

First you should confirm you can ping your bebop.
$ ping 192.168.43.1

Then you have to press the on/off button 4 times, to activate the telenet server at the bebop.

IanMin1022, 2017-09-17:
Hi, @mhct

I've done every thing and it seemed fine but nothing worked after all. It made beep sound but when I checked ifconfig on drone shell it was still same and I couldnt ping for the settee IP, i.g) 192.168.0.x.

Can I get some advice from you?

bilelxlab, 2017-09-26:
Hi @mhct 
i get this when i execute the connect scritp:
ESSID: bebop
PASSWORD: bebop
ADDRESS: 192.168.0.14
DRONE_IP: 192.168.43.1
Connection closed by foreign host.

can you help ?

tnaegeli, 2017-09-26:
I'm not sure if this still works with the actual parrot firmware. But what
shouls still work is chance the ip to: 192.168.x.1 and then use different
usb wifi dongles. This is a workaround if you can not connect with the
drone to a Router.

best,
Tobi

IanMin1022, 2017-09-27:
@bilelxlab 
can you ping?

@tnaegeli 
I accessed with my own IP 192.168.0.9 and 192.168.0.18 but there is a problem here. I tried to connect both IP using roslaunch bebop_driver bebop_node.laucnh by changing launch file and bebop.cpp and I found out that I can't use another port for bebop 2 except 44444. Did you succeed to make multi drone with other way? Thanks

bilelxlab, 2017-09-27:
@tnaegeli 
do you mean change the ip adress in the roslauch with (<param name="bebop_ip" value="192.168.1.14" />) ? 
when i do that i get :  
[ERROR] [1506503854.147514340]: [ARDISCOVERY_Connection] 11:17:34:147 | ARDISCOVERY_Socket_Connect:721 - connect() failed: 101 Network is unreachable => Try reconnecting after 1 seconds

@lanMin1022
no i can't ping !

IanMin1022, 2017-09-27:
@bilelxlab If you can’t ping then it means your PC and your drone is not on same AP. And the x.x.43.1 thingie- thats wired connection. If you want to connect your PC and drone on same AP like both are 192.168.43.x or 192.168.42.x or if you made your drone as a client mode, some other IP address.

tnaegeli, 2017-09-28:
@bilelxlab <https://github.com/bilelxlab>
In the readme there is a second part called: To change the IP permanently.
If you are only interested in having multiple bebops flying then use x usb
wifi dongle. Each drone needs a own subnet. This means change the ip (as
described in the readme) to 192.168.101.1 for the first drone,
192.168.102.1 for the second and so on (or whatever you like 192.168.x.1).
Then you can connect each usb wifi dongle to the drone and you have as many
local networks as drones. Change then the ip in the bebop.launch file:

 <group ns="q6/real">
        <node pkg="bebop_driver" name="bebop_driver"
type="bebop_driver_node" output="screen">
            <param name="camera_info_url" value="$(arg camera_info_url)" />
            <param name="bebop_ip" value="192.168.106.1" />
            <rosparam command="load" file="$(arg config_file)" />
        </node>
        <include file="$(find bebop_description)/launch/description.launch"
/>
    </group>

use a new console for each drone (it turned out that the parrot driver has
problems if you start more than one drone in a single launch file). I hope
this helped.

If you still need to connect multiple drones to a single router, i'm not
sure if this still works. I tested it the last time half a year ago and
then it was not really stable anymore.

best,
Tobi

bilelxlab, 2017-09-28:
@IanMin1022  

hemmm they are not on the same AP 
my router is  192.168.0.0
that why in the script connect i gave ADDRESS= 192.168.0.14  ( if it's not like this what should i do ?) 

or you mean i should do : ifconfig eth0 192.168.0.14  before  unsing connect ?

IanMin1022, 2017-09-29:
@tnaegeli 
I think if I get access to another port of Bebop, I can just do anything in same launch file. I'm still figuring out how to get another accessible port except 44444. Anyway seems like you are out of this project now. Thanks, and good luck for your next one.

@bilelxlab 
First, I think your PC and drone need to be on same AP and  plus, before you launch the connect script, would you just get in to Bebop's shell and check ifconfig in there? And mmm... maybe I'm wrong but as I know router's IP is supposed to end with 1, e.g. 192.168.0.1.

tnaegeli, 2017-09-29:
@Ian_Min Sorry it was a missunderstanding. i thought you simply want to
connect and fly several drones with your computer which is no problem and
works gread. We fly here in the lab 4 Bebops without any issues.

tnaegeli, 2017-09-29:
> @tnaegeli <https://github.com/tnaegeli> That's great. Did you make it by
> executing 4 launch files or other way?

@Ian_Min  Since the newest updates (sdk & onboadr code) i have problems
with a single lauch file (before it was working) and now it works 1 out of
5 trials. But using 4 different lauch files works great and there is
absolute no problem.

bilelxlab, 2017-10-02:
@IanMin1022   ifconfig shows this ! 

eth0      Link encap:Ethernet  HWaddr A0:14:3D:BC:A7:00  
          inet addr:192.168.0.105  Bcast:192.168.0.255  Mask:255.255.255.0
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:233
          TX packets:156 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:19190 (18.7 KiB)

lo        Link encap:Local Loopback  
          inet addr:127.0.0.1  Mask:255.0.0.0
          UP LOOPBACK RUNNING  MTU:16436  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

rndis0    Link encap:Ethernet  HWaddr AE:26:88:F8:C8:F1  
          inet addr:192.168.43.1  Bcast:192.168.43.255  Mask:255.255.255.0
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:82 errors:0 dropped:0 overruns:0 frame:0
          TX packets:80 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:9072 (8.8 KiB)  TX bytes:15144 (14.7 KiB)

my router is (192.168.0.1)  i gave the good ssid and PW  but i can't ping the drone !!!!  can't understand what is wrong !!!!!

bilelxlab, 2017-10-03:
@tnaegeli  sorry i missed your  email about the wifi usb dongle (Thank you for the details)

1) about changing the IP adress: i am not trying to change it permanently i am trying just to connect to my router  when i push the bebop botton 3 times this doesn't  work too ?

tnaegeli, 2017-10-03:
@bilelxlab i did this one ywar ago but have problems with the connection
now. Therefore i switched to the usb-Wifi solution and have absolute no
problems flying with multiple drones. Also the parrot app works without
problems after changing the ip

IanMin1022, 2017-11-02:
@tnaegeli heyy, I know you stopped this project one year ago but have you tried to check the internal program of Bebop drone?
I wanna change it to land when it loses the wifi signal or when the ros PID thread stops. However when I opened the inside using adb tool, there were too many things so, now, I have no idea how/where to start.
Can I get any clue if you have some idea? Thank you :D

tnaegeli, 2017-11-02:
maybe you can find something here (An unofficial Bebop drone hacking guide
1.6) :
https://onedrive.live.com/?authkey=%21AIZaZ4S5jq12EWI&cid=7E20CE1F838C9EB7&id=7E20CE1F838C9EB7%21220&parId=7E20CE1F838C9EB7%21105&o=OneUp

best,
Tobi

IanMin1022, 2017-11-02:
@tnaegeli  Thanks a lot!!!

carel-chris, 2018-09-25:
> @tnaegeli heyy, I know you stopped this project one year ago but have you tried to check the internal program of Bebop drone?
> I wanna change it to land when it loses the wifi signal or when the ros [...]

@IanMin1022 Could you get this working? If so details would be highly appreciated.

Kind regards

mastercomputer17, 2018-10-27:
hi mani iam from montreal. iam new in this and  plz if you can help me. couse when the bebop 2 starts, the red light only blinks and stay in that state. I installed ros and I tried to install the drivers but it did not work. Then I wrote ip adrees and I deleted all the folders that were in the internal memory. and now this is the internal memory. I copied other folders from another bebop and put it in internal memory but it did not work. i dont know what to do so can you tell me what is the best solution and how do i do it or how can i install all those files one more time

Christopher9Ernest, 2020-07-13:
Hi. I am trying to connect three bebop 2 to the same router so that I can control them at the same time. I am using Ubuntu linux 16.04, Bebop_autonomy and my router is  TP-LINK ac1200 tl-wdr5620. I have tried all the procedure provided by (https://github.com/tnaegeli/multiple_bebops) but my bebop 2 doesnt connect to my secured WiFi. I dont know if it is because of the router or something else? Should I try to change permanently the default IP address of the Bebop 2 to see if it will connect to the router?