AutonomyLab/bebop_autonomy #23: Socket access permission denied

ashnap123, 2015-12-21:
Hi,

This ROS package looks great, awesome work. I am however having an issue, whenever I try to launch the bebop_node.launch I get the following issue:

ARDISCOVERY_Device_NewARNetworkAL:238 - error: Socket access permission denied

I also get similar error message when trying to run the SDK example, BebopDroneReceiveStream. I followed all the instructions and the build process reported no errors. Im using Ubuntu 14.04

Thanks for any help in advance!

ashnap123, 2015-12-21:
Hi,

This is a problem with the drone not starting up properly. If you get an error like this simply restart the drone until it works. Sorry to waste your time.

mani-monaj, 2015-12-22:
Great.

Possible related issue discussed on official dev forum: http://forum.developer.parrot.com/t/bebopdronereceivestream-socket-error/444/3

closed 2015-12-22 by mani-monaj