AutonomyLab/bebop_autonomy #35: Failed to build package 'bebop_driver' because the following command

hajungong007, 2016-03-13:
cd /home/uav-ros-3/bebop_ws/build/bebop_driver && /home/uav-ros-3/bebop_ws/build/bebop_driver/build_env.sh /usr/bin/make --jobserver-fds=3,5 -j; cd -

# Path to log:
cat /home/uav-ros-3/bebop_ws/build/build_logs/bebop_driver.log

Exited with return code: 2 
                                                                                
[build] Runtime: 20.5 seconds

mani-monaj, 2016-03-14:
Hi.

Could you please:

1. Ask this question on http://answers.ros.org and tag it with `bebop_autonomy`
2. Provide more details about the error you are receiving.

closed 2016-03-14 by mani-monaj

hajungong007, 2016-03-16:
I have got it! Thanks!

CODEC_ID_H264 instead of  AV_CODEC_ID_H264
PIX_FMT_YUV420P instead of  AV_PIX_FMT_YUV420P

mani-monaj, 2016-03-16:
Thanks for the report. I've created a new issue for it.

commit 8da8e188 (2016-03-22): Fix libav API inconsistency issues

- Remove depreacated use of CODEC_* and PIX_FMT_*. Use AV_* instead.
- Add backward compatibility for older libav API versions

Closes #30
Closes #35
Closes #36