AutonomyLab/bebop_autonomy #46: bebop_autonomy not compiling

anuppari, 2016-04-30:
I've tried following the instructions [here](http://bebop-autonomy.readthedocs.io/en/latest/installation.html) for building the package on a new computer, but the package does not compile. At some point during the bebop_driver compilation, an xterm window pops up and says
```
Can't execvp ./build.sh: No such file or directory
```
Seems like during the process of downloading the ARDroneSDK, the build script doesn't get downloaded. I see that in the CMakeLists of bebop_driver, a certain commit of the arsdk_manifest is pulled, which has a release.xml that links to a certain commit of arsdk_package that has the build.sh in the correct place, but for some reason the file is not pulled down.

The `repo sync` command spits out a number of 404 Not Found, but at the end says that fetching and syncing were completed 100%, so I'm not sure what the issue is.

mani-monaj, 2016-05-02:
Thanks for reporting this. I will work on releasing a fix as soon as possible.

grigisf, 2016-05-02:
Here's a work around: 

I forked an earlier version of the arsdk (3.8 release) and made the following changes to the CMakeLists.txt file of bebop_driver package:

line 67:  COMMAND echo "y" | ./repo init -u https://github.com/grigisf/arsdk_manifests.git
line 70:  COMMAND ./build.sh -p arsdk-native -t build-sdk || xterm -e ./build.sh -p arsdk-native -t build-sdk

This way the build is successful (I get a lot of warnings though).

commit c7edd80f (2016-05-04): Fix inline build of arsdk to use the frozen manifest (fixes #46)

- Prior to this release, the build script would always compile the
development branch of ARSDK. This fix ensures that instead of
`default.xml` manifest file - which represents the dev version of ARSDK
package - `release.xml` is used by `repo`. This manifest file includes a
certain hash for each ARSDK package that enforces a consistent build for
ARSDK.
commit 8d9dea06 (2016-05-04): Merge pull request #47 from AutonomyLab/46-fix-arsdk-build

Fix inline build of arsdk to use the frozen manifest (fixes #46)
closed 2016-05-04 by mani-monaj