AutonomyLab/bebop_autonomy #48: Bebop Disconnects From Wireless Network When Driver is Launched

anuppari, 2016-05-06:
We are running the Bebop by connecting it to a router. The Bebop connects and we were running the driver just fine. However, today after running successfully we replaced the battery and restarted the Bebop. It connects to the network but when the driver is launched the Bebop disconnects itself from the network and the driver fails to complete the launch. An error comes up saying the connection is lost because it took too long to receive the first frame. We could see no change in the driver but still tried deleting the driver completely and rebuilding. After rebuilding the problem persisted. Does anyone know what the problem might be? Here is the output of the driver launch: 

> SUMMARY
========

PARAMETERS
 * /bebop/bebop_driver/bebop_ip: 192.168.1.101
 * /bebop/bebop_driver/camera_info_url: package://bebop_d...
 * /bebop/bebop_driver/reset_settings: True
 * /bebop/bebop_driver/states/enable_altitudechanged: True
 * /bebop/bebop_driver/states/enable_autotakeoffmodechanged: True
 * /bebop/bebop_driver/states/enable_camerastate_orientation: True
 * /bebop/bebop_driver/states/enable_commonstate_batterystatechanged: True
 * /bebop/bebop_driver/states/enable_commonstate_wifisignalchanged: True
 * /bebop/bebop_driver/states/enable_controllerstate_ispilotingchanged: True
 * /bebop/bebop_driver/states/enable_flightplanstate_availabilitystatechanged: True
 * /bebop/bebop_driver/states/enable_flightplanstate_componentstatelistchanged: True
 * /bebop/bebop_driver/states/enable_gpsstate_numberofsatellitechanged: True
 * /bebop/bebop_driver/states/enable_mavlinkstate_mavlinkfileplayingstatechanged: True
 * /bebop/bebop_driver/states/enable_mavlinkstate_mavlinkplayerrorstatechanged: True
 * /bebop/bebop_driver/states/enable_mediastreamingstate_videoenablechanged: True
 * /bebop/bebop_driver/states/enable_numberofsatellitechanged: True
 * /bebop/bebop_driver/states/enable_overheatstate_overheatchanged: True
 * /bebop/bebop_driver/states/enable_pilotingstate_altitudechanged: True
 * /bebop/bebop_driver/states/enable_pilotingstate_attitudechanged: True
 * /bebop/bebop_driver/states/enable_pilotingstate_flattrimchanged: True
 * /bebop/bebop_driver/states/enable_pilotingstate_flyingstatechanged: True
 * /bebop/bebop_driver/states/enable_pilotingstate_navigatehomestatechanged: True
 * /bebop/bebop_driver/states/enable_pilotingstate_positionchanged: True
 * /bebop/bebop_driver/states/enable_pilotingstate_speedchanged: True
 * /rosdistro: indigo
 * /rosversion: 1.11.19

NODES
  /bebop/
    bebop_driver (bebop_driver/bebop_driver_node)

auto-starting new master
process[master]: started with pid [3672]
ROS_MASTER_URI=http://workstation:11311/

setting /run_id to 02a842b4-13a4-11e6-8438-3c970edc4507
process[rosout-1]: started with pid [3685]
started core service [/rosout]
process[bebop/bebop_driver-2]: started with pid [3702]
[ INFO] [1462550582.810552466]: Initializing nodelet with 8 worker threads.
[ INFO] [1462550582.878636277]: [BebopSDK] 12:03:02:878 | Bebop:198 - Bebop Cnstr()
[ INFO] [1462550582.878705567]: Nodelet Cstr
[ INFO] [1462550582.886062153]: Connecting to Bebop ...
[ INFO] [1462550582.891177876]: [CB] 12:03:02:891 | Ardrone3PilotingStateFlatTrimChanged:319 - [STATES] Enabling states/ARDrone3/PilotingState/FlatTrimChanged
[ INFO] [1462550582.893775132]: [CB] 12:03:02:893 | Ardrone3PilotingStateFlyingStateChanged:363 - [STATES] Enabling states/ARDrone3/PilotingState/FlyingStateChanged
[ INFO] [1462550582.896615254]: [CB] 12:03:02:896 | Ardrone3PilotingStateNavigateHomeStateChanged:465 - [STATES] Enabling states/ARDrone3/PilotingState/NavigateHomeStateChanged
[ INFO] [1462550582.898468033]: [CB] 12:03:02:898 | Ardrone3PilotingStatePositionChanged:523 - [STATES] Enabling states/ARDrone3/PilotingState/PositionChanged
[ INFO] [1462550582.900179366]: [CB] 12:03:02:900 | Ardrone3PilotingStateSpeedChanged:588 - [STATES] Enabling states/ARDrone3/PilotingState/SpeedChanged
[ INFO] [1462550582.901976164]: [CB] 12:03:02:901 | Ardrone3PilotingStateAttitudeChanged:653 - [STATES] Enabling states/ARDrone3/PilotingState/AttitudeChanged
[ INFO] [1462550582.904435691]: [CB] 12:03:02:904 | Ardrone3PilotingStateAltitudeChanged:769 - [STATES] Enabling states/ARDrone3/PilotingState/AltitudeChanged
[ INFO] [1462550582.908986874]: [CB] 12:03:02:908 | Ardrone3MediaStreamingStateVideoEnableChanged:1045 - [STATES] Enabling states/ARDrone3/MediaStreamingState/VideoEnableChanged
[ INFO] [1462550582.912266005]: [CB] 12:03:02:912 | Ardrone3CameraStateOrientation:1096 - [STATES] Enabling states/ARDrone3/CameraState/Orientation
[ INFO] [1462550582.916254270]: [CB] 12:03:02:916 | Ardrone3GPSStateNumberOfSatelliteChanged:1256 - [STATES] Enabling states/ARDrone3/GPSState/NumberOfSatelliteChanged
[ INFO] [1462550582.922437928]: [CB] 12:03:02:922 | CommonCommonStateBatteryStateChanged:141 - [STATES] Enabling states/common/CommonState/BatteryStateChanged
[ INFO] [1462550582.930734351]: [CB] 12:03:02:930 | CommonCommonStateWifiSignalChanged:503 - [STATES] Enabling states/common/CommonState/WifiSignalChanged
[ INFO] [1462550582.936090787]: [CB] 12:03:02:936 | CommonOverHeatStateOverHeatChanged:714 - [STATES] Enabling states/common/OverHeatState/OverHeatChanged
[ INFO] [1462550582.938812280]: [CB] 12:03:02:938 | CommonControllerStateisPilotingChanged:809 - [STATES] Enabling states/common/ControllerState/isPilotingChanged
[ INFO] [1462550582.940673323]: [CB] 12:03:02:940 | CommonMavlinkStateMavlinkFilePlayingStateChanged:860 - [STATES] Enabling states/common/MavlinkState/MavlinkFilePlayingStateChanged
[ INFO] [1462550582.943380837]: [CB] 12:03:02:943 | CommonMavlinkStateMavlinkPlayErrorStateChanged:925 - [STATES] Enabling states/common/MavlinkState/MavlinkPlayErrorStateChanged
[ INFO] [1462550582.950948562]: [CB] 12:03:02:950 | CommonFlightPlanStateAvailabilityStateChanged:1201 - [STATES] Enabling states/common/FlightPlanState/AvailabilityStateChanged
[ INFO] [1462550582.953564100]: [CB] 12:03:02:953 | CommonFlightPlanStateComponentStateListChanged:1252 - [STATES] Enabling states/common/FlightPlanState/ComponentStateListChanged
[ INFO] [1462550583.055508151]: [BebopSDK] 12:03:03:055 | CommandReceivedCallback:112 - Command Received Callback LWP id is: 139736760979177
[ INFO] [1462550583.236001806]: [BebopSDK] 12:03:03:235 | Connect:287 - BebopSDK inited, lwp_id: 3702
[ WARN] [1462550583.236126052]: Resetting all settings ...
[... 60 lines cut ...]
[ INFO] [1462550589.712917599]: [CB] 12:03:09:712 | UpdateBebopFromROS:796 - NetworkSettingsWifiSelectionType changed!
[ INFO] [1462550589.712975222]: [CB] 12:03:09:712 | UpdateBebopFromROS:828 - Sending NetworkSettingsWifiSelection changes to bebop
[ INFO] [1462550589.713032354]: [CB] 12:03:09:713 | UpdateBebopFromROS:938 - GPSSettingsHomeTypeType changed!
[ERROR] [1462550589.713071706]: [CB] 12:03:09:713 | UpdateBebopFromROS:946 - Value of GPSSettingsHomeType was not initialized either by Bebop or Params.
[ INFO] [1462550589.725599154]: Enabling video stream ...
[ WARN] [1462550589.725720016]: [BebopSDK] 12:03:09:725 | StartStreaming:325 - Video streaming started ...
[ INFO] [1462550589.725891685]: Nodelet lwp_id: 3702
[ INFO] [1462550589.725954965]: Camera publisher thread lwp_id: 3998
[ INFO] [1462550589.726075612]: bebop_driver nodelet loaded.
[ INFO] [1462550589.832409415]: [ARSTREAM_Reader] 12:03:09:832 | ARSTREAM_Reader_RunDataThread:455 - Missed -65535 frames !
[ INFO] [1462550589.832671765]: [BebopSDK] 12:03:09:832 | FrameReceivedCallback:144 - Frame Recv & Decode LWP id: 139741055946476
[ INFO] [1462550589.834209624]: [Decoder] 12:03:09:834 | InitCodec:127 - H264 Codec is initialized!
[ INFO] [1462550589.834269139]: [Decoder] 12:03:09:834 | Decode:201 - Waiting for the first IFrame.
[ERROR] [1462550589.834340208]: [BebopSDK] 12:03:09:834 | FrameReceivedCallback:171 - Video decode failed
[ INFO] [1462550589.845375550]: [Decoder] 12:03:09:845 | Decode:201 - Waiting for the first IFrame.
[ERROR] [1462550589.845471645]: [BebopSDK] 12:03:09:845 | FrameReceivedCallback:171 - Video decode failed
[ INFO] [1462550589.872389018]: [Decoder] 12:03:09:872 | Decode:201 - Waiting for the first IFrame.
[ERROR] [1462550589.872495407]: [BebopSDK] 12:03:09:872 | FrameReceivedCallback:171 - Video decode failed
[ INFO] [1462550589.917476501]: [Decoder] 12:03:09:917 | Decode:201 - Waiting for the first IFrame.
[ERROR] [1462550589.917603219]: [BebopSDK] 12:03:09:917 | FrameReceivedCallback:171 - Video decode failed
[ INFO] [1462550589.939233707]: [Decoder] 12:03:09:939 | Decode:201 - Waiting for the first IFrame.
[ERROR] [1462550589.939332592]: [BebopSDK] 12:03:09:939 | FrameReceivedCallback:171 - Video decode failed
[ INFO] [1462550594.978263919]: [ARNETWORKAL_WifiNetwork] 12:03:14:978 | ARNETWORKAL_WifiNetwork_Receive:860 - [0x7f17e424bdf0] connection lost (too long time without reception)

>

anuppari, 2016-05-06:
Some additional info:

1. We had this problem previously, and switching to this commit e00b40d0a891f344470717c07e0ec9b424d96c66 seemed to have fixed the problem. We thought this might mean the driver pulls an updated version of the SDK, however now, the computer is not even connected to the internet, and without any changes as far as we can see, the issue reappeared.

2.  Currently we are using a Bebop 2. However, using a Bebop 1 with [this script](https://github.com/Parrot-Developers/Samples/wiki/Convert-Bebop-to-client-instead-of-access-point) gives the same issue. Similarly, resetting the access point, as well as using a different access point yields the same issue. It seems that when the driver works, it works for all combinations of Bebop's, routers, and  computers, and when it doesn't work, it doesn't work for any.

3. Through pinging the bebop at a high rate, and adding delays in the driver, it seems the disconnect happens at this [line](https://github.com/AutonomyLab/bebop_autonomy/blob/e00b40d0a891f344470717c07e0ec9b424d96c66/bebop_driver/src/bebop_driver_nodelet.cpp#L127). It seems that when the parameters are reset, it forces the bebop to disconnect from the router. We have checked the band and channel settings in the [cfg file](https://github.com/AutonomyLab/bebop_autonomy/blob/e00b40d0a891f344470717c07e0ec9b424d96c66/bebop_driver/cfg/autogenerated/bebop_ardrone3.cfg), and they match the router settings, so we are not sure what the issue is.

It seems really weird that it can work fine at one minute, and then permanently break the next time it's launched. We don't really know what to check next. Any insight would be greatly appreciated.

mani-monaj, 2016-05-14:
Can you try setting `reset_settings` to false, either [here](https://github.com/AutonomyLab/bebop_autonomy/blob/indigo-devel/bebop_driver/config/defaults.yaml) or in your custom configuration file?

anuppari, 2016-07-20:
I've tried doing that and it still has the same problem. Looking at the output of the node launch more closely, I see the following:

```
[ INFO] [1469023592.390956954]: Dynamic reconfigure callback with level: -1
[ INFO] [1469023592.391067208]: [CB] 10:06:32:391 | UpdateBebopFromROS:796 - NetworkSettingsWifiSelectionType changed!
[ INFO] [1469023592.391111051]: [CB] 10:06:32:391 | UpdateBebopFromROS:828 - Sending NetworkSettingsWifiSelection changes to bebop
```

Since the settings are not being reset (confirmed since I don't see `[ WARN] [1469023091.321904669]: Resetting all settings ...` in the output), I'm not sure what is causing the dynamic reconfigure callback to be called?

Turning off the dynamic reconfigure (i.e., commenting out [this line](https://github.com/AutonomyLab/bebop_autonomy/blob/e00b40d0a891f344470717c07e0ec9b424d96c66/bebop_driver/src/bebop_driver_nodelet.cpp#L127) ) seems to make everything work, though I'm not sure for how long considering we had things working temporarily before, and we would like to enable dynamic parameters.

mani-monaj, 2016-07-22:
It's strange. Have you ever modified any of the on-board config files of your Bebop? What version of `bebop_autonomy` are you using? What is the firmware version of your Bebop?

mani-monaj, 2016-10-14:
@anuppari Have you had a chance to look more into this issue?

risingsunho, 2016-11-23:
i have a same problem

anuppari, 2016-11-25:
We did not modify any of the on-board config files, other than adding the long press script linked above to change the bebop from AP mode to client mode. We tried using the latest version, as well as an older commit (mentioned above) that we thought would fix the issue, but both ended up not working.

I'm not sure about the firmware version, but I think it was the latest that was available at the time. Unfortunately, I no longer have access to the drones anymore, but @bellz867 is continuing the work with the drones and should have access to them. He would also be the one still running into this issue and interested in the fix.

risingsunho, 2016-11-25:
i have experimented bebop drone with setting its ip 192.168.42.9~13.
but, when i set a drone ip to 192.168.42.1, it works well but other ip doesn't work

mani-monaj, 2016-12-07:
Thanks @anuppari for the update. The problem might be caused by the script you mentioned. I will close this issue for now. @bellz867 please feel free to re-open this issue if you have more information and/or a fix.

closed 2016-12-07 by mani-monaj