AutonomyLab/bebop_autonomy #49: Raw IMU data stream and other sensors

MohitShridhar, 2016-05-07:
Great package! Works like a charm! 

Do you have plans to support raw IMU data streams? And also other sensors: barometer & ultrasound?
Plus, the documentation says that the odometry measurements are computed by integrating 'visual-inertial' velocities. Is the Bebop doing some sort of visual-inertial SLAM onboard? How are these velocities computed?

Thanks

mani-monaj, 2016-05-14:
> Do you have plans to support raw IMU data streams? And also other sensors: barometer & ultrasound?

At the moment those values are not streamed from the Bebop to the host device. This is a missing feature of the firmware/SDK.  If Parrot adds this feature to their firmware/SDK, we will integrate that into the driver for sure.

[Here](https://github.com/cucx/bebop) you can find a method to store those data onboard.

> Plus, the documentation says that the odometry measurements are computed by integrating 'visual-inertial' velocities. Is the Bebop doing some sort of visual-inertial SLAM onboard? How are these velocities computed?

I will add more information to the documentation about this. Bebop does not perform on-board SLAM. As far as I know, Bebop performs on-board visual-inertial odometery estimation by fusing optical flow from its downward facing camera stream and inertial readings from its IMU.

MohitShridhar, 2016-05-14:
Sounds good! Thanks!

closed 2016-05-14 by MohitShridhar

davidvar, 2018-04-13:
Do you know if the visual-inertial odometry is using a Kalman filter or some filtering algorithm to estimate the drone Position?

fabrizioschiano, 2018-04-13:
> Here you can find a method to store those data onboard.
@mani-monaj hi, but the link is broken, could you update your answer? Thanks in advance. 

@davidvar I would also be interested to understand this!

LArayane, 2020-07-20:
@mani-monaj Is there any updates regarding IMU data. and Also I want to see the link you sent but it is broken, can you resend. Thanks