AutonomyLab/bebop_autonomy #50: Bebop Driver Frequently Outputting Warnings

bellz867, 2016-05-09:
When testing the most recent driver on a Bebop 2, a set of warnings are frequently output. The warnings are:

> [ WARN] [1462807887.981096145]: [ARSTREAM2_H264Filter] 11:31:27:981 | ARSTREAM2_H264Filter_fillMissingEndOfFrame:892 - Missing NALU is probably a slice
[ WARN] [1462807887.981218202]: [ARSTREAM2_H264Filter] 11:31:27:981 | ARSTREAM2_H264Filter_fillMissingEndOfFrame:950 - Skipped P slice NALU output size: 13
[ WARN] [1462807887.983282285]: [ARSTREAM2_H264Filter] 11:31:27:983 | ARSTREAM2_H264Filter_fillMissingSlices:768 - Missing NALU is probably a SPS, PPS or SEI or on previous AU => OK

In addition to this warning set, sometimes independent and sometimes at the same time, the following warning shows up:

> [ WARN] [1462808246.418777361]: [BebopSDK] 11:37:26:418 | FrameReceivedCallback:196 - Previous frame might have been missed.

Is this a normal warning? It seems to be happening very frequently which makes control difficult. Thank you for your time and any information.

mani-monaj, 2016-05-14:
The warnings come from underlying `ARSTREAM2` library of ARSDK and are simply relayed by the driver. Could you please check the new branch of the driver that is based on SDK 3.9.x and see if you see reduced number of messages? The code is in `arsdk_3_9` branch.

shino-kei, 2016-05-25:
I also encountered the same warnings with bebop2 (firmware 3.2.0), Ubnutu14.04 and ROS indigo.
So, I checked out and built arsdk_3_9 branch but console output same warnings...

Unfortunately I don't have  3.1.0 firmware data for bebop2, so I cannot test with firmware 3.1.0. 
 But, according to Parrot Developer forum, this problem may be caused by latest firmware 3.2.0 for
bebop and bebop2.

- http://developer.parrot.com/blog/2016/play-bebop-stream-on-vlc/
- http://forum.developer.parrot.com/t/receive-video-on-ubuntu-specific-questions/2616

So I tried the sample program on the first link above and succeed in RTP video streaming with VLC player, Bebop2(3.2.0) with ARSDK3(3.9.1).

Trinto, 2016-09-17:
Did you solve it?