AutonomyLab/bebop_autonomy #52: control_camera topic

DLabo, 2016-05-19:
Is camera_control topic working correctly ? According to 
https://media.readthedocs.org/pdf/bebop-autonomy/latest/bebop-autonomy.pdf

to  move  Bebop’s  virtual  camera I should  publish  a  message  of  type  geometry_msgs/Twist  to
camera_control topic. But then the topic requires a message of type std_msgs/Empty that does not allow to control the camera

~/catkin_ws$ rostopic info /bebop/camera_control
Type: std_msgs/Empty

Publishers: None

Subscribers: 
 * /bebop/bebop_driver

mani-monaj, 2016-05-21:
I can not reproduce the issue locally. Could you please list the steps to reproduce the bug (incl. the target platform, firmware and driver version)?

DLabo, 2016-05-23:
I am simply trying to take off and then to move the camera in order to look the ground. So the command is:
rostopic pub -r 1 /bebop/camera_control geometry_msgs/Twist -- '[0.0, 0.0, 0.0]' '[0.0,0.5,0.0]'

This seems not possible since the camera_control topic requires Empty messages.

bebop drone with software 2.0.57 
 driver bebop autonomy 0.5.1
linux ubuntu 14.04 - Ros Indigo

thanks

mani-monaj, 2016-05-25:
It is a bit odd. Could you please:

1. Share the result of `rosnode info` on bebop driver node?
2. Share a picture of your ROS computation graph using `rqt_graph`?

mani-monaj, 2016-06-19:
Did you have a chance to confirm the issue?

mani-monaj, 2016-10-02:
This is fixed in d43463c44775fced1c6277aa5bf3c9ba95a490ec

closed 2016-10-02 by mani-monaj