AutonomyLab/bebop_autonomy #55: Joystick mapping for different controllers

aaravrav142, 2016-06-07:
Hi,

I cant seem to figure out how to map different joysticks with the bebop2. I am trying to configure the xbox 360 wireless controller on ROS indigo. Can you help me with this?

Thanks
Alex

bellz867, 2016-06-07:
Hey,
I've had that issue with xbox before, the easiest way to figure out the mapping is run the joy node. You can echo the joy topic and press each xbox button to see which index it is on buttons array. Then do the same for the joysticks and the axes array.
Hope this helps.

aaravrav142, 2016-06-08:
Hi
@bellz867 Thanks for the reply. 
I will try to do as you suggested. Can you share your xbox launch and config yaml file with me to get an idea. 
Thanks a lot 
--
Alex

bellz867, 2016-06-08:
I only use a launch file no config. See attached.
[joyNodeLaunch.launch.zip](https://github.com/AutonomyLab/bebop_autonomy/files/305221/joyNodeLaunch.launch.zip)

Hope it helps.

thomas-bamford, 2016-06-08:
Hi,
I use an xbox 360 wireless controller for my lab's Bebop 2. Below are the config and launch files I created for this controller.

[xbox360.zip](https://github.com/AutonomyLab/bebop_autonomy/files/305387/xbox360.zip)

The main thing you should change is the launch file's "dev" argument to match the location of your controller as mentioned by @bellz867. To find more information about your controller:

http://wiki.ros.org/joy/Tutorials/ConfiguringALinuxJoystick

aaravrav142, 2016-06-14:
Hi @thomas-bamford and @bellz867,

Thanks for sharing your launch files. I was able to test it using xbox controller. I changed some keys.

Many thanks for the help. 

Alex

closed 2016-06-14 by aaravrav142

mani-monaj, 2016-06-18:
@thomas-bamford and @bellz867 I would be more than happy to incorporate your changes to `bebop_tools` package if you send a pull request.