AutonomyLab/bebop_autonomy #58: Compile bebop_autonomy on Odroid XU4 (ARM Cortex A15)

tnaegeli, 2016-06-20:
Hi All,
I'm trying to compile the bebop_autonomy package on an odroid xu4 (indigo) and have compile errors. I found similar problems in the ARDrone autonomy package:
https://github.com/AutonomyLab/ardrone_autonomy/issues/71
but i'm not able to figure out how to apply this patch to the bebop_autonomy package (different structure).
Does anyone have similar problems/issues?

the compiler output:
### 
> /home/odroid/catkin_ws/devel/arsdk/build/alchemy/scripts/alchemake 
gcc-4.8.real: error: unrecognized command line option '-m32'
gcc-4.8.real: error: unrecognized command line option '-msse'
gcc-4.8.real: error: unrecognized command line option '-msse2'
gcc-4.8.real: error: unrecognized command line option '-mssse3'
gcc-4.8.real: error: unrecognized command line option '-m32'
gcc-4.8.real: error: unrecognized command line option '-msse'
gcc-4.8.real: error: unrecognized command line option '-msse2'
gcc-4.8.real: error: unrecognized command line option '-mssse3'
readelf: Error: '/home/odroid/catkin_ws/devel/arsdk/out/Unix-base/build/a.out': No such file
mkdir: invalid mode 'ssse3'
----------------------------------------------------------------------
+ ALCHEMY_WORKSPACE_DIR = /home/odroid/catkin_ws/devel/arsdk
+ TARGET_PRODUCT = Unix
+ TARGET_PRODUCT_VARIANT = base
+ TARGET_OS = linux
+ TARGET_OS_FLAVOUR = native
+ TARGET_LIBC = native
+ TARGET_ARCH = x86
+ TARGET_CPU = 
+ TARGET_OUT = /home/odroid/catkin_ws/devel/arsdk/out/Unix-base
+ TARGET_CONFIG_DIR = /home/odroid/catkin_ws/devel/arsdk/products/Unix/base/config
+ TARGET_CC_PATH = /usr/bin/gcc
+ TARGET_CC_VERSION = 4.8
----------------------------------------------------------------------
Scanning /home/odroid/catkin_ws/devel/arsdk/packages for makefiles...
Prebuilt module ncurses marked as overriden
Found 30 makefiles
Found 58 modules
make[3]: warning: jobserver unavailable: using -j1.  Add `+' to parent make rule.
Autotools: json => Configuring
configure: error: in `/home/odroid/catkin_ws/devel/arsdk/out/Unix-base/build/json/json-c-0.9':
configure: error: C compiler cannot create executables
See `config.log' for more details
make[3]: *** [/home/odroid/catkin_ws/devel/arsdk/out/Unix-base/build/json/json-0.9.configured] Error 77
[ 96%] Built target ARSDKBuildUtils
[ 97%] [ 98%] Building CXX object bebop_autonomy/bebop_driver/CMakeFiles/libbebop.dir/src/bebop.cpp.o
Building CXX object bebop_autonomy/bebop_driver/CMakeFiles/libbebop.dir/src/bebop_video_decoder.cpp.o
In file included from /home/odroid/catkin_ws/src/bebop_autonomy/bebop_driver/src/bebop_video_decoder.cpp:25:0:
/home/odroid/catkin_ws/src/bebop_autonomy/bebop_driver/include/bebop_driver/bebop_video_decoder.h:30:50: fatal error: libARController/ARCONTROLLER_Error.h: No such file or directory
   #include "libARController/ARCONTROLLER_Error.h"
                                                  ^
compilation terminated.
make[2]: *** [bebop_autonomy/bebop_driver/CMakeFiles/libbebop.dir/src/bebop_video_decoder.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
/home/odroid/catkin_ws/src/bebop_autonomy/bebop_driver/src/bebop.cpp:37:40: fatal error: libARCommands/ARCommands.h: No such file or directory
   #include "libARCommands/ARCommands.h"
                                        ^
compilation terminated.
make[2]: *** [bebop_autonomy/bebop_driver/CMakeFiles/libbebop.dir/src/bebop.cpp.o] Error 1
make[1]: *** [bebop_autonomy/bebop_driver/CMakeFiles/libbebop.dir/all] Error 2
make: *** [all] Error 2
Invoking "make -j4" failed

mani-monaj, 2016-07-20:
I am working on a fix for this issue based on this patch: Parrot-Developers/alchemy#6

Which version of the driver are you using at the moment?

tnaegeli, 2016-07-20:
Hi Mani,

Wow, great! I started with crosscompiling but in the end descided to first
do the trajectory planning algorithm before i start to fix the issues.
Good question. i'm still using the same version i pulled when i tried to
compile and wrote you (actually not sure which version it was). The sdk is
in devel/arsdk, do you know where to look the version up?

best,
Tobi

mani-monaj, 2016-07-22:
> The sdk isin devel/arsdk, do you know where to look the version up?

Please just let me know the latest git commit hash of `bebop_autonomy` you are using (`$ git rev-parse HEAD`)

tnaegeli, 2016-07-23:
hi Mani,

my head is

ca452eb5d1c884dc388ab4eba11717ca1b97577f

best,

Tobi

wilpulling, 2016-07-23:
Hi Mani, 

Excited to see you're working on this. I've been trying to install the [bebop_autonomy ROS driver](http://bebop-autonomy.readthedocs.io/en/latest/installation.html) on my Raspberry Pi 3 (Ubuntu Mate 16.04, ROS Kinetic), and ran into the same '-m32', '-msse', 'msse2', etc. errors as tnaegeli. These are also the errors which originally prevented me from [building the SDK directly](http://developer.parrot.com/docs/SDK3/#how-to-build-the-sdk). 

I'd like to better understand the fix you're working on implementing. Will it be incorporated into the ROS driver, such that installing the driver while using an ARM system will simply work? Or will it mean that we'll be able to build the SDK on an ARM system, perhaps by using a different build variant (e.g. instead of "./build.sh -p arsdk-native...", "./build.sh -p arsdk-arm...")? 

Also, can you possibly give a very general time frame for the fix? For example, should we expect a "couple of days," or a "couple of weeks"? 

Lastly, in case the fix is going to take a little while: If i were to cross compile the SDK and bring it onto my RasPi, would I then be able to install the ROS driver without seeing these errors? Would it need to be in a specific directory? Or, would the driver try to build the SDK by itself anyway and fail?

Thanks so much for your support on this issue, you're a life saver!

-Wilson!

samA94, 2016-08-08:
Do any of you have updates on this?  I'm having the same issues (Raspberry Pi 2, Ubuntu 14.04, Indigo).  If any of you figured out a way around this issue, I would appreciate it if you would let me know what you did.

mani-monaj, 2016-09-22:
I've been working on patching and releasing Parrot ARSDK as an independent package here: https://github.com/AutonomyLab/parrot_arsdk

The idea is to make it compile on different platforms supported by the ROS build farm. The ultimate goal is make that install-able via `apt-get`.

The good news is:

1. The patched version compiles on `i386`, `amd64`, `armhf` and `arm64` architectures: http://wiki.ros.org/parrot_arsdk (check the _Jenkins Jobs_ section)
2. It is available via official ROS repositories (i.e `apt-get install ros-kinetic-parrot-arsdk` works)

The bad news is:

1. It is not yet integrated in `bebop_autonomy` (The experimental integration is in `2-parrot-sdk` branch of the driver)
2. It is not based on the latest SDK version. My patched version is at version `3.9.1`, while the latest SDK is now at version `3.9.10`.

I am going to upgrade `parrot_arsdk` to `3.9.10` and finish the integration very soon. In the mean time you can check my patches on `parrot_arsdk` or use it directly on your non `x86_64` platform.

bellz867, 2016-10-12:
I have compiled and run the bebop_autonomy package on the Odroid XU4 with 16.04 mate and ros kinetic. I installed the parrot_arsdk using `sudo apt-get install ros-kinetic-parrot-arsdk`. After installing I followed the build instructions in the docs, http://bebop-autonomy.readthedocs.io/en/latest/installation.html.
Thanks for the help!

mani-monaj, 2016-10-14:
@bellz867 Thanks for confirming that it is actually working. Once the binaries for Parrot SDK 3.10.x are released to _Indigo_ and _Jade_ (They are already available in _Kinetic_), I will start the process of releasing `bebop_autonomy` as a binary as well. That makes it way easier to run the driver on platforms like _Odroid_.

closed 2016-10-14 by mani-monaj