AutonomyLab/bebop_autonomy #60: Simulation support for the parrot bebop drone 1 or 2 on ROS, gazebo etc ??

aaravrav142, 2016-07-04:
Hi,

I was wondering if there is any ROS simulation support for the Parrot bebop drone.  It seems the Parrot ARdrone has good support for the simulation test but could not find any information for the bebop drone. The drone itself flies well using the bebop_autonomy, but I would still not risk the drone crashing while testing some developed algorithms. I would deeply appreciate if someone can point me in right direction. 

Thanks 

Alex

mani-monaj, 2016-07-04:
> I was wondering if there is any ROS simulation support for the Parrot bebop drone.

Unfortunately I don't know of any.

>  It seems the Parrot ARdrone has good support for the simulation test but could not find any information for the bebop drone. 

[tum_simulator](http://wiki.ros.org/tum_simulator) is based on [tu-darmstadt-ros-pkg](http://wiki.ros.org/tu-darmstadt-ros-pkg/). I've never examined that in depth, but my intuition is creating a similar simulator for Bebop should not be that difficult. The 3D model of the Bebop is [officially available](https://sketchfab.com/models/6e7dccf233db40b7921f1c6fc5a6e36d) and determining a simplified dynamic model of the drone is doable if one has access to a motion-capture system.

KathiravanNatarajan, 2016-12-28:
Hi Mani-Monaj, 
Request you help me on simulating the parrot bebop on gazebo simulator using ROS commands.

Please find this. http://stackoverflow.com/questions/41326090/how-to-fly-a-parrot-bebop-drone-in-tum-simulator
Regards,
Kathir

mani-monaj, 2016-12-28:
@KathiravanNatarajan please find my answer [here](http://stackoverflow.com/a/41365756/1215297). You can post your future questions on [ROS Answers](http://answers.ros.org/questions/) instead of SO.

closed 2016-12-28 by mani-monaj

gsilano, 2019-07-22:
Hello everyone. Most likely I arrived a few years late, but I think it may be useful for future visitors to the page. My colleagues and I, at the University of Sannio in Benevento, Italy, developed a ROS package for simulating the behavior of the Parrot Bebop 2. The ROS package is compatible with both Indigo Igloo and Kinetic Kame distros of ROS. Moreover, we are working to make the code compatible with Parrot Sphinx (it simulates the firmware behavior thus, making it possible to put it in the control loop). Here the link to the repository available on GitHub.

https://github.com/gsilano/BebopS

Victorjgc, 2021-03-16:
> Hello everyone. Most likely I arrived a few years late, but I think it may be useful for future visitors to the page. My colleagues and I, at the University of Sannio in Benevento, Italy, developed a ROS package for simulating the behavior of the Parrot Bebop 2. The ROS package is compatible with both Indigo Igloo and Kinetic Kame distros of ROS. Moreover, we are working to make the code compatible with Parrot Sphinx (it simulates the firmware behavior thus, making it possible to put it in the control loop). Here the link to the repository available on GitHub.
> 
> https://github.com/gsilano/BebopS
> 
> **EDIT January 5th, 2020**
> 
> Now, the ROS package is compatible with ROS Melodic and Gazebo 9, ROS Kinetic and Gazebo 7 and ROS Kinetic and Gazebo 9.

I tried to install it but it has a lot of errors most of them abot: hoovering_example duplicated. I do not why I just follow the steps for Kinetics and Gazebo 7.

gsilano, 2021-03-16:
Open an issue in the repository, and let me know what error you get.