AutonomyLab/bebop_autonomy #62: Set maximum vertical speed and pitch/roll rotation speed

hoangtungdinh, 2016-07-28:
Hi,

Bebop SDK allows us to set the maximum vertical speed and pitch/roll rotation speed. 

http://developer.parrot.com/docs/SDK3/#a-name-ardrone3-speedsettings-maxverticalspeed-a
http://developer.parrot.com/docs/SDK3/#a-name-ardrone3-speedsettings-maxpitchrollrotationspeed-a

It would be nice if we could set it via ROS.

Thank you for your great work.

thomas-bamford, 2016-07-28:
Hi,

These parameters can be set via ROS. See:

SpeedSettingsMaxVerticalSpeedCurrent
& SpeedSettingsMaxPitchRollRotationSpeedCurrent

http://bebop-autonomy.readthedocs.io/en/latest/autogenerated/ardrone3_settings_param.html

hoangtungdinh, 2016-07-28:
Thanks a lot for pointing that out and a fast response!

closed 2016-07-28 by hoangtungdinh