AutonomyLab/bebop_autonomy #65: Can't turn off video stabilization

TomRoussel, 2016-09-09:
When changing the parameter "PictureSettingsVideoStabilizationModeMode", nothing happens. Normally if the parameter is 3, there shouldn't be any stabilization, but when reading "/bebop/image_raw" we saw that the video was still being stabilized. 
The other modes of that parameter also do not visually change the image feed.

mani-monaj, 2016-09-22:
I saw this behaviour when I released v0.5.1. I think it is due a bug in the firmware and/or the SDK. I will update the driver soon to use the newest SDK and work with the latest firmware. Hopefully that will fix this issue. Could you please share the firmware version of your Bebop and its model?

TomRoussel, 2016-09-22:
Yes of course. We are using the Parrot Bebop 2 drone, firmware version is 3.1.0.

commit e9725597 (2016-10-02): Full SDK 3.10.x support (Fixes #68, Fixes #67)

- BebopInTheLoop tests pass on both Bebop 1 and 2, on firmware 3.2 and
3.9
- ROS API change: states/ARDrone3/.. topics renamed to
states/ardrone3/..
- Disabling video stabilization works with Firmware 3.9 (tested on Bebop
    2) - Close #65
closed 2016-10-02 by mani-monaj