AutonomyLab/bebop_autonomy #66: Cannot launch bebop_node.launch

Trinto, 2016-09-12:
Hey guys. Getting following error: 

> [ WARN] [1473683814.559167827]: [ARSTREAM2_H264Filter] 14:36:54:559 | ARSTREAM2_H264Filter_generateGrayIFrame:618 - Gray I slice NALU output size: 1629
[ INFO] [1473683814.559236865]: [BebopSDK] 14:36:54:559 | FrameReceivedCallback:174 - Frame Recv & Decode LWP id: 12687
[ INFO] [1473683814.665935263]: [Decoder] 14:36:54:665 | InitCodec:133 - H264 Codec is initialized!
[ INFO] [1473683814.666004930]: [Decoder] 14:36:54:665 | Decode:235 - Updating H264 codec parameters (Buffer Size: 36) ...
[bebop/bebop_driver-2] process has died [pid 12321, exit code -11, cmd /home/xxx/bebop_ws/devel/lib/bebop_driver/bebop_driver_node __name:=bebop_driver __log:=/home/xxx/.ros/log/90320dda-78e5-11e6-b3ff-0024e8cfd983/bebop-bebop_driver-2.log].
log file: /home/xxx/.ros/log/90320dda-78e5-11e6-b3ff-0024e8cfd983/bebop-bebop_driver-2*.log

> 

Also getting some errors like:
`
[ERROR] [1473683809.673369512]: [ARCONTROLLER_Network] 14:36:49:673 | ARCONTROLLER_Network_ReaderRun:940 - ARCOMMANDS_Decoder_DecodeBuffer () failed : 2 ARDrone3.PictureSettingsState.UNKNOWN -> Unknown co
[ERROR] [1473683809.691899960]: [ARCONTROLLER_Network] 14:36:49:691 | ARCONTROLLER_Network_ReaderRun:940 - ARCOMMANDS_Decoder_DecodeBuffer () failed : 2 ARDrone3.PictureSettingsState.UNKNOWN -> Unknown co

 `
i finsished catkin with:

Finished  <<< bebop_driver                         [ 7 minutes and 40.0 seconds ]
Starting  >>> bebop_tools                                                      
Finished  <<< bebop_tools                          [ 2.0 seconds ]             
[build] Summary: All 5 packages succeeded!                                     
[build]   Ignored:   1 packages were skipped or are blacklisted.               
[build]   Warnings:  1 packages succeeded with warnings.                       
[build]   Abandoned: None.                                                     
[build]   Failed:    None.                                                     
[build] Runtime: 7 minutes and 59.8 seconds total.         

could someone help me plz?

Trinto, 2016-09-15:
i guess its about the picturesettingsstate. bebop has 3 picture types and he isnt sure, which i want to use. Someone know how to fix it?

Trinto, 2016-09-17:
Could someone help me please?

jacobperron, 2016-09-17:
I saw your [post on ROS answers](http://answers.ros.org/question/243933/connecting-bebop-drone-with-ros/?answer=243934#post-id-243934). I'll try to help you there :)

mani-monaj, 2016-09-22:
Possibly related to #67 #68. If so, could you please close this issue in favour of those?

closed 2016-09-22 by Trinto

Kornelia11101, 2018-07-19:
Hi, did you fix the problem? :)
Thanks!