AutonomyLab/bebop_autonomy #68: HALP!  It's not working!

copterGuy, 2016-09-21:
Hi, 

Just got a brand new Bebop 2 drone last week and it works great with the controller & phone app.  

So I installed ROS indigo on Ubuntu 14.04 and built / installed the bebop_autonomy ROS package according to the documentation instructions  Got all the way up to the part where I run the driver with "roslaunch bebop_driver bebop_node.launch" and then I get the result below (process dies with errors mentioning ARCOMMANDS_Decoder_DecodeBuffer () failed : 2 ARDrone3.PictureSettingsState.UNKNOWN)...  

>> roslaunch bebop_driver bebop_node.launch

... logging to /home/username/.ros/log/09c2f872-8051-11e6-a79c-a0a8cd7a2911/roslaunch-username-comp-22267.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://username-comp:42539/

SUMMARY
========

PARAMETERS
 * /bebop/bebop_driver/bebop_ip: 192.168.42.1
 * /bebop/bebop_driver/camera_info_url: package://bebop_d...
 * /bebop/bebop_driver/cmd_vel_timeout: 0.2
 * /bebop/bebop_driver/odom_frame_id: odom
 * /bebop/bebop_driver/publish_odom_tf: True
 * /bebop/bebop_driver/reset_settings: True
 * /bebop/bebop_driver/states/enable_altitudechanged: True
 * /bebop/bebop_driver/states/enable_autotakeoffmodechanged: True
 * /bebop/bebop_driver/states/enable_camerastate_orientation: True
 * /bebop/bebop_driver/states/enable_commonstate_batterystatechanged: True
 * /bebop/bebop_driver/states/enable_commonstate_wifisignalchanged: True
 * /bebop/bebop_driver/states/enable_controllerstate_ispilotingchanged: True
 * /bebop/bebop_driver/states/enable_flightplanstate_availabilitystatechanged: True
 * /bebop/bebop_driver/states/enable_flightplanstate_componentstatelistchanged: True
 * /bebop/bebop_driver/states/enable_gpsstate_numberofsatellitechanged: True
 * /bebop/bebop_driver/states/enable_mavlinkstate_mavlinkfileplayingstatechanged: True
 * /bebop/bebop_driver/states/enable_mavlinkstate_mavlinkplayerrorstatechanged: True
 * /bebop/bebop_driver/states/enable_mediastreamingstate_videoenablechanged: True
 * /bebop/bebop_driver/states/enable_numberofsatellitechanged: True
 * /bebop/bebop_driver/states/enable_overheatstate_overheatchanged: True
 * /bebop/bebop_driver/states/enable_pilotingstate_altitudechanged: True
 * /bebop/bebop_driver/states/enable_pilotingstate_attitudechanged: True
 * /bebop/bebop_driver/states/enable_pilotingstate_flattrimchanged: True
 * /bebop/bebop_driver/states/enable_pilotingstate_flyingstatechanged: True
 * /bebop/bebop_driver/states/enable_pilotingstate_navigatehomestatechanged: True
 * /bebop/bebop_driver/states/enable_pilotingstate_positionchanged: True
 * /bebop/bebop_driver/states/enable_pilotingstate_speedchanged: True
 * /bebop/robot_description: <?xml version="1....
 * /rosdistro: indigo
 * /rosversion: 1.11.20

NODES
  /bebop/
    bebop_driver (bebop_driver/bebop_driver_node)
    robot_state_publisher (robot_state_publisher/robot_state_publisher)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
process[bebop/bebop_driver-1]: started with pid [22288]
process[bebop/robot_state_publisher-2]: started with pid [22289]
[ INFO] [1474500555.912267356]: Initializing nodelet with 4 worker threads.
[ INFO] [1474500555.957847044]: [BebopSDK] 18:29:15:957 | Bebop:230 - Bebop Cnstr()
[ INFO] [1474500555.957905316]: Nodelet Cstr
[ INFO] [1474500555.965846269]: Connecting to Bebop ...
[ INFO] [1474500555.969096582]: [CB] 18:29:15:969 | Ardrone3PilotingStateFlatTrimChanged:320 - [STATES] Enabling states/ARDrone3/PilotingState/FlatTrimChanged
[ INFO] [1474500555.970596368]: [CB] 18:29:15:970 | Ardrone3PilotingStateFlyingStateChanged:364 - [STATES] Enabling states/ARDrone3/PilotingState/FlyingStateChanged
[ INFO] [1474500555.972415353]: [CB] 18:29:15:972 | Ardrone3PilotingStateNavigateHomeStateChanged:466 - [STATES] Enabling states/ARDrone3/PilotingState/NavigateHomeStateChanged
[ INFO] [1474500555.973526332]: [CB] 18:29:15:973 | Ardrone3PilotingStatePositionChanged:524 - [STATES] Enabling states/ARDrone3/PilotingState/PositionChanged
[ INFO] [1474500555.974568931]: [CB] 18:29:15:974 | Ardrone3PilotingStateSpeedChanged:589 - [STATES] Enabling states/ARDrone3/PilotingState/SpeedChanged
[ INFO] [1474500555.975523155]: [CB] 18:29:15:975 | Ardrone3PilotingStateAttitudeChanged:654 - [STATES] Enabling states/ARDrone3/PilotingState/AttitudeChanged
[... 123 lines cut ...]
[ INFO] [1474500562.373560988]: [CB] 18:29:22:373 | UpdateBebopFromROS:323 - Sending PilotingSettingsMaxDistance changes to bebop
[ INFO] [1474500562.388250848]: Value for PilotingSettingsMaxDistanceValue recved: 160
[ INFO] [1474500562.388383775]: [CB] 18:29:22:388 | Update:352 - Checking if PilotingSettingsMaxDistanceValue exists in params ...
[ INFO] [1474500562.389550285]: [CB] 18:29:22:389 | Update:360 -   Yes
[ INFO] [1474500562.390181171]: Enabling video stream ...
[ WARN] [1474500562.390235425]: [BebopSDK] 18:29:22:390 | StartStreaming:364 - Video streaming started ...
[ INFO] [1474500562.390333428]: Nodelet lwp_id: 22288
[ INFO] [1474500562.390367663]: [CameraThread] thread lwp_id: 22622
[ INFO] [1474500562.390400980]: [AuxThread] thread lwp_id: 22623
[ INFO] [1474500562.390427443]: bebop_driver nodelet loaded.
[ WARN] [1474500562.413070576]: [ARSTREAM2_H264Filter] 18:29:22:413 | ARSTREAM2_H264Filter_RtpReceiverNaluCallback:1201 - ARSTREAM2_RtpReceiver NALU buffer is too small, truncated AU (or maybe it's the fi
[ WARN] [1474500562.413162221]: [ARSTREAM2_H264Filter] 18:29:22:413 | ARSTREAM2_H264Filter_sync:161 - SPS/PPS sync OK
[ INFO] [1474500562.413203086]: [BebopSDK] 18:29:22:413 | DecoderConfigCallback:147 - H264 configuration packet received: #SPS: 27 #PPS: 9 (MP4? 0)
[ WARN] [1474500562.413264830]: [ARSTREAM2_H264Filter] 18:29:22:413 | ARSTREAM2_H264Filter_generateGrayIFrame:568 - Waiting for a slice to generate gray I-frame
[ WARN] [1474500562.413331715]: [ARSTREAM2_H264Filter] 18:29:22:413 | ARSTREAM2_H264Filter_generateGrayIFrame:568 - Waiting for a slice to generate gray I-frame
[ WARN] [1474500562.417604668]: [ARSTREAM2_H264Filter] 18:29:22:417 | ARSTREAM2_H264Filter_generateGrayIFrame:618 - Gray I slice NALU output size: 1629
[ INFO] [1474500562.417696384]: [BebopSDK] 18:29:22:417 | FrameReceivedCallback:174 - Frame Recv & Decode LWP id: 22625
[ INFO] [1474500562.419384238]: [Decoder] 18:29:22:419 | InitCodec:133 - H264 Codec is initialized!
[ INFO] [1474500562.419423624]: [Decoder] 18:29:22:419 | Decode:235 - Updating H264 codec parameters (Buffer Size: 36) ...
[bebop/bebop_driver-1] process has died [pid 22288, exit code -11, cmd /home/username/bebop_ws/devel/lib/bebop_driver/bebop_driver_node __name:=bebop_driver __log:=/home/username/.ros/log/09c2f872-8051-11e6-a79c-a0a8cd7a2911/bebop-bebop_driver-1.log].
log file: /home/username/.ros/log/09c2f872-8051-11e6-a79c-a0a8cd7a2911/bebop-bebop_driver-1*.log
^C[bebop/robot_state_publisher-2] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

mani-monaj, 2016-09-22:
`bebop_autonomy` needs to be updated to work with the latest firmware released by Parrot for Bebop drones. I will start working on it later this week and hopefully will release a new version as soon as possible.

copterGuy, 2016-09-22:
Sweet - you are awesome... thanks much!

Cheers

Trinto, 2016-09-22:
look at my post here. I also fixed it by habdling myself : http://answers.ros.org/question/243933/connecting-bebop-drone-with-ros/?answer=243934

rupumped, 2016-10-01:
Is there any workaround for the time being? I'm experiencing a similar issue whenever I try to run the driver node/nodelet.

Trinto, 2016-10-01:
Sry, i dont underdtand your question :X. 

Mit freundlichen Grüßen
Dennis Weber

rupumped, 2016-10-01:
Trinto, thank you for replying. The launch code (roslaunch bebop_tools bebop_nodelet_iv.launch) fails at exactly the same point as the original poster (i.e., after "Updating H264 codec parameters", the process dies).

I tried looking at the answer posted here: http://answers.ros.org/question/243933/connecting-bebop-drone-with-ros/?answer=243934, but couldn't understand exactly what should be done. If you can provide step-by-step instructions, I would appreciate that a lot.

Trinto, 2016-10-01:
Its about the error code -11. its a segmentation fault. chance in bebop_driver -> src -> bebop-decoder.cpp

       frame_rgb_raw_ptr_ = reinterpret_cast<uint8_t*>(av_malloc(num_bytes * sizeof(uint8_t)));
       ThrowOnCondition(frame_rgb_raw_ptr_ == NULL,

into

       frame_rgb_raw_ptr_ = reinterpret_cast<uint8_t*>(av_malloc(num_bytes * sizeof(uint64_t)));
       ThrowOnCondition(frame_rgb_raw_ptr_ == NULL,

Now the frames gave more ram-space to safe a frame.

closed 2016-10-02 by mani-monaj

mani-monaj, 2016-10-02:
I've updated the driver to support SDK 3.10.x. I've also fixed the issue of insufficient buffer size. You should be able to stream the video in 480p or 720p from Bebop 2.

rupumped, 2016-10-02:
Thank you, Trinto! My problem is now solved.

Summary:
In addition to following the instructions on the website, one must also install the additional dependencies:
```
sudo apt-get install ros-kinetic-robot-state-publisher
sudo apt-get install python-rostopic
```
Then, navigate to:
`~/bebop_ws/src/bebop_autonomy/bebop_driver/src/`
And gedit the file bebop_video_decoder.cpp as Trinto describes above. After saving and exiting the file, navigate to:
`cd ~/bebop_ws/build/bebop_driver`
and remake with the command:
`make`
Now that you have fixed the error in the camera driver, start the driver by entering:
```
source ~/bebop_ws/devel/setup.bash
roslaunch bebop_tools bebop_nodelet_iv.launch
```
To publish commands to the drone, open a new terminal and enter something like:
`rostopic pub --once /bebop/takeoff std_msgs/Empty`

Andres1508, 2018-11-19:
Hello! I'm using Ubuntu 16.04 + ROS Kinetic + bebop_autonomy, after runbebop_nodelet_iv.launch the terminal show me this error:

[ERROR] [ARSTREAM2_H264] 13:03:12:376 | ARSTREAM_H264_AuNaluFifoPopFreeItem:1022 . AU-NALU FIFO is Full
[WARN] [15422649230.892135823] : [ARSTREAM_h264] 12:40:30:891 | ARSTREAM2:H264_AufifoUnrefBuffer:705 - FIXME! Ref count is already null, this should not happen!

Any ideas?

commit e9725597 (2016-10-02): Full SDK 3.10.x support (Fixes #68, Fixes #67)

- BebopInTheLoop tests pass on both Bebop 1 and 2, on firmware 3.2 and
3.9
- ROS API change: states/ARDrone3/.. topics renamed to
states/ardrone3/..
- Disabling video stabilization works with Firmware 3.9 (tested on Bebop
    2) - Close #65