AutonomyLab/bebop_autonomy #72: Warning with building bebop_driver

gwurb, 2016-10-23:
Hi All,

I am having difficulty wih the catkin build.  I am following the instructions on the wiki and the build seems to go until some 404s.  The outcome is this:
[build] Summary: All 5 packages succeeded!                                                                                                                                              
[build]   Ignored:   1 packages were skipped or are blacklisted.                                                                                                                        
[build]   Warnings:  1 packages succeeded with warnings.                                                                                                                                
[build]   Abandoned: None.                                                                                                                                                              
[build]   Failed:    None.                                                                                                                                                              
[build] Runtime: 43 minutes and 36.6 seconds total.                                                                                                                                     
[build] Note: Workspace packages have changed, please re-source setup files to use them.

I take it this is not good enough to proceed further and it should be sorted out before continuing.  

The output for the errors is this (seems like Parrot SDK problem?):
Starting  >>> bebop_driver                                                                                                                                                              
________________________________________________________________________________________________________________________________________________________________________________________
Warnings   << bebop_driver:make /home/xx/bebop_ws/logs/bebop_driver/build.make.000.log                                                                                                  
gpg: keyring `/home/xx/.repoconfig/gnupg/secring.gpg' created
gpg: keyring `/home/xx/.repoconfig/gnupg/pubring.gpg' created
gpg: /home/xx/.repoconfig/gnupg/trustdb.gpg: trustdb created
gpg: key 920F5C65: public key "Repo Maintainer <repo@android.kernel.org>" imported
gpg: key 692B382C: public key "Conley Owens <cco3@android.com>" imported
gpg: Total number processed: 2
gpg:               imported: 2  (RSA: 1)
Get https://gerrit.googlesource.com/git-repo/clone.bundle
Get https://gerrit.googlesource.com/git-repo
remote: Finding sources: 100% (5/5)
remote: Total 5 (delta 2), reused 5 (delta 2)
Unpacking objects: 100% (5/5), done.
From https://gerrit.googlesource.com/git-repo
   82f6798..4d5bb68  master     -> origin/master
Get https://github.com/Parrot-Developers/arsdk_manifests.git
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0curl: (22) The requested URL returned error: 404 Not Found
Server does not provide clone.bundle; ignoring.
remote: Counting objects: 80, done.        
remote: Total 80 (delta 0), reused 0 (delta 0), pack-reused 80        
From https://github.com/Parrot-Developers/arsdk_manifests
 * [new branch]      master     -> origin/master
 * [new tag]         ARSDK3_version_3_10_0 -> ARSDK3_version_3_10_0
 * [new tag]         ARSDK3_version_3_10_1 -> ARSDK3_version_3_10_1
 * [new tag]         ARSDK3_version_3_8 -> ARSDK3_version_3_8
 * [new tag]         ARSDK3_version_3_8_1 -> ARSDK3_version_3_8_1
 * [new tag]         ARSDK3_version_3_8_2 -> ARSDK3_version_3_8_2
 * [new tag]         ARSDK3_version_3_8_3 -> ARSDK3_version_3_8_3
 * [new tag]         ARSDK3_version_3_9_0 -> ARSDK3_version_3_9_0
 * [new tag]         ARSDK3_version_3_9_1 -> ARSDK3_version_3_9_1
 * [new tag]         ARSDK3_version_3_9_2 -> ARSDK3_version_3_9_2
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0   % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0 % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0curl: (22) The requested URL returned error: 404 Not Found
Server does not provide clone.bundle; ignoring.
curl: (22) The requested URL returned error: 404 Not Found
Server does not provide clone.bundle; ignoring.
curl: (22) The requested URL returned error: 404 Not Found
Server does not provide clone.bundle; ignoring.
curl: (22) The requested URL returned error: 404 Not Found
Server does not provide clone.bundle; ignoring.
remote: Counting objects: 446, done.        
remote: Counting objects: 482, done.        
remote: Counting objects: 1421, done.        
remote: Counting objects: 659, done.        
remote: Total 446 (delta 0), reused 0 (delta 0), pack-reused 446        
[... 254 lines cut ...]
packages/libARNetworkAL/Sources/Wifi/ARNETWORKAL_WifiNetwork.c: In function 'ARNETWORKAL_WifiNetwork_Receive':
packages/libARNetworkAL/Sources/Wifi/ARNETWORKAL_WifiNetwork.c:903:13: warning: ignoring return value of 'read', declared with attribute warn_unused_result [-Wunused-result]
             read (receiverObject->fifo[0], &dump, 1);
             ^
packages/libARDiscovery/Sources/ARDISCOVERY_Connection.c: In function 'ARDISCOVERY_Connection_DeviceAccept':
packages/libARDiscovery/Sources/ARDISCOVERY_Connection.c:816:13: warning: ignoring return value of 'read', declared with attribute warn_unused_result [-Wunused-result]
             read (connectionData->abortPipe[0], &dump, 10);
             ^
packages/libARDiscovery/Sources/ARDISCOVERY_Connection.c: In function 'ARDISCOVERY_Connection_RxPending':
packages/libARDiscovery/Sources/ARDISCOVERY_Connection.c:947:13: warning: ignoring return value of 'read', declared with attribute warn_unused_result [-Wunused-result]
             read (connectionData->abortPipe[0], &dump, 10);
             ^
packages/libARDiscovery/Sources/ARDISCOVERY_Connection.c: In function 'ARDISCOVERY_Connection_Unlock':
packages/libARDiscovery/Sources/ARDISCOVERY_Connection.c:1036:9: warning: ignoring return value of 'write', declared with attribute warn_unused_result [-Wunused-result]
         write (connectionData->abortPipe[1], buff, 1);
         ^
[I] In '/home/xx/bebop_ws/devel/.private/bebop_driver/arsdk': ln -fs arsdk-native /home/xx/bebop_ws/devel/.private/bebop_driver/arsdk/out/arsdk-native/../Unix-base
[I] Finished task 'build-sdk'
/home/xx/bebop_ws/src/bebop_autonomy/bebop_driver/src/bebop.cpp: In member function ‘void bebop_driver::Bebop::Connect(ros::NodeHandle&, ros::NodeHandle&, const string&)’:
/home/xx/bebop_ws/src/bebop_autonomy/bebop_driver/src/bebop.cpp:236:6: note: variable tracking size limit exceeded with -fvar-tracking-assignments, retrying without
 void Bebop::Connect(ros::NodeHandle& nh, ros::NodeHandle& priv_nh, const std::string& bebop_ip)
      ^
cd /home/xx/bebop_ws/build/bebop_driver; catkin build --get-env bebop_driver | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
........................................................................................................................................................................................
Finished  <<< bebop_driver                         [ 40 minutes and 26.9 seconds ]

mani-monaj, 2016-10-24:
Could you please include the environment you are building the package on? (i.e. the OS and ROS version).

gwurb, 2016-10-26:
Hi, the Linux is Ubuntu 15.04 vivid, ROS jade with roscpp 1.11.20 (above is from Jade). I also tried Ubuntu 16.04.1 and ROS kinetic with roscpp 1.12.5.

I followed instructions from:
http://wiki.ros.org/jade/Installation/Ubuntu
http://bebop-autonomy.readthedocs.io/en/latest/installation.html

gwurb, 2016-11-02:
Hi Mani,

You have not been able to reproduce it?  Do you think its Parrot SDK problem?

Thanks

mani-monaj, 2016-11-03:
Could you please clean your workspace and re-compile the driver? I've updated the driver to use pre-built binaries of Parrot SDK (#75).

[Installation — bebop_autonomy indigo-devel documentation](http://bebop-autonomy.readthedocs.io/en/latest/installation.html)

gwurb, 2016-11-08:
Thanks Mani, that seems to have improved the situation.  

I do still get a warning, if that makes a difference... (I am only just starting to read about ROS so can't test much to see if it makes a difference or not)

Starting  >>> bebop_driver                                                                                                                                
__________________________________________________________________________________________________________________________________________________________
Warnings   << bebop_driver:make /home/xx/bebop_ws/logs/bebop_driver/build.make.000.log                                                          
/home/xx/bebop_ws/src/bebop_autonomy/bebop_driver/src/bebop.cpp: In member function ‘void bebop_driver::Bebop::Connect(ros::NodeHandle&, ros::NodeHandle&, const string&)’:
/home/xx/bebop_ws/src/bebop_autonomy/bebop_driver/src/bebop.cpp:236:6: note: variable tracking size limit exceeded with -fvar-tracking-assignments, retrying without
 void Bebop::Connect(ros::NodeHandle& nh, ros::NodeHandle& priv_nh, const std::string& bebop_ip)
      ^
cd /home/xx/bebop_ws/build/bebop_driver; catkin build --get-env bebop_driver | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
..........................................................................................................................................................
Finished  <<< bebop_driver                         [ 3 minutes and 26.5 seconds ]                                                                         
Starting  >>> bebop_tools                                                                                                                                 
Finished  <<< bebop_tools                          [ 3.6 seconds ]                                                                                        
[build] Summary: All 5 packages succeeded!                                                                                                                
[build]   Ignored:   1 packages were skipped or are blacklisted.                                                                                          
[build]   Warnings:  1 packages succeeded with warnings.                                                                                                  
[build]   Abandoned: None.                                                                                                                                
[build]   Failed:    None.

gwurb, 2016-11-08:
Oh, just to add another warning with Ubuntu 16.04 with ROS Kinetic

Starting  >>> bebop_driver                                                                                                                                                           
_____________________________________________________________________________________________________________________________________________________________________________________
Warnings   << bebop_driver:cmake /home/xx/bebop_ws/logs/bebop_driver/build.cmake.000.log                                                                                     
CMake Warning at /opt/ros/kinetic/share/catkin/cmake/catkin_package.cmake:166 (message):
  catkin_package() DEPENDS on 'system_lib' but neither
  'system_lib_INCLUDE_DIRS' nor 'system_lib_LIBRARIES' is defined.
Call Stack (most recent call first):
  /opt/ros/kinetic/share/catkin/cmake/catkin_package.cmake:102 (_catkin_package)
  CMakeLists.txt:40 (catkin_package)

CMake Warning at /opt/ros/kinetic/share/catkin/cmake/catkin_package.cmake:166 (message):
  catkin_package() DEPENDS on 'libavcodec' but neither
  'libavcodec_INCLUDE_DIRS' nor 'libavcodec_LIBRARIES' is defined.
Call Stack (most recent call first):
  /opt/ros/kinetic/share/catkin/cmake/catkin_package.cmake:102 (_catkin_package)
  CMakeLists.txt:40 (catkin_package)

CMake Warning at /opt/ros/kinetic/share/catkin/cmake/catkin_package.cmake:166 (message):
  catkin_package() DEPENDS on 'libavformat' but neither
  'libavformat_INCLUDE_DIRS' nor 'libavformat_LIBRARIES' is defined.
Call Stack (most recent call first):
  /opt/ros/kinetic/share/catkin/cmake/catkin_package.cmake:102 (_catkin_package)
  CMakeLists.txt:40 (catkin_package)

CMake Warning at /opt/ros/kinetic/share/catkin/cmake/catkin_package.cmake:166 (message):
  catkin_package() DEPENDS on 'libswscale' but neither
  'libswscale_INCLUDE_DIRS' nor 'libswscale_LIBRARIES' is defined.
Call Stack (most recent call first):
  /opt/ros/kinetic/share/catkin/cmake/catkin_package.cmake:102 (_catkin_package)
  CMakeLists.txt:40 (catkin_package)

cd /home/xx/bebop_ws/build/bebop_driver; catkin build --get-env bebop_driver | catkin env -si  /usr/bin/cmake /home/xx/bebop_ws/src/bebop_autonomy/bebop_driver --no-warn-unused-cli -DCATKIN_DEVEL_PREFIX=/home/xx/bebop_ws/devel/.private/bebop_driver -DCMAKE_INSTALL_PREFIX=/home/xx/bebop_ws/install -DCMAKE_BUILD_TYPE=RelWithDebInfo; cd -
.....................................................................................................................................................................................
Finished  <<< bebop_driver                         [ 4 minutes and 47.0 seconds ]                                                                                                    
Starting  >>> bebop_tools                                                                                                                                                            
Finished  <<< bebop_tools                          [ 7.2 seconds ]                                                                                                                   
[build] Summary: All 5 packages succeeded!                                                                                                                                           
[build]   Ignored:   1 packages were skipped or are blacklisted.                                                                                                                     
[build]   Warnings:  1 packages succeeded with warnings.                                                                                                                             
[build]   Abandoned: None.                                                                                                                                                           
[build]   Failed:    None.

mani-monaj, 2016-11-08:
Thanks @gwurb for the update. Although the warnings does not seem to affect the performance of the driver at the moment, they will break the build eventually in the future. I've marked this issue as an issue that needs to be fixed.

araujorayza, 2022-02-07:
> Oh, just to add another warning with Ubuntu 16.04 with ROS Kinetic
>
> ```
> Starting  >>> [...]

I am having the exact same problem. These warnings appear when trying to "catkin build" bebop autonomy on Ubuntu 16.04, ROS Kinetic.