AutonomyLab/bebop_autonomy #74: Document the way cmd_vel is applied to the drone and the simplified dynamic model

mani-monaj, 2016-11-01:


hoangtungdinh, 2016-11-01:
Hi Mani, I'm in need of this information urgently. Is there anywhere (in the code or in the SDK docs for example) we can find it or derive it?

Thanks for your help.

mani-monaj, 2016-11-03:
I've updated the documentation to indicate how different components of the `cmd_vel` topic affect the different degrees of the Bebop: [Sending Commands to Bebop — bebop_autonomy indigo-devel documentation](http://bebop-autonomy.readthedocs.io/en/latest/piloting.html#piloting)

You can also take a look at the following package as an example of how to control the velocity of the Bebop using these mappings (and a simplified dynamic model): [AutonomyLab/bebop_vel_ctrl: Velocity controller for Parrot Bebop Drone (ROS package)](https://github.com/AutonomyLab/bebop_vel_ctrl)

I will update the docs soon with more information about this dynamic model.

SeongjooShin, 2018-06-11:
Hello Mani, I would like to control the speed of the bebop. So I used AutonomyLab/bebop_vel ctrl package. I want to run the drone consistently at 0.3m/s. The speed is measured and the error is 0.1 to 0.6 m/s. So I think I'm using the package wrong. Could you tell me how to use it?