AutonomyLab/bebop_autonomy #76: Video stream frequently freezing on takeoff

wilpulling, 2016-11-04:
This has been happening almost every flight since yesterday (never previously). While the Bebop 2 is on the ground, it streams the video perfectly. Then, I take off. The video has always frozen for a second in the very first moments of takeoff, but now instead of coming back permanently, it comes back, gets a lot of bad artifacts, and then freezes. I retain control of the drone (can send flying commands), but I've confirmed that the subscriber to the image_raw topic is receiving the same frame over and over again. It's not blocking, and the system as a whole works, there are just no different frames coming in from the subscriber. (Alternately, could the subscriber not be getting high enough quality frames from the drone, and just re-sending the last "good" frame it got to the callback in each spin cycle?)

In the rare instances that I'm able to take off without the video freezing, the same thing ultimately happens between 2 and 5 minutes into my flight. 

Has anyone else experienced anything like this? Any recommended fixes?

mani-monaj, 2016-11-04:
Thanks for the detailed problem report @wilpulling . Are you using the latest changes I've merged into `indigo-devel`? If yes, have you cleaned up your workspace and re-build everything from scratch?

wilpulling, 2016-11-05:
It ended up being a hardware issue on my machine, so no change was actually needed. Thanks for the quick reply @mani-monaj!

mani-monaj, 2016-11-06:
You are welcome. Thanks for the follow-up.

closed 2016-11-06 by mani-monaj