AutonomyLab/bebop_autonomy #77: `Video streaming started` encounters `shared_ptr` error

weiweikong, 2016-12-06:
My platform is 
```
- Indigo
- Ubuntu 14.04
- Disco Platform
```
When running `roslaunch bebop_driver bebop_node.launch`, we received
```
 INFO] [1481012040.781340550]: Initializing nodelet with 4 worker threads.
[ INFO] [1481012040.842065688]: [BebopSDK] 16:14:00:841 | Bebop:225 - Bebop Cnstr()
[ INFO] [1481012040.842178176]: Nodelet Cstr
[ INFO] [1481012040.852265261]: Connecting to Bebop ...
[ INFO] [1481012040.858184131]: [CB] 16:14:00:858 | Ardrone3PilotingStateFlatTrimChanged:382 - [STATES] Enabling states/ardrone3/PilotingState/FlatTrimChanged
[ INFO] [1481012040.860158026]: [CB] 16:14:00:860 | Ardrone3PilotingStateFlyingStateChanged:426 - [STATES] Enabling states/ardrone3/PilotingState/FlyingStateChanged
[ INFO] [1481012040.862821376]: [CB] 16:14:00:862 | Ardrone3PilotingStateNavigateHomeStateChanged:528 - [STATES] Enabling states/ardrone3/PilotingState/NavigateHomeStateChanged
[ INFO] [1481012040.864662541]: [CB] 16:14:00:864 | Ardrone3PilotingStatePositionChanged:586 - [STATES] Enabling st
....
```
which seems that ROS has been correctly connect with Parrot Disco, but after a while, we receive the following errors
```
[ INFO] [1481012048.090233874]: Enabling video stream ...
[ WARN] [1481012048.090320800]: [BebopSDK] 16:14:08:090 | StartStreaming:359 - Video streaming started ...
[ INFO] [1481012048.090545332]: Nodelet lwp_id: 8498
[ INFO] [1481012048.090600575]: [CameraThread] thread lwp_id: 8922
[ INFO] [1481012048.090657502]: bebop_driver nodelet loaded.
[ INFO] [1481012048.090768550]: [AuxThread] thread lwp_id: 8923
bebop_driver_node: /usr/local/include/boost/smart_ptr/shared_ptr.hpp:648: typename boost::detail::sp_member_access<T>::type boost::shared_ptr<T>::operator->() const [with T = const bebop_msgs::Ardrone3PilotingStatePositionChanged_<std::allocator<void> >; typename boost::detail::sp_member_access<T>::type = const bebop_msgs::Ardrone3PilotingStatePositionChanged_<std::allocator<void> >*]: Assertion `px != 0' failed.
[bebop/bebop_driver-2] process has died [pid 8498, exit code -6, cmd /home/alvin/bebop_ws/devel/lib/bebop_driver/bebop_driver_node __name:=bebop_driver __log:=/home/alvin/.ros/log/f0e64cc0-bb8b-11e6-ac45-34de1a0f21c0/bebop-bebop_driver-2.log].
log file: /home/alvin/.ros/log/f0e64cc0-bb8b-11e6-ac45-34de1a0f21c0/bebop-bebop_driver-2*.log

```
I googled and find that it might be the `boost::shared_ptr` error, however, I am not so familiar with boost and `bebop_autonomy` source code. 

Would it be possible share some information about this issue？

Thanks.

mani-monaj, 2016-12-07:
Thanks for this bug report. Unfortunately I do not have access to a Disco FPV and I can not confirm that this driver works (at least without further modification) with the Disco. However, the issue you are facing might be caused by a bug in the driver's code.

Could you please change [this line](https://github.com/AutonomyLab/bebop_autonomy/blob/indigo-devel/bebop_driver/src/bebop_driver_nodelet.cpp#L539) from:

```c
if ((gps_state_ptr->header.stamp - last_gps_time).toSec() > util::eps)
```

to

```c
if (gps_state_ptr && ((gps_state_ptr->header.stamp - last_gps_time).toSec() > util::eps))
```

and recompile/test the driver?

weiweikong, 2016-12-12:
@mani-monaj Hi, thank you for your kind reply.
I recompiled the whole package as you recommended, and received the similar errors
```
...
[ WARN] [1481511105.839228628]: [ARNETWORK_Receiver] 10:51:45:839 | ARNETWORK_Receiver_ThreadRun:271 - [0x7f75342812c0] Received an old frame for buffer 127
[ WARN] [1481511105.841021120]: [ARNETWORK_Receiver] 10:51:45:840 | ARNETWORK_Receiver_ThreadRun:271 - [0x7f75342812c0] Received an old frame for buffer 127
[ WARN] [1481511105.841111476]: [ARNETWORK_Receiver] 10:51:45:841 | ARNETWORK_Receiver_ThreadRun:271 - [0x7f75342812c0] Received an old frame for buffer 127
[ INFO] [1481511109.201209276]: Dynamic reconfigure callback with level: -1
[ INFO] [1481511109.232303990]: Enabling video stream ...
[ WARN] [1481511109.232424631]: [BebopSDK] 10:51:49:232 | StartStreaming:359 - Video streaming started ...
[ INFO] [1481511109.232769860]: Nodelet lwp_id: 10372
[ INFO] [1481511109.232848724]: [CameraThread] thread lwp_id: 10758
[ INFO] [1481511109.232944684]: [AuxThread] thread lwp_id: 10759
[ INFO] [1481511109.233010189]: bebop_driver nodelet loaded.
bebop_driver_node: /usr/include/boost/smart_ptr/shared_ptr.hpp:653: typename boost::detail::sp_member_access<T>::type boost::shared_ptr<T>::operator->() const [with T = const bebop_msgs::Ardrone3PilotingStateSpeedChanged_<std::allocator<void> >; typename boost::detail::sp_member_access<T>::type = const bebop_msgs::Ardrone3PilotingStateSpeedChanged_<std::allocator<void> >*]: Assertion `px != 0' failed.
[bebop/bebop_driver-2] process has died [pid 10372, exit code -6, cmd /home/amax/Workspace/120.bebop_autonomy/devel/lib/bebop_driver/bebop_driver_node __name:=bebop_driver __log:=/home/amax/.ros/log/df071712-c015-11e6-9a86-001d4320131b/bebop-bebop_driver-2.log].
log file: /home/amax/.ros/log/df071712-c015-11e6-9a86-001d4320131b/bebop-bebop_driver-2*.log
^C[rosout-1] killing on exit
[master] killing on exit
^Cshutting down processing monitor...
... shutting down processing monitor complete
done
```
by running 
```
roslaunch bebop_driver bebop_node.launch 
```
Should we change all the `shared_ptr` related codes? Thanks.

mani-monaj, 2016-12-18:
Thanks for the update. I think the problem is that the driver's code does not properly guard against all cases of NULL [shared] pointers before accessing them. 

Looking at the code and your error report, I guess the problem is from [this line](https://github.com/AutonomyLab/bebop_autonomy/blob/indigo-devel/bebop_driver/src/bebop_driver_nodelet.cpp#L562):

```c
if ((speed_esd_ptr->header.stamp - last_speed_time).toSec() > util::eps) 
```

Could you please change it to 

```c
if (speed_esd_ptr && ((speed_esd_ptr->header.stamp - last_speed_time).toSec() > util::eps))
````

Then recompile and test?

Since I do not have access to a Disco, it is helpful if you could run the driver in a debugger such as `gdb`. This is what I use to run a Nodelet with `gdb`:

```
$ rosrun --prefix "gdb --args" nodelet nodelet standalone bebop_driver/BebopDriverNodelet
```

Prior to running this command, please `catkin clean` your workspace and recompile it with `catkin build -DCMAKE_BUILD_TYPE=Debug [or -DCMAKE_BUILD_TYPE=RelWithDebInfo`. When `gdb` starts, use the `run` command to execute the code. When it SEGFAULTS, use `bt` to get the call stack that led to the fault. You may find [this tutorial helpful](https://beej.us/guide/bggdb/).

weiweikong, 2016-12-23:
@mani-monaj Hi, sorry for my late testing. The camera streams was receivedl, as shown 

[image]

but there were some warnings related to H264Filter
```
[ WARN] [1482484759.094475629]: [ARSTREAM2_H264Filter] 17:19:19:094 | ARSTREAM2_H264Filter_fillMissingSlices:1660 - #222 AUTS:879945444 Missing NALU is probably a slice
[ WARN] [1482484759.094522639]: [ARSTREAM2_H264Filter] 17:19:19:094 | ARSTREAM2_H264Filter_fillMissingSlices:1703 - #222 AUTS:879945444 Current slice is not a P-slice, aborting
[ WARN] [1482484759.097729950]: [ARSTREAM2_H264Filter] 17:19:19:097 | ARSTREAM2_H264Filter_enqueueCurrentAu:1006 - AU output cancelled (!outputIncompleteAu)
[ WARN] [1482484759.925153540]: [ARSTREAM2_H264Filter] 17:19:19:925 | ARSTREAM2_H264Filter_fillMissingSlices:1621 - #226 AUTS:880779600 previousSliceFirstMb:960 previousSliceMbCount:160 currentSliceFirstM

```

Another problem is, I could not access the autopilot state, following are the topic list and service list.
```
/bebop/autoflight/navigate_home
/bebop/autoflight/pause
/bebop/autoflight/start
/bebop/autoflight/stop
/bebop/bebop_driver/parameter_descriptions
/bebop/bebop_driver/parameter_updates
/bebop/camera_control
/bebop/camera_info
/bebop/cmd_vel
/bebop/fix
/bebop/flattrim
/bebop/flip
/bebop/image_raw
/bebop/image_raw/compressed
/bebop/image_raw/compressed/parameter_descriptions
/bebop/image_raw/compressed/parameter_updates
/bebop/image_raw/compressedDepth
/bebop/image_raw/compressedDepth/parameter_descriptions
/bebop/image_raw/compressedDepth/parameter_updates
/bebop/image_raw/theora
/bebop/image_raw/theora/parameter_descriptions
/bebop/image_raw/theora/parameter_updates
/bebop/joint_states
/bebop/land
/bebop/odom
/bebop/record
/bebop/reset
/bebop/snapshot
/bebop/states/ardrone3/CameraState/Orientation
/bebop/states/ardrone3/GPSState/NumberOfSatelliteChanged
/bebop/states/ardrone3/MediaStreamingState/VideoEnableChanged
/bebop/states/ardrone3/PilotingState/AltitudeChanged
/bebop/states/ardrone3/PilotingState/AttitudeChanged
/bebop/states/ardrone3/PilotingState/FlatTrimChanged
/bebop/states/ardrone3/PilotingState/FlyingStateChanged
/bebop/states/ardrone3/PilotingState/NavigateHomeStateChanged
/bebop/states/ardrone3/PilotingState/PositionChanged
/bebop/states/ardrone3/PilotingState/SpeedChanged
/bebop/states/common/CommonState/BatteryStateChanged
/bebop/states/common/CommonState/WifiSignalChanged
/bebop/states/common/FlightPlanState/AvailabilityStateChanged
/bebop/states/common/FlightPlanState/ComponentStateListChanged
/bebop/states/common/MavlinkState/MavlinkFilePlayingStateChanged
/bebop/states/common/MavlinkState/MavlinkPlayErrorStateChanged
/bebop/states/common/OverHeatState/OverHeatChanged
/bebop/takeoff
/rosout
/rosout_agg
/tf
```
these optics seems dead becasue I could not use `rostopic echo` to get data and the `rqt_graph` shows nothing about this topcis.
```
/bebop/bebop_driver/get_loggers
/bebop/bebop_driver/list
/bebop/bebop_driver/load_nodelet
/bebop/bebop_driver/set_logger_level
/bebop/bebop_driver/set_parameters
/bebop/bebop_driver/unload_nodelet
/bebop/image_raw/compressed/set_parameters
/bebop/image_raw/compressedDepth/set_parameters
/bebop/image_raw/theora/set_parameters
/bebop/set_camera_info
/rosout/get_loggers
/rosout/set_logger_level
```

mani-monaj, 2016-12-28:
Thanks @weiweikong for the update.

1. To clarify, does the changes I propose fix the segfault issue? Could you please send a Pull Request with all those changes if that's the case?
2. Could you please run the driver, do a `rosbag recored -a` and upload/share the resulting bag file with me? It can be huge since it will include non-compressed image topics. The order I suggest to run the commands is:

1. Run `roscore`
2. `rosbag record -a`
3. Launch the driver
4. Kill the recored script after 30 seconds or so.

weiweikong, 2017-01-02:
Sorry for my late replay.
## `rosbag record -a` failed
When I run the command line
```
rosbag record -a
```
there were following warnings and the nodes crashed
```
[ INFO] [1483364228.753029394]: [CB] 21:37:08:752 | Update:1420 -   No
[ INFO] [1483364228.754733563]: Value for NetworkSettingsWifiSelectionBand recved: 0
[ INFO] [1483364228.754828603]: [CB] 21:37:08:754 | Update:1438 - Checking if NetworkSettingsWifiSelectionBand exists in params ...
[ INFO] [1483364228.755995494]: [CB] 21:37:08:755 | Update:1441 -   No
[ INFO] [1483364228.757409919]: Value for NetworkSettingsWifiSelectionChannel recved: 6
[ INFO] [1483364228.757477382]: [CB] 21:37:08:757 | Update:1459 - Checking if NetworkSettingsWifiSelectionChannel exists in params ...
[ INFO] [1483364228.758744974]: [CB] 21:37:08:758 | Update:1462 -   No
[ INFO] [1483364228.771349406]: Value for PictureSettingsVideoStabilizationModeMode recved: 3
[ INFO] [1483364228.771445612]: [CB] 21:37:08:771 | Update:1539 - Checking if PictureSettingsVideoStabilizationModeMode exists in params ...
[ INFO] [1483364228.772760891]: [CB] 21:37:08:772 | Update:1547 -   Yes
[ INFO] [1483364228.780480640]: Value for GPSSettingsReturnHomeDelayDelay recved: 10
[ INFO] [1483364228.780566953]: [CB] 21:37:08:780 | Update:1939 - Checking if GPSSettingsReturnHomeDelayDelay exists in params ...
[ INFO] [1483364228.781891282]: [CB] 21:37:08:781 | Update:1947 -   Yes
[ INFO] [1483364228.783519477]: Value for GPSSettingsHomeTypeType recved: 0
[ INFO] [1483364228.783627727]: [CB] 21:37:08:783 | Update:1859 - Checking if GPSSettingsHomeTypeType exists in params ...
[ INFO] [1483364228.784873432]: [CB] 21:37:08:784 | Update:1867 -   Yes
[ INFO] [1483364232.038771032]: Dynamic reconfigure callback with level: -1
[ INFO] [1483364232.038956437]: [CB] 21:37:12:038 | UpdateBebopFromROS:309 - PilotingSettingsMaxDistanceValue changed!
[ INFO] [1483364232.039010804]: [CB] 21:37:12:038 | UpdateBebopFromROS:323 - Sending PilotingSettingsMaxDistance changes to bebop
[ INFO] [1483364232.056417989]: Value for PilotingSettingsMaxDistanceValue recved: 200
[ INFO] [1483364232.056491649]: [CB] 21:37:12:056 | Update:351 - Checking if PilotingSettingsMaxDistanceValue exists in params ...
[ INFO] [1483364232.057692696]: [CB] 21:37:12:057 | Update:359 -   Yes
[ INFO] [1483364232.072941835]: Enabling video stream ...
[ WARN] [1483364232.073075975]: [BebopSDK] 21:37:12:073 | StartStreaming:359 - Video streaming started ...
[ INFO] [1483364232.073392469]: Nodelet lwp_id: 19021
[ INFO] [1483364232.073478692]: [CameraThread] thread lwp_id: 19433
[ INFO] [1483364232.073590521]: [AuxThread] thread lwp_id: 19434
[ INFO] [1483364232.073664378]: bebop_driver nodelet loaded.
[ WARN] [1483364232.155402193]: [ARSTREAM2_H264Filter] 21:37:12:155 | ARSTREAM2_H264Filter_RtpReceiverNaluCallback:2174 - ARSTREAM2_RtpReceiver NALU buffer is too small, truncated AU (or maybe it's the fi
[ WARN] [1483364232.191659456]: [ARSTREAM2_H264Filter] 21:37:12:191 | ARSTREAM2_H264Filter_enqueueCurrentAu:1018 - AU output cancelled (waitForSync)
[ WARN] [1483364232.327678192]: [ARSTREAM2_H264Filter] 21:37:12:327 | ARSTREAM2_H264Filter_sync:550 - SPS/PPS sync OK
[ INFO] [1483364232.327748749]: [BebopSDK] 21:37:12:327 | DecoderConfigCallback:147 - H264 configuration packet received: #SPS: 26 #PPS: 9 (MP4? 0)
[ WARN] [1483364232.327901892]: [ARSTREAM2_H264Filter] 21:37:12:327 | ARSTREAM2_H264Filter_generateGrayIFrame:1254 - Waiting for a slice to generate gray I-frame
[ WARN] [1483364232.327967009]: [ARSTREAM2_H264Filter] 21:37:12:327 | ARSTREAM2_H264Filter_generateGrayIFrame:1254 - Waiting for a slice to generate gray I-frame
[ WARN] [1483364232.328274023]: [ARSTREAM2_H264Filter] 21:37:12:328 | ARSTREAM2_H264Filter_generateGrayIFrame:1322 - Gray I slice NALU output size: 3609
[ INFO] [1483364232.328330583]: [BebopSDK] 21:37:12:328 | FrameReceivedCallback:174 - Frame Recv & Decode LWP id: 19436
[ INFO] [1483364232.329614947]: [Decoder] 21:37:12:329 | InitCodec:117 - H264 Codec is partially initialized!
[ INFO] [1483364232.329667044]: [Decoder] 21:37:12:329 | Decode:263 - Updating H264 codec parameters (Buffer Size: 35) ...
[ERROR] [1483364232.337201280]: [Decoder] 21:37:12:337 | Decode:302 - Frame size changed to 1280 x 720
[ INFO] [1483364232.337291103]: [Decoder] 21:37:12:337 | ReallocateBuffers:123 - Buffer reallocation request
[ INFO] [1483364232.345410540]: camera calibration URL: package://bebop_driver/data/bebop_camera_calib.yaml
[ WARN] [1483364232.849684167]: [ARSTREAM2_H264Filter] 21:37:12:849 | ARSTREAM2_H264Filter_fillMissingSlices:1660 - #4 AUTS:163231289 Missing NALU is probably a slice
[ WARN] [1483364232.849788518]: [ARSTREAM2_H264Filter] 21:37:12:849 | ARSTREAM2_H264Filter_fillMissingSlices:1688 - #4 AUTS:163231289 Streaming info is not available
[ WARN] [1483364232.879619300]: [ARSTREAM2_H264Filter] 21:37:12:879 | ARSTREAM2_H264Filter_enqueueCurrentAu:1006 - AU output cancelled (!outputIncompleteAu)
[ WARN] [1483364233.867120714]: [ARSTREAM2_H264Filter] 21:37:13:867 | ARSTREAM2_H264Filter_fillMissingSlices:1621 - #9 AUTS:164232278 previousSliceFirstMb:2560 previousSliceMbCount:160 currentSliceFirstMb
[ WARN] [1483364233.867220104]: [ARSTREAM2_H264Filter] 21:37:13:867 | ARSTREAM2_H264Filter_fillMissingSlices:1660 - #9 AUTS:164232278 Missing NALU is probably a slice
[... cut ...]
[bebop/bebop_driver-1] process has died [pid 16478, exit code -11, cmd /home/amax/Workspace/120.bebop_autonomy/devel/lib/bebop_driver/bebop_driver_node __name:=bebop_driver __log:=/home/amax/.ros/log/67d14ee8-d0ef-11e6-ade1-001d4320131b/bebop-bebop_driver-1.log].
log file: /home/amax/.ros/log/67d14ee8-d0ef-11e6-ade1-001d4320131b/bebop-bebop_driver-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
```

## Other topics information
Would it be possible to access autopilot data such as IMU, gyro raw data through the driver by ROS node?

weiweikong, 2017-01-11:
@mani-monaj  Any update about other topic information with ROS node?

weiweikong, 2017-01-19:
Any update @mani-monaj ?

mani-monaj, 2017-02-08:
@weiweikong Sorry for my delayed response. I will be back on the development track very soon. I will update this issue with my answers. Any update on your side?

weiweikong, 2017-02-11:
@mani-monaj No more update. : (

thomas-bamford, 2017-06-23:
Hello @weiweikong,
Sorry for the late reply. The bebop_autonomy ros driver is currently limited to the information that is provided by the Parrot, so information such as IMU, gyro raw data is not available. Here is a link to the list of information that these Parrot drones can provide [http://developer.parrot.com/docs/reference/all/index.html](url). We hope to work towards providing support for the Disco platform (which upgrading to the latest SDK will add some new information for).

thomas-bamford, 2018-07-03:
Disco is supported in #168.

closed 2018-07-03 by thomas-bamford