AutonomyLab/bebop_autonomy #80: Take off on Disco

eder1234, 2016-12-09:
Hi,
I am using this:

- Ubuntu 14.04
- ROS Indigo
- Bebop_autonomy
- Disco FPV

My goal is to create a package to fly the drone and receive the images as I am doing with the Bebop 2.
But, at this moment, I had not successfully taken off using the command /bebop/takeoff as I had done with the Bebop 2 before. When I send the message I look at the communication with the command rqt_graph and everything seems normal, but the drone does not start the motor. Is there another command to, or another way, to take off the Disco?

Thank you very much.
Eder Alejandro Rodriguez Martinez

mani-monaj, 2016-12-18:
Thanks for the bug report @eder1234 . Unfortunately I do not have access to a *Disco* at the moment. Could you please share more information about the current state of support with *Disco*? Does the driver run successfully? Can you receive images or ready any data from the drone using the driver? Could you please paste the complete console output here?

eder1234, 2016-12-20:
Hi Man
Here is the message when I run the driver.
[0m[ INFO] [1482244634.766589083]: Initializing nodelet with 4 worker threads.[0m
[0m[ INFO] [1482244634.809607169]: [BebopSDK] 15:37:14:809 | Bebop:225 - Bebop Cnstr()[0m
[0m[ INFO] [1482244634.809652350]: Nodelet Cstr[0m
[0m[ INFO] [1482244634.816541630]: Connecting to Bebop ...[0m
[0m[ INFO] [1482244634.819845208]: [CB] 15:37:14:819 | Ardrone3PilotingStateFlatTrimChanged:382 - [STATES] Enabling states/ardrone3/PilotingState/FlatTrimChanged[0m
[0m[ INFO] [1482244634.821221083]: [CB] 15:37:14:821 | Ardrone3PilotingStateFlyingStateChanged:426 - [STATES] Enabling states/ardrone3/PilotingState/FlyingStateChanged[0m
[0m[ INFO] [1482244634.822905359]: [CB] 15:37:14:822 | Ardrone3PilotingStateNavigateHomeStateChanged:528 - [STATES] Enabling states/ardrone3/PilotingState/NavigateHomeStateChanged[0m
[0m[ INFO] [1482244634.824098971]: [CB] 15:37:14:824 | Ardrone3PilotingStatePositionChanged:586 - [STATES] Enabling states/ardrone3/PilotingState/PositionChanged[0m
[0m[ INFO] [1482244634.825236332]: [CB] 15:37:14:825 | Ardrone3PilotingStateSpeedChanged:651 - [STATES] Enabling states/ardrone3/PilotingState/SpeedChanged[0m
[0m[ INFO] [1482244634.826320252]: [CB] 15:37:14:826 | Ardrone3PilotingStateAttitudeChanged:716 - [STATES] Enabling states/ardrone3/PilotingState/AttitudeChanged[0m
[0m[ INFO] [1482244634.827901639]: [CB] 15:37:14:827 | Ardrone3PilotingStateAltitudeChanged:832 - [STATES] Enabling states/ardrone3/PilotingState/AltitudeChanged[0m
[0m[ INFO] [1482244634.832280542]: [CB] 15:37:14:832 | Ardrone3MediaStreamingStateVideoEnableChanged:1245 - [STATES] Enabling states/ardrone3/MediaStreamingState/VideoEnableChanged[0m
[0m[ INFO] [1482244634.833942927]: [CB] 15:37:14:833 | Ardrone3CameraStateOrientation:1347 - [STATES] Enabling states/ardrone3/CameraState/Orientation[0m
[0m[ INFO] [1482244634.836429212]: [CB] 15:37:14:836 | Ardrone3GPSStateNumberOfSatelliteChanged:1565 - [STATES] Enabling states/ardrone3/GPSState/NumberOfSatelliteChanged[0m
[0m[ INFO] [1482244634.843784543]: [CB] 15:37:14:843 | CommonCommonStateBatteryStateChanged:146 - [STATES] Enabling states/common/CommonState/BatteryStateChanged[0m
[0m[ INFO] [1482244634.848406687]: [CB] 15:37:14:848 | CommonCommonStateWifiSignalChanged:508 - [STATES] Enabling states/common/CommonState/WifiSignalChanged[0m
[0m[ INFO] [1482244634.853547321]: [CB] 15:37:14:853 | CommonOverHeatStateOverHeatChanged:963 - [STATES] Enabling states/common/OverHeatState/OverHeatChanged[0m
[0m[ INFO] [1482244634.855245656]: [CB] 15:37:14:855 | CommonMavlinkStateMavlinkFilePlayingStateChanged:1058 - [STATES] Enabling states/common/MavlinkState/MavlinkFilePlayingStateChanged[0m
[0m[ INFO] [1482244634.856495941]: [CB] 15:37:14:856 | CommonMavlinkStateMavlinkPlayErrorStateChanged:1123 - [STATES] Enabling states/common/MavlinkState/MavlinkPlayErrorStateChanged[0m
[0m[ INFO] [1482244634.860036700]: [CB] 15:37:14:860 | CommonFlightPlanStateAvailabilityStateChanged:1457 - [STATES] Enabling states/common/FlightPlanState/AvailabilityStateChanged[0m
[0m[ INFO] [1482244634.861244477]: [CB] 15:37:14:861 | CommonFlightPlanStateComponentStateListChanged:1508 - [STATES] Enabling states/common/FlightPlanState/ComponentStateListChanged[0m
[0m[ INFO] [1482244634.875121001]: [ARCONTROLLER_Network] 15:37:14:875 | ARCONTROLLER_Network_GetAvailableSocketPort:108 - d2c_port port: 44360[0m
[0m[ INFO] [1482244634.875296858]: [ARCONTROLLER_Stream2] 15:37:14:875 | ARCONTROLLER_Stream2_Open_Socket:122 - udp local port stream: 56436[0m
[0m[ INFO] [1482244634.875326239]: [ARCONTROLLER_Stream2] 15:37:14:875 | ARCONTROLLER_Stream2_Open_Socket:122 - udp local port control: 43145[0m
[0m[ INFO] [1482244634.961508511]: [BebopSDK] 15:37:14:961 | CommandReceivedCallback:113 - Command Received Callback LWP id is: 5129[0m
[0m[ INFO] [1482244635.152472204]: [BebopSDK] 15:37:15:152 | Connect:321 - BebopSDK inited, lwp_id: 4978[0m
[33m[ WARN] [1482244635.152604266]: Resetting all settings ...[0m
[0m[ INFO] [1482244635.152678773]: [BebopSDK] 15:37:15:152 | ResetAllSettings:409 - All settings of the drone have been reset to default values.[0m
[0m[ INFO] [1482244635.207019467]: Value for GPSSettingsHomeTypeType recved: 0[0m
[0m[ INFO] [1482244635.207056541]: [CB] 15:37:15:207 | Update:1859 - Checking if GPSSettingsHomeTypeType exists in params ...[0m
[0m[ INFO] [1482244635.207860044]: [CB] 15:37:15:207 | Update:1862 -   No[0m
[0m[ INFO] [1482244635.216029535]: Value for PictureSettingsVideoRecordingModeMode recved: 0[0m
[0m[ INFO] [1482244635.216079853]: [CB] 15:37:15:216 | Update:1619 - Checking if PictureSettingsVideoRecordingModeMode exists in params ...[0m
[0m[ INFO] [1482244635.216921133]: [CB] 15:37:15:216 | Update:1622 -   No[0m
[0m[ INFO] [1482244635.217697236]: Value for PictureSettingsVideoFramerateFramerate recved: 2[0m
[0m[ INFO] [1482244635.217723179]: [CB] 15:37:15:217 | Update:1699 - Checking if PictureSettingsVideoFramerateFramerate exists in params ...[0m
[0m[ INFO] [1482244635.218279119]: [CB] 15:37:15:218 | Update:1702 -   No[0m
[0m[ INFO] [1482244635.219192746]: Value for PictureSettingsVideoResolutionsType recved: 0[0m
[0m[ INFO] [1482244635.219229394]: [CB] 15:37:15:219 | Update:1779 - Checking if PictureSettingsVideoResolutionsType exists in params ...[0m
[0m[ INFO] [1482244635.219934185]: [CB] 15:37:15:219 | Update:1782 -   No[0m
[... cut ...]
[bebop/bebop_driver-1] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
[1mdone[0m
I also run the tool rqt_image_view and select the topics /bebop/image_raw, /bebop/image_raw/compressed,
/bebop/image_raw/compressedDepth, /bebop/image_raw/theora, the displayed messages are at the bottom.
The image raw is shown in the window.
Also I published the message to take off as I had done with the bebop2 but nothing happens.
Can you include me in the contributors list?
Regards

eder1234, 2016-12-20:
Hi Mani,

By the way I forgot to ask this in the last email.

When I receive the camera's information in the /bebop/camera_info (for the bebop2), I get the vector

P = [400.182373, 0, 323.081936, 0, 0, 403.197845, 172.320207, 0, 0, 0, 1, 0],

which, I guess, is the projection matrix. Is it a 4 x 3 or a 3 x 4 matrix (I had found both in literature)?

Thank you very much.

Regards

________________________________
De: eder rodriguez <eder_rdz_mtz@hotmail.com>
Enviado: martes, 20 de diciembre de 2016 09:01:17 a. m.
Para: AutonomyLab/bebop_autonomy
Asunto: Re: [AutonomyLab/bebop_autonomy] Take off on Disco (#80)

Hi Man
Here is the message when I run the driver.
[0m[ INFO] [1482244634.766589083]: Initializing nodelet with 4 worker threads.[0m
[0m[ INFO] [1482244634.809607169]: [BebopSDK] 15:37:14:809 | Bebop:225 - Bebop Cnstr()[0m
[0m[ INFO] [1482244634.809652350]: Nodelet Cstr[0m
[0m[ INFO] [1482244634.816541630]: Connecting to Bebop ...[0m
[0m[ INFO] [1482244634.819845208]: [CB] 15:37:14:819 | Ardrone3PilotingStateFlatTrimChanged:382 - [STATES] Enabling states/ardrone3/PilotingState/FlatTrimChanged[0m
[0m[ INFO] [1482244634.821221083]: [CB] 15:37:14:821 | Ardrone3PilotingStateFlyingStateChanged:426 - [STATES] Enabling states/ardrone3/PilotingState/FlyingStateChanged[0m
[0m[ INFO] [1482244634.822905359]: [CB] 15:37:14:822 | Ardrone3PilotingStateNavigateHomeStateChanged:528 - [STATES] Enabling states/ardrone3/PilotingState/NavigateHomeStateChanged[0m
[0m[ INFO] [1482244634.824098971]: [CB] 15:37:14:824 | Ardrone3PilotingStatePositionChanged:586 - [STATES] Enabling states/ardrone3/PilotingState/PositionChanged[0m
[0m[ INFO] [1482244634.825236332]: [CB] 15:37:14:825 | Ardrone3PilotingStateSpeedChanged:651 - [STATES] Enabling states/ardrone3/PilotingState/SpeedChanged[0m
[0m[ INFO] [1482244634.826320252]: [CB] 15:37:14:826 | Ardrone3PilotingStateAttitudeChanged:716 - [STATES] Enabling states/ardrone3/PilotingState/AttitudeChanged[0m
[0m[ INFO] [1482244634.827901639]: [CB] 15:37:14:827 | Ardrone3PilotingStateAltitudeChanged:832 - [STATES] Enabling states/ardrone3/PilotingState/AltitudeChanged[0m
[0m[ INFO] [1482244634.832280542]: [CB] 15:37:14:832 | Ardrone3MediaStreamingStateVideoEnableChanged:1245 - [STATES] Enabling states/ardrone3/MediaStreamingState/VideoEnableChanged[0m
[0m[ INFO] [1482244634.833942927]: [CB] 15:37:14:833 | Ardrone3CameraStateOrientation:1347 - [STATES] Enabling states/ardrone3/CameraState/Orientation[0m
[0m[ INFO] [1482244634.836429212]: [CB] 15:37:14:836 | Ardrone3GPSStateNumberOfSatelliteChanged:1565 - [STATES] Enabling states/ardrone3/GPSState/NumberOfSatelliteChanged[0m
[0m[ INFO] [1482244634.843784543]: [CB] 15:37:14:843 | CommonCommonStateBatteryStateChanged:146 - [STATES] Enabling states/common/CommonState/BatteryStateChanged[0m
[0m[ INFO] [1482244634.848406687]: [CB] 15:37:14:848 | CommonCommonStateWifiSignalChanged:508 - [STATES] Enabling states/common/CommonState/WifiSignalChanged[0m
[0m[ INFO] [1482244634.853547321]: [CB] 15:37:14:853 | CommonOverHeatStateOverHeatChanged:963 - [STATES] Enabling states/common/OverHeatState/OverHeatChanged[0m
[0m[ INFO] [1482244634.855245656]: [CB] 15:37:14:855 | CommonMavlinkStateMavlinkFilePlayingStateChanged:1058 - [STATES] Enabling states/common/MavlinkState/MavlinkFilePlayingStateChanged[0m
[0m[ INFO] [1482244634.856495941]: [CB] 15:37:14:856 | CommonMavlinkStateMavlinkPlayErrorStateChanged:1123 - [STATES] Enabling states/common/MavlinkState/MavlinkPlayErrorStateChanged[0m
[0m[ INFO] [1482244634.860036700]: [CB] 15:37:14:860 | CommonFlightPlanStateAvailabilityStateChanged:1457 - [STATES] Enabling states/common/FlightPlanState/AvailabilityStateChanged[0m
[0m[ INFO] [1482244634.861244477]: [CB] 15:37:14:861 | CommonFlightPlanStateComponentStateListChanged:1508 - [STATES] Enabling states/common/FlightPlanState/ComponentStateListChanged[0m
[0m[ INFO] [1482244634.875121001]: [ARCONTROLLER_Network] 15:37:14:875 | ARCONTROLLER_Network_GetAvailableSocketPort:108 - d2c_port port: 44360[0m
[0m[ INFO] [1482244634.875296858]: [ARCONTROLLER_Stream2] 15:37:14:875 | ARCONTROLLER_Stream2_Open_Socket:122 - udp local port stream: 56436[0m
[0m[ INFO] [1482244634.875326239]: [ARCONTROLLER_Stream2] 15:37:14:875 | ARCONTROLLER_Stream2_Open_Socket:122 - udp local port control: 43145[0m
[0m[ INFO] [1482244634.961508511]: [BebopSDK] 15:37:14:961 | CommandReceivedCallback:113 - Command Received Callback LWP id is: 5129[0m
[0m[ INFO] [1482244635.152472204]: [BebopSDK] 15:37:15:152 | Connect:321 - BebopSDK inited, lwp_id: 4978[0m
[33m[ WARN] [1482244635.152604266]: Resetting all settings ...[0m
[0m[ INFO] [1482244635.152678773]: [BebopSDK] 15:37:15:152 | ResetAllSettings:409 - All settings of the drone have been reset to default values.[0m
[0m[ INFO] [1482244635.207019467]: Value for GPSSettingsHomeTypeType recved: 0[0m
[0m[ INFO] [1482244635.207056541]: [CB] 15:37:15:207 | Update:1859 - Checking if GPSSettingsHomeTypeType exists in params ...[0m
[0m[ INFO] [1482244635.207860044]: [CB] 15:37:15:207 | Update:1862 -   No[0m
[0m[ INFO] [1482244635.216029535]: Value for PictureSettingsVideoRecordingModeMode recved: 0[0m
[0m[ INFO] [1482244635.216079853]: [CB] 15:37:15:216 | Update:1619 - Checking if PictureSettingsVideoRecordingModeMode exists in params ...[0m
[0m[ INFO] [1482244635.216921133]: [CB] 15:37:15:216 | Update:1622 -   No[0m
[... cut ...]
[bebop/bebop_driver-1] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
[1mdone[0m
I also run the tool rqt_image_view and select the topics /bebop/image_raw, /bebop/image_raw/compressed,
/bebop/image_raw/compressedDepth, /bebop/image_raw/theora, the displayed messages are at the bottom.
The image raw is shown in the window.
Also I published the message to take off as I had done with the bebop2 but nothing happens.
Can you include me in the contributors list?
Regards

mani-monaj, 2016-12-28:
Thanks for the detailed console log.

Unfortunately I can not see any hints about the root cause of the issue in the log. Based on Parrot's documentation, the `takeoff` command should work both on *Bebop* and *Disco*. Can you fly the *Disco* with Parrot's app without any issue?

> which, I guess, is the projection matrix. Is it a 4 x 3 or a 3 x 4 matrix (I had found both in literature)?

It's a [3x4 matrix](http://docs.ros.org/api/sensor_msgs/html/msg/CameraInfo.html).

eder1234, 2016-12-29:
I had not tried yet to take off the parrot with the app (free flight). When I try it so, I let you know.

Thank you.

Happy Holidays

thomas-bamford, 2017-06-23:
Hello @eder1234,
Have you made any progress with this work? I have access to a Disco to work on providing support for it in this ROS package. I will try this weekend to takeoff and fly the Disco with ROS and will provide any launch files or tips in the documentation, so if you have any pointers it would be very much appreciated.
Thanks.

eder1234, 2017-06-23:
Hello,

Actually I had not access to the drone, some my remarks are the same than before.

My supervisor has done some experiments. Let me ask him this afternoon and I give you the remarks.

Alejandro

eder1234, 2017-06-30:
Hello,

I got the answer of my supervisor:

Problèmes Disco/Free flight pro :
1. L’application sous android s’affiche dans le mauvais sens de l’écran de la tablette FlightSight HD900 et on ne peut pas réorienter l’application. C’est génant avec un pare-soleil qui devient inutilisable.
2. L’exportation des données de vol avec le logiciel Flightdata manager en fichier CSV fait apparaître de nombreuses valeurs nulles de roulis, tangage et lacet (cf. pièce jointe).

Problems Disco/Free flight pro:

  1.  The android's app displays the screen in a wrong directions in the tablet FlightSight HD900 and it can't be reoriented by the app. It is incovenient when using the viewfinder.
  2.  In the data exploration, of the software Flighdata manager, the CSV contains a lot of nul values for the rotation readings (displayed in the picture).

I hope this information is helpful. If you have more doubts, let me know.

Cordially

Eder Alejandro Rodriguez Martinez

________________________________
De: eder rodriguez <eder_rdz_mtz@hotmail.com>
Enviado: viernes, 23 de junio de 2017 04:32:17 a. m.
Para: AutonomyLab/bebop_autonomy
Asunto: Re: [AutonomyLab/bebop_autonomy] Take off on Disco (#80)

Hello,

Actually I had not access to the drone, some my remarks are the same than before.

My supervisor has done some experiments. Let me ask him this afternoon and I give you the remarks.

Alejandro

thomas-bamford, 2018-07-03:
Support added in #168.

closed 2018-07-03 by thomas-bamford