AutonomyLab/bebop_autonomy #81: Bebop Autonomy - cannot run the drivers. PLEASE HELP!!!

kenisvery, 2016-12-19:
Just got a new Bebop 2 drones the installation were done but i could run the driver. Below are the error shows. 

kenisvery@kenisvery-Sensei:~$ source ~/bebop_ws/devel/setup.bash
kenisvery@kenisvery-Sensei:~$ roslaunch bebop_driver bebop_node.launch
... logging to /home/kenisvery/.ros/log/7514a9ce-c339-11e6-a825-0800275ad713/roslaunch-kenisvery-Sensei-15254.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://kenisvery-Sensei:38304/

SUMMARY
========

PARAMETERS
 * /bebop/bebop_driver/bebop_ip: 192.168.42.1
 * /bebop/bebop_driver/camera_info_url: package://bebop_d...
 * /bebop/bebop_driver/cmd_vel_timeout: 0.2
 * /bebop/bebop_driver/odom_frame_id: odom
 * /bebop/bebop_driver/publish_odom_tf: True
 * /bebop/bebop_driver/reset_settings: True
 * /bebop/bebop_driver/states/enable_altitudechanged: True
 * /bebop/bebop_driver/states/enable_autotakeoffmodechanged: True
 * /bebop/bebop_driver/states/enable_camerastate_orientation: True
 * /bebop/bebop_driver/states/enable_commonstate_batterystatechanged: True
 * /bebop/bebop_driver/states/enable_commonstate_wifisignalchanged: True
 * /bebop/bebop_driver/states/enable_controllerstate_ispilotingchanged: True
 * /bebop/bebop_driver/states/enable_flightplanstate_availabilitystatechanged: True
 * /bebop/bebop_driver/states/enable_flightplanstate_componentstatelistchanged: True
 * /bebop/bebop_driver/states/enable_gpsstate_numberofsatellitechanged: True
 * /bebop/bebop_driver/states/enable_mavlinkstate_mavlinkfileplayingstatechanged: True
 * /bebop/bebop_driver/states/enable_mavlinkstate_mavlinkplayerrorstatechanged: True
 * /bebop/bebop_driver/states/enable_mediastreamingstate_videoenablechanged: True
 * /bebop/bebop_driver/states/enable_numberofsatellitechanged: True
 * /bebop/bebop_driver/states/enable_overheatstate_overheatchanged: True
 * /bebop/bebop_driver/states/enable_pilotingstate_altitudechanged: True
 * /bebop/bebop_driver/states/enable_pilotingstate_attitudechanged: True
 * /bebop/bebop_driver/states/enable_pilotingstate_flattrimchanged: True
 * /bebop/bebop_driver/states/enable_pilotingstate_flyingstatechanged: True
 * /bebop/bebop_driver/states/enable_pilotingstate_navigatehomestatechanged: True
 * /bebop/bebop_driver/states/enable_pilotingstate_positionchanged: True
 * /bebop/bebop_driver/states/enable_pilotingstate_speedchanged: True
 * /bebop/robot_description: <?xml version="1....
 * /rosdistro: indigo
 * /rosversion: 1.11.20

NODES
  /bebop/
    bebop_driver (bebop_driver/bebop_driver_node)
    robot_state_publisher (robot_state_publisher/robot_state_publisher)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
process[bebop/bebop_driver-1]: started with pid [15275]
process[bebop/robot_state_publisher-2]: started with pid [15276]
[ INFO] [1481857456.714137182]: Initializing nodelet with 4 worker threads.
[ INFO] [1481857456.752061519]: [BebopSDK] 11:04:16:751 | Bebop:225 - Bebop Cnstr()
[ INFO] [1481857456.752197549]: Nodelet Cstr
[ INFO] [1481857456.761891007]: Connecting to Bebop ...
[ INFO] [1481857456.770876837]: [CB] 11:04:16:770 | Ardrone3PilotingStateFlatTrimChanged:382 - [STATES] Enabling states/ardrone3/PilotingState/FlatTrimChanged
[ INFO] [1481857456.772738835]: [CB] 11:04:16:772 | Ardrone3PilotingStateFlyingStateChanged:426 - [STATES] Enabling states/ardrone3/PilotingState/FlyingStateChanged
[ INFO] [1481857456.774621472]: [CB] 11:04:16:774 | Ardrone3PilotingStateNavigateHomeStateChanged:528 - [STATES] Enabling states/ardrone3/PilotingState/NavigateHomeStateChanged
[ INFO] [1481857456.775928606]: [CB] 11:04:16:775 | Ardrone3PilotingStatePositionChanged:586 - [STATES] Enabling states/ardrone3/PilotingState/PositionChanged
[ INFO] [1481857456.777135670]: [CB] 11:04:16:777 | Ardrone3PilotingStateSpeedChanged:651 - [STATES] Enabling states/ardrone3/PilotingState/SpeedChanged
[ INFO] [1481857456.778312966]: [CB] 11:04:16:778 | Ardrone3PilotingStateAttitudeChanged:716 - [STATES] Enabling states/ardrone3/PilotingState/AttitudeChanged
[ INFO] [1481857456.780080986]: [CB] 11:04:16:780 | Ardrone3PilotingStateAltitudeChanged:832 - [STATES] Enabling states/ardrone3/PilotingState/AltitudeChanged
[ INFO] [1481857456.788196375]: [CB] 11:04:16:788 | Ardrone3MediaStreamingStateVideoEnableChanged:1245 - [STATES] Enabling states/ardrone3/MediaStreamingState/VideoEnableChanged
[ INFO] [1481857456.790389817]: [CB] 11:04:16:790 | Ardrone3CameraStateOrientation:1347 - [STATES] Enabling states/ardrone3/CameraState/Orientation
[ INFO] [1481857456.792943585]: [CB] 11:04:16:792 | Ardrone3GPSStateNumberOfSatelliteChanged:1565 - [STATES] Enabling states/ardrone3/GPSState/NumberOfSatelliteChanged
[ INFO] [1481857456.803285097]: [CB] 11:04:16:803 | CommonCommonStateBatteryStateChanged:146 - [STATES] Enabling states/common/CommonState/BatteryStateChanged
[ INFO] [1481857456.809015864]: [CB] 11:04:16:808 | CommonCommonStateWifiSignalChanged:508 - [STATES] Enabling states/common/CommonState/WifiSignalChanged
[ INFO] [1481857456.821506286]: [CB] 11:04:16:821 | CommonOverHeatStateOverHeatChanged:963 - [STATES] Enabling states/common/OverHeatState/OverHeatChanged
[ INFO] [1481857456.823228374]: [CB] 11:04:16:823 | CommonMavlinkStateMavlinkFilePlayingStateChanged:1058 - [STATES] Enabling states/common/MavlinkState/MavlinkFilePlayingStateChanged
[ INFO] [1481857456.824408761]: [CB] 11:04:16:824 | CommonMavlinkStateMavlinkPlayErrorStateChanged:1123 - [STATES] Enabling states/common/MavlinkState/MavlinkPlayErrorStateChanged
[ INFO] [1481857456.836052919]: [CB] 11:04:16:836 | CommonFlightPlanStateAvailabilityStateChanged:1457 - [STATES] Enabling states/common/FlightPlanState/AvailabilityStateChanged
[ INFO] [1481857456.837566241]: [CB] 11:04:16:837 | CommonFlightPlanStateComponentStateListChanged:1508 - [STATES] Enabling states/common/FlightPlanState/ComponentStateListChanged
[ INFO] [1481857456.852159451]: [ARCONTROLLER_Network] 11:04:16:852 | ARCONTROLLER_Network_GetAvailableSocketPort:108 - d2c_port port: 40252
[ INFO] [1481857456.852344804]: [ARCONTROLLER_Stream2] 11:04:16:852 | ARCONTROLLER_Stream2_Open_Socket:122 - udp local port stream: 49286
[ INFO] [1481857456.852381267]: [ARCONTROLLER_Stream2] 11:04:16:852 | ARCONTROLLER_Stream2_Open_Socket:122 - udp local port control: 38631
[ERROR] [1481857457.855801241]: [ARDISCOVERY_Connection] 11:04:17:855 | ARDISCOVERY_Connection_ControllerInitSocket:703 - connect() SO_ERROR: 111 Connection refused
[ERROR] [1481857457.856189450]: [ARDISCOVERY_Device] 11:04:17:856 | ARDISCOVERY_Device_NewARNetworkAL:241 - error: Socket access permission denied
[ERROR] [1481857457.856306349]: [ARCONTROLLER_Network] 11:04:17:856 | ARCONTROLLER_Network_New:335 - error: Error during the getting of the ARNETWORKAL_Manager from the device
[ INFO] [1481857457.856652852]: [ARCONTROLLER_Device] 11:04:17:856 | ARCONTROLLER_Device_StartRun:4815 - Start failed or canceled.
[ERROR] [1481857457.856795816]: [ARCONTROLLER_Device] 11:04:17:856 | ARCONTROLLER_Device_StopNetwork:5288 - Error occured durring unregistering ARCommands to the feature ARCONTROLLER_DICTIONARY_KEY_ARDRO
[ERROR] [1481857457.856856501]: [ARCONTROLLER_Device] 11:04:17:856 | ARCONTROLLER_Device_StopNetwork:5297 - Error occured durring unregistering ARCommands to the feature ARCONTROLLER_DICTIONARY_KEY_COMMO
[ERROR] [1481857457.856942739]: [ARCONTROLLER_Device] 11:04:17:856 | ARCONTROLLER_Device_StopRun:4858 - StopNetwork failed with error :Bad parameters
[ERROR] [1481857457.857060846]: [ARCONTROLLER_Device] 11:04:17:857 | ARCONTROLLER_Device_StartRun:4823 - Start fail error :Error during the getting of the ARNETWORKAL_Manager from the device
[ INFO] [1481857457.857706200]: [BebopSDK] 11:04:17:857 | Cleanup:326 - Bebop Cleanup()
[FATAL] [1481857457.858296649]: Init failed: Waiting for device failed: No error
[ INFO] [1481857457.858635163]: Bebop Nodelet Dstr: 0
[ INFO] [1481857457.858692114]: Killing Camera Thread ...
[ INFO] [1481857457.858734963]: Killing Aux Thread ...
[bebop/bebop_driver-1] process has died [pid 15275, exit code -11, cmd /home/kenisvery/bebop_ws/devel/lib/bebop_driver/bebop_driver_node __name:=bebop_driver __log:=/home/kenisvery/.ros/log/7514a9ce-c339-11e6-a825-0800275ad713/bebop-bebop_driver-1.log].
log file: /home/kenisvery/.ros/log/7514a9ce-c339-11e6-a825-0800275ad713/bebop-bebop_driver-1*.log

Thank you for your attentions

mani-monaj, 2016-12-19:
Can you `ping` the Bebop?

```bash
$ ping 192.168.42.1
```

kenisvery, 2016-12-20:
Yeah, This is the following result...

SUMMARY
========

PARAMETERS
 * /bebop/bebop_driver/bebop_ip: 192.168.42.1
 * /bebop/bebop_driver/camera_info_url: package://bebop_d...
 * /bebop/bebop_driver/cmd_vel_timeout: 0.2
 * /bebop/bebop_driver/odom_frame_id: odom
 * /bebop/bebop_driver/publish_odom_tf: True
 * /bebop/bebop_driver/reset_settings: True
 * /bebop/bebop_driver/states/enable_altitudechanged: True
 * /bebop/bebop_driver/states/enable_autotakeoffmodechanged: True
 * /bebop/bebop_driver/states/enable_camerastate_orientation: True
 * /bebop/bebop_driver/states/enable_commonstate_batterystatechanged: True
 * /bebop/bebop_driver/states/enable_commonstate_wifisignalchanged: True
 * /bebop/bebop_driver/states/enable_controllerstate_ispilotingchanged: True
 * /bebop/bebop_driver/states/enable_flightplanstate_availabilitystatechanged: True
 * /bebop/bebop_driver/states/enable_flightplanstate_componentstatelistchanged: True
 * /bebop/bebop_driver/states/enable_gpsstate_numberofsatellitechanged: True
 * /bebop/bebop_driver/states/enable_mavlinkstate_mavlinkfileplayingstatechanged: True
 * /bebop/bebop_driver/states/enable_mavlinkstate_mavlinkplayerrorstatechanged: True
 * /bebop/bebop_driver/states/enable_mediastreamingstate_videoenablechanged: True
 * /bebop/bebop_driver/states/enable_numberofsatellitechanged: True
 * /bebop/bebop_driver/states/enable_overheatstate_overheatchanged: True
 * /bebop/bebop_driver/states/enable_pilotingstate_altitudechanged: True
 * /bebop/bebop_driver/states/enable_pilotingstate_attitudechanged: True
 * /bebop/bebop_driver/states/enable_pilotingstate_flattrimchanged: True
 * /bebop/bebop_driver/states/enable_pilotingstate_flyingstatechanged: True
 * /bebop/bebop_driver/states/enable_pilotingstate_navigatehomestatechanged: True
 * /bebop/bebop_driver/states/enable_pilotingstate_positionchanged: True
 * /bebop/bebop_driver/states/enable_pilotingstate_speedchanged: True
 * /bebop/robot_description: <?xml version="1....
 * /rosdistro: indigo
 * /rosversion: 1.11.20

NODES
  /bebop/
    bebop_driver (bebop_driver/bebop_driver_node)
    robot_state_publisher (robot_state_publisher/robot_state_publisher)

auto-starting new master
process[master]: started with pid [5741]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to ce2052d0-c5f4-11e6-9f09-0800275ad713
process[rosout-1]: started with pid [5754]
started core service [/rosout]
process[bebop/bebop_driver-2]: started with pid [5757]
process[bebop/robot_state_publisher-3]: started with pid [5758]
[ INFO] [1482156592.070147107]: Initializing nodelet with 4 worker threads.
[ INFO] [1482156592.814424277]: [BebopSDK] 22:09:52:812 | Bebop:225 - Bebop Cnstr()
[ INFO] [1482156592.814561060]: Nodelet Cstr
[ INFO] [1482156592.820646419]: Connecting to Bebop ...
[ INFO] [1482156592.826915114]: [CB] 22:09:52:826 | Ardrone3PilotingStateFlatTrimChanged:382 - [STATES] Enabling states/ardrone3/PilotingState/FlatTrimChanged
[ INFO] [1482156592.828129280]: [CB] 22:09:52:828 | Ardrone3PilotingStateFlyingStateChanged:426 - [STATES] Enabling states/ardrone3/PilotingState/FlyingStateChanged
[ INFO] [1482156592.829886578]: [CB] 22:09:52:829 | Ardrone3PilotingStateNavigateHomeStateChanged:528 - [STATES] Enabling states/ardrone3/PilotingState/NavigateHomeStateChanged
[ INFO] [1482156592.831006114]: [CB] 22:09:52:830 | Ardrone3PilotingStatePositionChanged:586 - [STATES] Enabling states/ardrone3/PilotingState/PositionChanged
[ INFO] [1482156592.832173434]: [CB] 22:09:52:832 | Ardrone3PilotingStateSpeedChanged:651 - [STATES] Enabling states/ardrone3/PilotingState/SpeedChanged
[ INFO] [1482156592.833466905]: [CB] 22:09:52:833 | Ardrone3PilotingStateAttitudeChanged:716 - [STATES] Enabling states/ardrone3/PilotingState/AttitudeChanged
[ INFO] [1482156592.835065926]: [CB] 22:09:52:835 | Ardrone3PilotingStateAltitudeChanged:832 - [STATES] Enabling states/ardrone3/PilotingState/AltitudeChanged
[ INFO] [1482156592.842632172]: [CB] 22:09:52:842 | Ardrone3MediaStreamingStateVideoEnableChanged:1245 - [STATES] Enabling states/ardrone3/MediaStreamingState/VideoEnableChanged
[ INFO] [1482156592.844621624]: [CB] 22:09:52:844 | Ardrone3CameraStateOrientation:1347 - [STATES] Enabling states/ardrone3/CameraState/Orientation
[ INFO] [1482156592.847331109]: [CB] 22:09:52:847 | Ardrone3GPSStateNumberOfSatelliteChanged:1565 - [STATES] Enabling states/ardrone3/GPSState/NumberOfSatelliteChanged
[ INFO] [1482156592.851038379]: [CB] 22:09:52:850 | CommonCommonStateBatteryStateChanged:146 - [STATES] Enabling states/common/CommonState/BatteryStateChanged
[ INFO] [1482156592.855199614]: [CB] 22:09:52:855 | CommonCommonStateWifiSignalChanged:508 - [STATES] Enabling states/common/CommonState/WifiSignalChanged
[ INFO] [1482156592.861323693]: [CB] 22:09:52:861 | CommonOverHeatStateOverHeatChanged:963 - [STATES] Enabling states/common/OverHeatState/OverHeatChanged
[ INFO] [1482156592.862888585]: [CB] 22:09:52:862 | CommonMavlinkStateMavlinkFilePlayingStateChanged:1058 - [STATES] Enabling states/common/MavlinkState/MavlinkFilePlayingStateChanged
[ INFO] [1482156592.863949848]: [CB] 22:09:52:863 | CommonMavlinkStateMavlinkPlayErrorStateChanged:1123 - [STATES] Enabling states/common/MavlinkState/MavlinkPlayErrorStateChanged
[... 29 lines cut ...]
[ INFO] [1482156598.922327623]: Killing Camera Thread ...
[ INFO] [1482156598.922517939]: Killing Aux Thread ...
[... cut ...]
64 bytes from 192.168.42.1: icmp_seq=7 ttl=63 time=4.09 ms
64 bytes from 192.168.42.1: icmp_seq=8 ttl=63 time=8.29 ms
64 bytes from 192.168.42.1: icmp_seq=9 ttl=63 time=4.11 ms
64 bytes from 192.168.42.1: icmp_seq=10 ttl=63 time=5.62 ms
64 bytes from 192.168.42.1: icmp_seq=11 ttl=63 time=5.23 ms
^F64 bytes from 192.168.42.1: icmp_seq=12 ttl=63 time=4.78 ms
^C
--- 192.168.42.1 ping statistics ---
12 packets transmitted, 12 received, 0% packet loss, time 11022ms
rtt min/avg/max/mdev = 1.795/5.806/12.870/2.591 ms

closed 2016-12-28 by kenisvery

RENO216, 2019-04-01:
@kenisvery Same issue, have you figured out the solution?