AutonomyLab/bebop_autonomy #85: ERROR: cannot launch node of type [bebop_autonomy/bebop_driver]: bebop_autonomy

kenisvery, 2017-01-03:
Trying to run keyboard controller with bebop autonomy but error occurred.

kenisvery@kenisvery-Sensei:~/bebop_ws$ roslaunch bebop_tutorials keyboard_controller.launch
... logging to /home/kenisvery/.ros/log/ee4ea42c-d1c4-11e6-94b8-0800275ad713/roslaunch-kenisvery-Sensei-15852.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://kenisvery-Sensei:33962/

SUMMARY
========

CLEAR PARAMETERS
 * /bebop_driver/

PARAMETERS
 * /bebop_driver/altitude_max: 3000
 * /bebop_driver/altitude_min: 50
 * /bebop_driver/control_vz_max: 200
 * /bebop_driver/control_yaw: 0.7
 * /bebop_driver/euler_angle_max: 0.1
 * /bebop_driver/flight_without_shell: 0
 * /bebop_driver/outdoor: 0
 * /rosdistro: indigo
 * /rosversion: 1.11.20

NODES
  /
    bebop_driver (bebop_autonomy/bebop_driver)
    keyboard_controller (bebop_tutorials/keyboard_controller.py)

auto-starting new master
process[master]: started with pid [15864]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to ee4ea42c-d1c4-11e6-94b8-0800275ad713
process[rosout-1]: started with pid [15877]
started core service [/rosout]
ERROR: cannot launch node of type [bebop_autonomy/bebop_driver]: bebop_autonomy
ROS path [0]=/opt/ros/indigo/share/ros
ROS path [1]=/home/kenisvery/bebop_ws/src
ROS path [2]=/home/kenisvery/catkin_ws/src
ROS path [3]=/opt/ros/indigo/share
ROS path [4]=/opt/ros/indigo/stacks
process[keyboard_controller-3]: started with pid [15889]
Traceback (most recent call last):
  File "/home/kenisvery/bebop_ws/src/bebop_tutorials/src/keyboard_controller.py", line 9, in <module>
    import roslib; roslib.load_manifest('bebop_tutorials')
  File "/opt/ros/indigo/lib/python2.7/dist-packages/roslib/launcher.py", line 62, in load_manifest
    sys.path = _generate_python_path(package_name, _rospack) + sys.path
  File "/opt/ros/indigo/lib/python2.7/dist-packages/roslib/launcher.py", line 98, in _generate_python_path
    packages = get_depends(pkg, rospack) 
  File "/opt/ros/indigo/lib/python2.7/dist-packages/roslib/launcher.py", line 51, in get_depends
    vals = rospack.get_depends(package, implicit=True)
  File "/usr/lib/python2.7/dist-packages/rospkg/rospack.py", line 237, in get_depends
    s.update(self.get_depends(p, implicit))
  File "/usr/lib/python2.7/dist-packages/rospkg/rospack.py", line 231, in get_depends
    names = [p.name for p in self.get_manifest(name).depends]
  File "/usr/lib/python2.7/dist-packages/rospkg/rospack.py", line 164, in get_manifest
    return self._load_manifest(name)
  File "/usr/lib/python2.7/dist-packages/rospkg/rospack.py", line 208, in _load_manifest
    retval = self._manifests[name] = parse_manifest_file(self.get_path(name), self._manifest_name, rospack=self)
  File "/usr/lib/python2.7/dist-packages/rospkg/rospack.py", line 200, in get_path
    raise ResourceNotFound(name, ros_paths=self._ros_paths)
rospkg.common.ResourceNotFound: bebop_autonomy
ROS path [0]=/opt/ros/indigo/share/ros
ROS path [1]=/home/kenisvery/bebop_ws/src
ROS path [2]=/home/kenisvery/catkin_ws/src
ROS path [3]=/opt/ros/indigo/share
ROS path [4]=/opt/ros/indigo/stacks
================================================================================REQUIRED process [keyboard_controller-3] has died!
process has died [pid 15889, exit code 1, cmd /home/kenisvery/bebop_ws/src/bebop_tutorials/src/keyboard_controller.py __name:=keyboard_controller __log:=/home/kenisvery/.ros/log/ee4ea42c-d1c4-11e6-94b8-0800275ad713/keyboard_controller-3.log].
log file: /home/kenisvery/.ros/log/ee4ea42c-d1c4-11e6-94b8-0800275ad713/keyboard_controller-3*.log
Initiating shutdown!
================================================================================
[keyboard_controller-3] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

Package.xml --> 

<package>
<name>bebop_tutorials</name>
<version>1.0.0</version>
<description>
Controllers developed for the "Up and flying with the bebop Drone and ROS" tutorial series
</description>
<maintainer email="kenisvery@gmail.com">kenisvery</maintainer>
<buildtool_depend>catkin</buildtool_depend>
<run_depend>bebop_autonomy</run_depend>
<run_depend>joy</run_depend>
<license>BSD</license>
<url>http://github.com/mikehamer/ardrone_tutorials</url>
</package>

CMakeLists.txt -->

cmake_minimum_required(VERSION 2.8)
project(bebop_tutorials)

find_package(catkin REQUIRED)
catkin_package()

#catkin_python_setup()

install(PROGRAMS src/drone_controller.py  
		 src/drone_status.py  
		 src/drone_video_display.py  
		 src/joystick_controller.py  
		 src/keyboard_controller.py
        DESTINATION ${CATKIN_PACKAGE_BIN_DESTINATION}
)

install(DIRECTORY launch/
  DESTINATION ${CATKIN_PACKAGE_SHARE_DESTINATION}/launch
)

mani-monaj, 2017-01-09:
1. Could you please format the code section of your submitted issue better?
2. Have you `source`ed the `setup.bash` file within Bebop's workspace?

kenisvery, 2017-01-09:
yes, i sourced the setup.bash every time i run the launch file in the bebop_ws.

`cmake_minimum_required(VERSION 2.8)
project(bebop_tutorials)

find_package(catkin REQUIRED)
catkin_package()

#catkin_python_setup()

install(PROGRAMS src/drone_controller.py  
		 src/drone_status.py  
		 src/drone_video_display.py  
		 src/joystick_controller.py  
		 src/keyboard_controller.py
        DESTINATION ${CATKIN_PACKAGE_BIN_DESTINATION}
)

install(DIRECTORY launch/
  DESTINATION ${CATKIN_PACKAGE_SHARE_DESTINATION}/launch
)`

`<package>
<name>bebop_tutorials</name><version>1.0.0</version><description>
		Controllers developed for the "Up and flying with the bebop Drone and ROS" tutorial series
	</description><maintainer email="kenisvery@gmail.com">kenisvery</maintainer><buildtool_depend>catkin</buildtool_depend><run_depend>bebop_autonomy</run_depend><run_depend>joy</run_depend><license>BSD</license><url>http://github.com/mikehamer/ardrone_tutorials</url>
</package>`

keyboard controller.py -->

`#!/usr/bin/env python

# The Keyboard Controller Node for the tutorial "Up and flying with the AR.Drone and ROS | Getting Started"
# https://github.com/mikehamer/ardrone_tutorials

# This controller extends the base DroneVideoDisplay class, adding a keypress handler to enable keyboard control of the drone

# Import the ROS libraries, and load the manifest file which through <depend package=... /> will give us access to the project dependencies
import roslib; roslib.load_manifest('bebop_tutorials')
import rospy

# Load the DroneController class, which handles interactions with the drone, and the DroneVideoDisplay class, which handles video display
from drone_controller import BasicDroneController
from drone_video_display import DroneVideoDisplay

# Finally the GUI libraries
from PySide import QtCore, QtGui

# Here we define the keyboard map for our controller (note that python has no enums, so we use a class)
class KeyMapping(object):
	PitchForward     = QtCore.Qt.Key.Key_E
	PitchBackward    = QtCore.Qt.Key.Key_D
	RollLeft         = QtCore.Qt.Key.Key_S
	RollRight        = QtCore.Qt.Key.Key_F
	YawLeft          = QtCore.Qt.Key.Key_W
	YawRight         = QtCore.Qt.Key.Key_R
	IncreaseAltitude = QtCore.Qt.Key.Key_Q
	DecreaseAltitude = QtCore.Qt.Key.Key_A
	Takeoff          = QtCore.Qt.Key.Key_Y
	Land             = QtCore.Qt.Key.Key_H
	Emergency        = QtCore.Qt.Key.Key_Space

# Our controller definition, note that we extend the DroneVideoDisplay class
class KeyboardController(DroneVideoDisplay):
	def __init__(self):
		super(KeyboardController,self).__init__()
		
		self.pitch = 0
		self.roll = 0
		self.yaw_velocity = 0 
[... 149 lines cut ...]

	def SendLand(self):
		# Send a landing message to the bebop driver
		# Note we send this in all states, landing can do no harm
		self.pubLand.publish(Empty())

	def SendEmergency(self):
		# Send an emergency (or reset) message to the bebop driver
		self.pubReset.publish(Empty())

	def SetCommand(self,roll=0,pitch=0,yaw_velocity=0,z_velocity=0):
		# Called by the main program to set the current command
		self.command.linear.x  = pitch
		self.command.linear.y  = roll
		self.command.linear.z  = z_velocity
		self.command.angular.z = yaw_velocity

	def SendCommand(self,event):
		# The previously set command is then sent out periodically if the drone is flying
		if self.status == DroneStatus.Flying or self.status == DroneStatus.GotoHover or self.status == DroneStatus.Hovering:
			self.pubCommand.publish(self.command)

`

I suspect it is caused by the publisher of the drone controller. It

mani-monaj, 2017-02-08:
It's hard for me to follow what you've pasted. 

1. Please consider [formatting your report](https://help.github.com/articles/about-writing-and-formatting-on-github/) properly
2. Please include only the information that you think might be useful. I quite frankly am overwhelmed with the amount of information provide.

You can also use [Questions - ROS Answers: Open Source Q&A Forum](http://answers.ros.org/questions/) forum to post this type of questions. Please use the tag: `bebop_autonomy`.

thomas-bamford, 2017-06-23:
Hello,
@kenisvery Have you made any progress in this effort? In the terminal output above, the program was crashing because the bebop_autonomy package was not found which could mean the ws was not sourced or not successfully built with catkin. I also noticed that in the dronevideodisplay code, you have imported a non-existing message 'from bebop_autonomy.msg import Navdata.' Please see [http://bebop-autonomy.readthedocs.io/en/latest/reading.html#states-aka-navdata](url) describing that the bebop sdk does not publish to a single topic anymore.

mani-monaj, 2017-06-27:
@kenisvery closing this ticket due to inactivity. Please feel free to re-open it if you have any updates.

closed 2017-06-27 by mani-monaj

sepideh2020, 2018-10-06:
install these packages
sudo apt-get install ros-kinetic-gazebo-ros-pkgs ros-kinetic-gazebo-ros-control
sudo apt-get install ros-kinetic-gazebo-ros-pkgs