AutonomyLab/bebop_autonomy #86: Running Bebop on Virtual Machine

kenisvery, 2017-01-04:
I am trying to run bebop autonomy on a virtual machine but I got this error. The driver installation is successful, and also tried on my other computer which is not VM but ubuntu 14.04.  Is it because of VM?

SUMMARY

PARAMETERS

/bebop/bebop_driver/bebop_ip: 192.168.42.1
/bebop/bebop_driver/camera_info_url: package://bebop_d...
/bebop/bebop_driver/cmd_vel_timeout: 0.2
/bebop/bebop_driver/odom_frame_id: odom
/bebop/bebop_driver/publish_odom_tf: True
/bebop/bebop_driver/reset_settings: True
/bebop/bebop_driver/states/enable_altitudechanged: True
/bebop/bebop_driver/states/enable_autotakeoffmodechanged: True
/bebop/bebop_driver/states/enable_camerastate_orientation: True
/bebop/bebop_driver/states/enable_commonstate_batterystatechanged: True
/bebop/bebop_driver/states/enable_commonstate_wifisignalchanged: True
/bebop/bebop_driver/states/enable_controllerstate_ispilotingchanged: True
/bebop/bebop_driver/states/enable_flightplanstate_availabilitystatechanged: True
/bebop/bebop_driver/states/enable_flightplanstate_componentstatelistchanged: True
/bebop/bebop_driver/states/enable_gpsstate_numberofsatellitechanged: True
/bebop/bebop_driver/states/enable_mavlinkstate_mavlinkfileplayingstatechanged: True
/bebop/bebop_driver/states/enable_mavlinkstate_mavlinkplayerrorstatechanged: True
/bebop/bebop_driver/states/enable_mediastreamingstate_videoenablechanged: True
/bebop/bebop_driver/states/enable_numberofsatellitechanged: True
/bebop/bebop_driver/states/enable_overheatstate_overheatchanged: True
/bebop/bebop_driver/states/enable_pilotingstate_altitudechanged: True
/bebop/bebop_driver/states/enable_pilotingstate_attitudechanged: True
/bebop/bebop_driver/states/enable_pilotingstate_flattrimchanged: True
/bebop/bebop_driver/states/enable_pilotingstate_flyingstatechanged: True
/bebop/bebop_driver/states/enable_pilotingstate_navigatehomestatechanged: True
/bebop/bebop_driver/states/enable_pilotingstate_positionchanged: True
/bebop/bebop_driver/states/enable_pilotingstate_speedchanged: True
/bebop/robot_description: <?xml version="1....
/rosdistro: indigo
/rosversion: 1.11.20
NODES
/bebop/
bebop_driver (bebop_driver/bebop_driver_node)
robot_state_publisher (robot_state_publisher/robot_state_publisher)

auto-starting new master
process[master]: started with pid [5741]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to ce2052d0-c5f4-11e6-9f09-0800275ad713
process[rosout-1]: started with pid [5754]
started core service [/rosout]
process[bebop/bebop_driver-2]: started with pid [5757]
process[bebop/robot_state_publisher-3]: started with pid [5758]
[ INFO] [1482156592.070147107]: Initializing nodelet with 4 worker threads.
[ INFO] [1482156592.814424277]: [BebopSDK] 22:09:52:812 | Bebop:225 - Bebop Cnstr()
[ INFO] [1482156592.814561060]: Nodelet Cstr
[ INFO] [1482156592.820646419]: Connecting to Bebop ...
[ INFO] [1482156592.826915114]: [CB] 22:09:52:826 | Ardrone3PilotingStateFlatTrimChanged:382 - [STATES] Enabling states/ardrone3/PilotingState/FlatTrimChanged
[ INFO] [1482156592.828129280]: [CB] 22:09:52:828 | Ardrone3PilotingStateFlyingStateChanged:426 - [STATES] Enabling states/ardrone3/PilotingState/FlyingStateChanged
[ INFO] [1482156592.829886578]: [CB] 22:09:52:829 | Ardrone3PilotingStateNavigateHomeStateChanged:528 - [STATES] Enabling states/ardrone3/PilotingState/NavigateHomeStateChanged
[ INFO] [1482156592.831006114]: [CB] 22:09:52:830 | Ardrone3PilotingStatePositionChanged:586 - [STATES] Enabling states/ardrone3/PilotingState/PositionChanged
[ INFO] [1482156592.832173434]: [CB] 22:09:52:832 | Ardrone3PilotingStateSpeedChanged:651 - [STATES] Enabling states/ardrone3/PilotingState/SpeedChanged
[ INFO] [1482156592.833466905]: [CB] 22:09:52:833 | Ardrone3PilotingStateAttitudeChanged:716 - [STATES] Enabling states/ardrone3/PilotingState/AttitudeChanged
[ INFO] [1482156592.835065926]: [CB] 22:09:52:835 | Ardrone3PilotingStateAltitudeChanged:832 - [STATES] Enabling states/ardrone3/PilotingState/AltitudeChanged
[ INFO] [1482156592.842632172]: [CB] 22:09:52:842 | Ardrone3MediaStreamingStateVideoEnableChanged:1245 - [STATES] Enabling states/ardrone3/MediaStreamingState/VideoEnableChanged
[ INFO] [1482156592.844621624]: [CB] 22:09:52:844 | Ardrone3CameraStateOrientation:1347 - [STATES] Enabling states/ardrone3/CameraState/Orientation
[ INFO] [1482156592.847331109]: [CB] 22:09:52:847 | Ardrone3GPSStateNumberOfSatelliteChanged:1565 - [STATES] Enabling states/ardrone3/GPSState/NumberOfSatelliteChanged
[ INFO] [1482156592.851038379]: [CB] 22:09:52:850 | CommonCommonStateBatteryStateChanged:146 - [STATES] Enabling states/common/CommonState/BatteryStateChanged
[ INFO] [1482156592.855199614]: [CB] 22:09:52:855 | CommonCommonStateWifiSignalChanged:508 - [STATES] Enabling states/common/CommonState/WifiSignalChanged
[ INFO] [1482156592.861323693]: [CB] 22:09:52:861 | CommonOverHeatStateOverHeatChanged:963 - [STATES] Enabling states/common/OverHeatState/OverHeatChanged
[ INFO] [1482156592.862888585]: [CB] 22:09:52:862 | CommonMavlinkStateMavlinkFilePlayingStateChanged:1058 - [STATES] Enabling states/common/MavlinkState/MavlinkFilePlayingStateChanged
[ INFO] [1482156592.863949848]: [CB] 22:09:52:863 | CommonMavlinkStateMavlinkPlayErrorStateChanged:1123 - [STATES] Enabling states/common/MavlinkState/MavlinkPlayErrorStateChanged
[ INFO] [1482156592.867003198]: [CB] 22:09:52:866 | CommonFlightPlanStateAvailabilityStateChanged:1457 - [STATES] Enabling states/common/FlightPlanState/AvailabilityStateChanged
[... 7 lines cut ...]
[ERROR] [1482156593.988943621]: [ARNETWORK_Sender] 22:09:53:988 | ARNETWORK_Sender_ProcessBufferToSend:417 - [0x7fa7702a1b60] Will retry sending data of buffer 11
[ERROR] [1482156594.507771560]: [ARNETWORK_Sender] 22:09:54:507 | ARNETWORK_Sender_ProcessBufferToSend:401 - [0x7fa7702a1b60] Timeout waiting for ack in buffer 11
[ERROR] [1482156594.507988195]: [ARNETWORK_Sender] 22:09:54:507 | ARNETWORK_Sender_ProcessBufferToSend:417 - [0x7fa7702a1b60] Will retry sending data of buffer 11
[ERROR] [1482156595.024421786]: [ARNETWORK_Sender] 22:09:55:024 | ARNETWORK_Sender_ProcessBufferToSend:401 - [0x7fa7702a1b60] Timeout waiting for ack in buffer 11
[ WARN] [1482156595.024829808]: [ARNETWORK_Sender] 22:09:55:024 | ARNETWORK_Sender_TimeOutCallback:651 - [0x7fa7702a1b60] Did timeout sending command : project = 0 | class = 4 | command = 1 - 0
[ERROR] [1482156595.541742172]: [ARNETWORK_Sender] 22:09:55:541 | ARNETWORK_Sender_ProcessBufferToSend:401 - [0x7fa7702a1b60] Timeout waiting for ack in buffer 11
[ERROR] [1482156595.542211651]: [ARNETWORK_Sender] 22:09:55:542 | ARNETWORK_Sender_ProcessBufferToSend:417 - [0x7fa7702a1b60] Will retry sending data of buffer 11
[ERROR] [1482156596.059634525]: [ARNETWORK_Sender] 22:09:56:059 | ARNETWORK_Sender_ProcessBufferToSend:401 - [0x7fa7702a1b60] Timeout waiting for ack in buffer 11
[ERROR] [1482156596.059987334]: [ARNETWORK_Sender] 22:09:56:059 | ARNETWORK_Sender_ProcessBufferToSend:417 - [0x7fa7702a1b60] Will retry sending data of buffer 11
[ERROR] [1482156596.577537789]: [ARNETWORK_Sender] 22:09:56:577 | ARNETWORK_Sender_ProcessBufferToSend:401 - [0x7fa7702a1b60] Timeout waiting for ack in buffer 11
[ERROR] [1482156596.577996018]: [ARNETWORK_Sender] 22:09:56:577 | ARNETWORK_Sender_ProcessBufferToSend:417 - [0x7fa7702a1b60] Will retry sending data of buffer 11
[ERROR] [1482156597.094186663]: [ARNETWORK_Sender] 22:09:57:094 | ARNETWORK_Sender_ProcessBufferToSend:401 - [0x7fa7702a1b60] Timeout waiting for ack in buffer 11
[ WARN] [1482156597.094536686]: [ARNETWORK_Sender] 22:09:57:094 | ARNETWORK_Sender_TimeOutCallback:651 - [0x7fa7702a1b60] Did timeout sending command : project = 0 | class = 4 | command = 2 - 0
[ERROR] [1482156597.609501995]: [ARNETWORK_Sender] 22:09:57:609 | ARNETWORK_Sender_ProcessBufferToSend:401 - [0x7fa7702a1b60] Timeout waiting for ack in buffer 11
[ERROR] [1482156597.609623550]: [ARNETWORK_Sender] 22:09:57:609 | ARNETWORK_Sender_ProcessBufferToSend:417 - [0x7fa7702a1b60] Will retry sending data of buffer 11
[ INFO] [1482156597.920629124]: [ARNETWORKAL_WifiNetwork] 22:09:57:920 | ARNETWORKAL_WifiNetwork_Receive:890 - [0x7fa77024dea0] connection lost (too long time without reception)
[ INFO] [1482156597.921158592]: [ARCONTROLLER_Device] 22:09:57:921 | ARCONTROLLER_Device_StartRun:4815 - Start failed or canceled.
[ERROR] [1482156598.916221411]: [ARCONTROLLER_Device] 22:09:58:916 | ARCONTROLLER_Device_StartRun:4823 - Start fail error :start canceled
[... cut ...]
[ INFO] [1482156598.922101411]: Bebop Nodelet Dstr: 0
[ INFO] [1482156598.922327623]: Killing Camera Thread ...
[ INFO] [1482156598.922517939]: Killing Aux Thread ...
[bebop/bebop_driver-2] process has died [pid 5757, exit code -11, cmd /home/kenisvery/bebop_ws/devel/lib/bebop_driver/bebop_driver_node __name:=bebop_driver __log:=/home/kenisvery/.ros/log/ce2052d0-c5f4-11e6-9f09-0800275ad713/bebop-bebop_driver-2.log].
log file: /home/kenisvery/.ros/log/ce2052d0-c5f4-11e6-9f09-0800275ad713/bebop-bebop_driver-2*.log

thomas-bamford, 2017-01-05:
Hi kenisvery,
To run bebop_autonomy on a VM you have to make sure that the VM has a network adapter that is bridged (default is NAT) and you are connected to the Bebop drone. The reason for this is presumably because the driver needs to connect directly to the drone's physical network. I run bebop_autonomy on VMware with a bridged network adapter and I have had no problems since setup.

Hope that helps,
Thomas

kenisvery, 2017-01-07:
Thanks!!!  It really works!

closed 2017-01-07 by kenisvery

ziranwu, 2017-11-21:
Got exactly the same error messages when trying to launch bebop_autonomy as a node on my Bebop 2. Ubuntu 16, direct WiFi connection to the drone's WiFi network using a USB wireless adapter on my computer, no VM. IP info showed the correct DHCP server 192.168.42.1. Please advise!

miller15, 2018-02-07:
Also having the same problem. But our network adapter is bridged by default since we are running ROS through docker, so no problems there. We get the same error printout as OP. We can ping the server at the IP address fine. Any other suggestions?