AutonomyLab/bebop_autonomy #87: Libraries/ commands for return home.

kenisvery, 2017-01-07:
Hi, 

Are there any msg and command that tell bebop to return it takeoff position with bebop autonomy (std_msgs, geometry_msgs.msgs)? If so, how could I publish it in python?

P.S.
Is it possible to use bebop autonomy with OpenCV for image processing/ optical flow?

mani-monaj, 2017-01-09:
> Are there any msg and command that tell bebop to return it takeoff position with bebop autonomy (std_msgs, geometry_msgs.msgs)?

Yes. But it GPS-based. Please check the following: http://bebop-autonomy.readthedocs.io/en/latest/piloting.html#navigate-home

> If so, how could I publish it in python?

Please follow the [ROS Python Tutorials](http://wiki.ros.org/rospy/Tutorials).

> Is it possible to use bebop autonomy with OpenCV for image processing/ optical flow?

Yes! You can subscribe to Bebop's video stream in any other ROS node and use OpenCV to process the stream (e.g. to calculate the Optical Flow). To convert from ROS Image format (`sensor_msgs::Image`) to OpenCV format, you can use [cv_bridge](http://wiki.ros.org/cv_bridge).

kenisvery, 2017-01-09:
WOW! bebop autonomy is very powerful!!! 

By the way, I'm trying to use a wifi router to connect multi bebop 2 drone, just like how tum ar drone does. Is it practically possible to achieve? 

plus, even if I connected to multiple bebop 2 drones,  how could I give a separate command to each drone? Do the drones do the exact same thing each time I command it with rostopic or compile a python program?

mani-monaj, 2017-02-08:
> By the way, I'm trying to use a wifi router to connect multi bebop 2 drone, just like how tum ar drone does. Is it practically possible to achieve?

I have not done it yet. But there has been some successful attempts. Please check #15 

> plus, even if I connected to multiple bebop 2 drones, how could I give a separate command to each drone?

You need to run each instance of the driver under its own namespace. This seems to be a good tutorial to familiarize yourself with the concept: [ROS Naming and Namespaces | NooTriX](http://nootrix.com/software/ros-namespaces/)

All `bebop_driver/launch/*` and `bebop_tools/launch/*` launch files accept an argument for setting the namespace. This is by default set to `bebop` but can be changed to any other name:

```bash
$ roslaunch bebop_tools bebop_nodelet_iv.launch namespace:=customnamespace 
```

donald2016, 2017-03-01:
>     Is it possible to use bebop autonomy with OpenCV for image processing/ optical flow?

> Yes! You can subscribe to Bebop's video stream in any other ROS node and use OpenCV to process the stream (e.g. to calculate the Optical Flow). To convert from ROS Image format (sensor_msgs::Image) to OpenCV format, you can use cv_bridge.

I have code my bebop to read the image from the front camera. According to your reply, calculating the Optical Flow means you use the front camera to find optical flow OR using the bottom camera and ultra-sonic sensor for Optical flow calculation?

Can I subscribe to the image from the bottom camera?

closed 2017-03-13 by mani-monaj