AutonomyLab/bebop_autonomy #88: Flying bebop in a circle autonomously.

kenisvery, 2017-01-12:
Hi,

I wrote a program to fly my parrot bebop 2 drone to fly in a circle to fly autonomously. It works well with the tum simulator. The ar drone turn a complete circle, but my bebop 2 drone did not do a perfect circle.

Link for video -> https://drive.google.com/file/d/0B6ZU9qP_1Y0zWGZ6OXVVNXdDczQ/view?usp=sharing 

Below links are my coding -> https://drive.google.com/file/d/0B6ZU9qP_1Y0zSks5eXJNZ3dfWGc/view?usp=sharing

thomas-bamford, 2017-01-12:
Hello,
This is not an issue with either the bebop driver or the drone itself. The reason it doesn't complete the circle is because the real world is not ideal. To get the drone to fly in a perfect circle you will need to write a controller that follows a trajectory through time and uses the drone's state (perhaps generated by robot_localization) to determine commands. This is a standard approach to the problem and there are many new approaches that may perform better through the use of machine learning techniques (see https://m.youtube.com/user/angelaschoe). The program you have written to send commands with no feedback can be very dangerous if your drone is disturbed during flight (i.e. Wind or connection dropout) - please only use this code in a controlled environment or simulation.

mani-monaj, 2017-02-08:
Thanks @thomas-bamford for your complete answer. You brought up fair and reasonable points.

closed 2017-02-08 by mani-monaj