AutonomyLab/bebop_autonomy #91: problem to run  bebop_driver_node

hosh0425, 2017-02-02:
when I run this command :
rosrun bebop_driver bebop_driver_node this is my output:
[ INFO] [1486043366.436598051]: Nodelet Cstr
[ INFO] [1486043366.447257915]: Connecting to Bebop ...
[ INFO] [1486043366.568235933]: [ARCONTROLLER_Network] 17:19:26:568 | ARCONTROLLER_Network_GetAvailableSocketPort:108 - d2c_port port: 55924
[ INFO] [1486043366.568500692]: [ARCONTROLLER_Stream2] 17:19:26:568 | ARCONTROLLER_Stream2_Open_Socket:122 - udp local port stream: 60982
[ INFO] [1486043366.568540941]: [ARCONTROLLER_Stream2] 17:19:26:568 | ARCONTROLLER_Stream2_Open_Socket:122 - udp local port control: 48044
[ INFO] [1486043366.715122632]: [BebopSDK] 17:19:26:715 | CommandReceivedCallback:113 - Command Received Callback LWP id is: 15683
[ INFO] [1486043366.851146087]: [BebopSDK] 17:19:26:851 | Connect:321 - BebopSDK inited, lwp_id: 15572
[ INFO] [1486043366.851210014]: Fetching all settings from the Drone ...
[ INFO] [1486043366.891831828]: Value for GPSSettingsHomeTypeType recved: 0
[ INFO] [1486043366.891906639]: [CB] 17:19:26:891 | Update:1859 - Checking if GPSSettingsHomeTypeType exists in params ...
[ INFO] [1486043366.892973808]: [CB] 17:19:26:892 | Update:1862 -   No
[ INFO] [1486043366.898245021]: Value for PictureSettingsVideoRecordingModeMode recved: 0
[ INFO] [1486043366.898378093]: [CB] 17:19:26:898 | Update:1619 - Checking if PictureSettingsVideoRecordingModeMode exists in params ...
[ INFO] [1486043366.899520028]: [CB] 17:19:26:899 | Update:1622 -   No
[ INFO] [1486043366.901585241]: Value for PictureSettingsVideoFramerateFramerate recved: 1
[ INFO] [1486043366.901631087]: [CB] 17:19:26:901 | Update:1699 - Checking if PictureSettingsVideoFramerateFramerate exists in params ...
[ INFO] [1486043366.903337834]: [CB] 17:19:26:903 | Update:1702 -   No
[ INFO] [1486043366.906059278]: Value for PictureSettingsVideoResolutionsType recved: 0
[ INFO] [1486043366.906107848]: [CB] 17:19:26:906 | Update:1779 - Checking if PictureSettingsVideoResolutionsType exists in params ...
[ INFO] [1486043366.907019928]: [CB] 17:19:26:906 | Update:1782 -   No
[ INFO] [1486043366.955435689]: Value for PilotingSettingsMaxAltitudeCurrent recved: 60.5351
[ INFO] [1486043366.955508211]: [CB] 17:19:26:955 | Update:111 - Checking if PilotingSettingsMaxAltitudeCurrent exists in params ...
[ INFO] [1486043366.956549601]: [CB] 17:19:26:956 | Update:114 -   No
[ INFO] [1486043366.957712612]: Value for PilotingSettingsMaxDistanceValue recved: 300
[ INFO] [1486043366.957764000]: [CB] 17:19:26:957 | Update:351 - Checking if PilotingSettingsMaxDistanceValue exists in params ...
[ INFO] [1486043366.958648367]: [CB] 17:19:26:958 | Update:354 -   No
[ INFO] [1486043366.959755208]: Value for PilotingSettingsNoFlyOverMaxDistanceShouldnotflyover recved: 0
[ INFO] [1486043366.959803189]: [CB] 17:19:26:959 | Update:431 - Checking if PilotingSettingsNoFlyOverMaxDistanceShouldnotflyover exists in params ...
[ INFO] [1486043366.960523897]: [CB] 17:19:26:960 | Update:434 -   No
[ INFO] [1486043366.965885029]: Value for PilotingSettingsMaxTiltCurrent recved: 28
[ INFO] [1486043366.965940749]: [CB] 17:19:26:965 | Update:191 - Checking if PilotingSettingsMaxTiltCurrent exists in params ...
[ INFO] [1486043366.966943998]: [CB] 17:19:26:966 | Update:194 -   No
[ INFO] [1486043366.968499138]: Value for PilotingSettingsAbsolutControlOn recved: 0
[ INFO] [1486043366.968555472]: [CB] 17:19:26:968 | Update:271 - Checking if PilotingSettingsAbsolutControlOn exists in params ...
[ INFO] [1486043366.969370373]: [CB] 17:19:26:969 | Update:274 -   No
[ INFO] [1486043366.970197004]: Value for PilotingSettingsBankedTurnValue recved: 1
[ INFO] [1486043366.970235733]: [CB] 17:19:26:970 | Update:511 - Checking if PilotingSettingsBankedTurnValue exists in params ...
[ INFO] [1486043366.970956358]: [CB] 17:19:26:970 | Update:514 -   No
[ INFO] [1486043366.971968341]: Value for SpeedSettingsMaxVerticalSpeedCurrent recved: 2
[ INFO] [1486043366.972011439]: [CB] 17:19:26:971 | Update:991 - Checking if SpeedSettingsMaxVerticalSpeedCurrent exists in params ...
[ INFO] [1486043366.972689086]: [CB] 17:19:26:972 | Update:994 -   No
[ INFO] [1486043366.973505060]: Value for SpeedSettingsMaxRotationSpeedCurrent recved: 54
[ INFO] [1486043366.973545687]: [CB] 17:19:26:973 | Update:1071 - Checking if SpeedSettingsMaxRotationSpeedCurrent exists in params ...
[ INFO] [1486043366.974196072]: [CB] 17:19:26:974 | Update:1074 -   No
[ INFO] [1486043366.975161162]: Value for SpeedSettingsHullProtectionPresent recved: 0
[ INFO] [1486043366.975203257]: [CB] 17:19:26:975 | Update:1151 - Checking if SpeedSettingsHullProtectionPresent exists in params ...
[ INFO] [1486043366.975888092]: [CB] 17:19:26:975 | Update:1154 -   No
[ INFO] [1486043366.976672341]: Value for SpeedSettingsOutdoorOutdoor recved: 1
[ INFO] [1486043366.976708414]: [CB] 17:19:26:976 | Update:1231 - Checking if SpeedSettingsOutdoorOutdoor exists in params ...
[ INFO] [1486043366.977345937]: [CB] 17:19:26:977 | Update:1234 -   No
[ INFO] [1486043366.978263132]: Value for SpeedSettingsMaxPitchRollRotationSpeedCurrent recved: 187
[ INFO] [1486043366.978370435]: [CB] 17:19:26:978 | Update:1311 - Checking if SpeedSettingsMaxPitchRollRotationSpeedCurrent exists in params ...
[ INFO] [1486043366.979861968]: [CB] 17:19:26:979 | Update:1314 -   No
[ INFO] [1486043366.980887680]: Value for NetworkSettingsWifiSelectionType recved: 1
[ INFO] [1486043366.980936364]: [CB] 17:19:26:980 | Update:1417 - Checking if NetworkSettingsWifiSelectionType exists in params ...
[ INFO] [1486043366.982645504]: [CB] 17:19:26:982 | Update:1420 -   No
[ INFO] [1486043366.983951034]: Value for NetworkSettingsWifiSelectionBand recved: 0
[ INFO] [1486043366.984009250]: [CB] 17:19:26:983 | Update:1438 - Checking if NetworkSettingsWifiSelectionBand exists in params ...
[ INFO] [1486043366.984882738]: [CB] 17:19:26:984 | Update:1441 -   No
[ INFO] [1486043366.985812074]: Value for NetworkSettingsWifiSelectionChannel recved: 6
[ INFO] [1486043366.985847009]: [CB] 17:19:26:985 | Update:1459 - Checking if NetworkSettingsWifiSelectionChannel exists in params ...
[ INFO] [1486043366.986588606]: [CB] 17:19:26:986 | Update:1462 -   No
[ INFO] [1486043367.067162908]: Value for PictureSettingsVideoStabilizationModeMode recved: 0
[ INFO] [1486043367.067338898]: [CB] 17:19:27:067 | Update:1539 - Checking if PictureSettingsVideoStabilizationModeMode exists in params ...
[ INFO] [1486043367.069767451]: [CB] 17:19:27:069 | Update:1542 -   No
[ INFO] [1486043367.070956315]: Value for PictureSettingsVideoRecordingModeMode recved: 0
[ INFO] [1486043367.070997331]: [CB] 17:19:27:070 | Update:1619 - Checking if PictureSettingsVideoRecordingModeMode exists in params ...
[ INFO] [1486043367.071633888]: [CB] 17:19:27:071 | Update:1627 -   Yes
[... 48 lines cut ...]
[ WARN] [1486043371.541696001]: [ARSTREAM2_H264Filter] 17:19:31:541 | ARSTREAM2_H264Filter_sync:550 - SPS/PPS sync OK
[ INFO] [1486043371.541831209]: [BebopSDK] 17:19:31:541 | DecoderConfigCallback:147 - H264 configuration packet received: #SPS: 27 #PPS: 9 (MP4? 0)
[ WARN] [1486043371.542048365]: [ARSTREAM2_H264Filter] 17:19:31:542 | ARSTREAM2_H264Filter_generateGrayIFrame:1254 - Waiting for a slice to generate gray I-frame
[ WARN] [1486043371.542111148]: [ARSTREAM2_H264Filter] 17:19:31:542 | ARSTREAM2_H264Filter_generateGrayIFrame:1254 - Waiting for a slice to generate gray I-frame
[ WARN] [1486043371.542202244]: [ARSTREAM2_H264Filter] 17:19:31:542 | ARSTREAM2_H264Filter_generateGrayIFrame:1322 - Gray I slice NALU output size: 1629
[ INFO] [1486043371.542254011]: [BebopSDK] 17:19:31:542 | FrameReceivedCallback:174 - Frame Recv & Decode LWP id: 15949
[ INFO] [1486043371.590554659]: [Decoder] 17:19:31:590 | InitCodec:117 - H264 Codec is partially initialized!
[ INFO] [1486043371.590641743]: [Decoder] 17:19:31:590 | Decode:263 - Updating H264 codec parameters (Buffer Size: 36) ...
[ERROR] [1486043371.592848698]: [Decoder] 17:19:31:592 | Decode:302 - Frame size changed to 856 x 480
[ INFO] [1486043371.592927675]: [Decoder] 17:19:31:592 | ReallocateBuffers:123 - Buffer reallocation request
[ INFO] [1486043371.621575436]: using default calibration URL
[ INFO] [1486043371.621686768]: camera calibration URL: file:///home/hosh/.ros/camera_info/bebop_front.yaml
[ INFO] [1486043371.621822064]: Unable to open camera calibration file [/home/hosh/.ros/camera_info/bebop_front.yaml]
[... cut ...]
[ WARN] [1486043394.325605317]: [ARSTREAM2_H264Filter] 17:19:54:325 | ARSTREAM2_H264Filter_fillMissingSlices:1621 - #570 AUTS:165771222 previousSliceFirstMb:702 previousSliceMbCount:54 currentSliceFirstMb
[ WARN] [1486043394.325810964]: [ARSTREAM2_H264Filter] 17:19:54:325 | ARSTREAM2_H264Filter_fillMissingSlices:1660 - #570 AUTS:165771222 Missing NALU is probably a slice
[ WARN] [1486043394.326062489]: [ARSTREAM2_H264Filter] 17:19:54:325 | ARSTREAM2_H264Filter_fillMissingSlices:1735 - #570 AUTS:165771222 Skipped P slice NALU output size: 13

I test this package on ubuntu 14.04 + indigo and 16.04+kinetic

mani-monaj, 2017-02-08:
@hosh0425 Thank you for your report. What is the problem exactly? Are you getting those *warnings* continuously? Can you see the video stream? 

Instead of using `rosrun` can you do `roslaunch bebop_tools bebop_nodelet_iv.launch`?

hoodsr, 2017-02-14:
I'm also getting these warnings continuously and am running it through `roslaunch bebop_tools bebop_nodelet_iv.launch`. I can see the video stream but it glitches occasionally. It doesn't interfere with using the drone other than the buggy video feed.

mani-monaj, 2017-02-14:
@hoodsr What version of the firmware are you running on Bebop?

hoodsr, 2017-02-14:
I was on 3.0.2 but upgraded to the latest release, 4.0.3. I have the same issue on both versions of the firmware.

mani-monaj, 2017-02-14:
I've released a new version of `parrot_arsdk` based on SDK 3.11.0. It takes some time for that to be available via `apt-get` and I am not sure if using that (and the updated driver) will actually fix this issue. Nevertheless, if you would like to try that:

1. Create a new `catkin` workspace.
2. Clone https://github.com/AutonomyLab/parrot_arsdk into the `src` folder.
3. Clone branch `sdk-3-11` branch of the driver 
(https://github.com/AutonomyLab/bebop_autonomy/tree/sdk-3-11) into the `src` folder.
4. Perform the `rosdep` step as [mentioned in the documentation](http://bebop-autonomy.readthedocs.io/en/latest/installation.html#compiling-from-source).
5. Recompile and run as usual. It should take a few minutes longer since it compiles the SDK as well.

Please let me know if it solves any of your issues. I will hopefully have some time this week to look into this.

dalas10, 2017-03-13:
I'm also getting these warnings continuously and am running it through roslaunch bebop_tools bebop_nodelet_iv.launch. I can't see the video stream. I'm using Ubuntu 14.04(through virtual box) and Ros Indigo.
When i run 
1)source ~/bebop_ws/devel/setup.bash
2)roslaunch bebop_tools bebop_nodelet_iv.launch
i am getting the following output (i get a fatal error 
[FATAL] [1489164201.565475177]: Init failed: Waiting for device failed: No error);

dalas@dalas-VirtualBox:~$ roslaunch bebop_tools bebop_nodelet_iv.launch
... logging to /home/dalas/.ros/log/a430438a-05b0-11e7-8ef4-0800271017ca/roslaunch-dalas-VirtualBox-2072.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://dalas-VirtualBox:44708/

SUMMARY
========

PARAMETERS
 * /bebop/bebop_nodelet/bebop_ip: 192.168.42.1
 * /bebop/bebop_nodelet/camera_info_url: package://bebop_d...
 * /bebop/bebop_nodelet/cmd_vel_timeout: 0.2
 * /bebop/bebop_nodelet/odom_frame_id: odom
 * /bebop/bebop_nodelet/publish_odom_tf: True
 * /bebop/bebop_nodelet/reset_settings: True
 * /bebop/bebop_nodelet/states/enable_altitudechanged: True
 * /bebop/bebop_nodelet/states/enable_autotakeoffmodechanged: True
 * /bebop/bebop_nodelet/states/enable_camerastate_orientation: True
 * /bebop/bebop_nodelet/states/enable_commonstate_batterystatechanged: True
 * /bebop/bebop_nodelet/states/enable_commonstate_wifisignalchanged: True
 * /bebop/bebop_nodelet/states/enable_controllerstate_ispilotingchanged: True
 * /bebop/bebop_nodelet/states/enable_flightplanstate_availabilitystatechanged: True
 * /bebop/bebop_nodelet/states/enable_flightplanstate_componentstatelistchanged: True
 * /bebop/bebop_nodelet/states/enable_gpsstate_numberofsatellitechanged: True
 * /bebop/bebop_nodelet/states/enable_mavlinkstate_mavlinkfileplayingstatechanged: True
 * /bebop/bebop_nodelet/states/enable_mavlinkstate_mavlinkplayerrorstatechanged: True
 * /bebop/bebop_nodelet/states/enable_mediastreamingstate_videoenablechanged: True
 * /bebop/bebop_nodelet/states/enable_numberofsatellitechanged: True
 * /bebop/bebop_nodelet/states/enable_overheatstate_overheatchanged: True
 * /bebop/bebop_nodelet/states/enable_pilotingstate_altitudechanged: True
 * /bebop/bebop_nodelet/states/enable_pilotingstate_attitudechanged: True
 * /bebop/bebop_nodelet/states/enable_pilotingstate_flattrimchanged: True
 * /bebop/bebop_nodelet/states/enable_pilotingstate_flyingstatechanged: True
 * /bebop/bebop_nodelet/states/enable_pilotingstate_navigatehomestatechanged: True
 * /bebop/bebop_nodelet/states/enable_pilotingstate_positionchanged: True
 * /bebop/bebop_nodelet/states/enable_pilotingstate_speedchanged: True
 * /bebop/robot_description: <?xml version="1....
 * /rosdistro: indigo
 * /rosversion: 1.11.20

NODES
  /bebop/
    bebop_image_view_nodelet (nodelet/nodelet)
    bebop_nodelet (nodelet/nodelet)
    bebop_nodelet_manager (nodelet/nodelet)
    robot_state_publisher (robot_state_publisher/robot_state_publisher)

auto-starting new master
process[master]: started with pid [2088]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to a430438a-05b0-11e7-8ef4-0800271017ca
process[rosout-1]: started with pid [2101]
started core service [/rosout]
process[bebop/bebop_nodelet_manager-2]: started with pid [2104]
process[bebop/bebop_nodelet-3]: started with pid [2105]
process[bebop/robot_state_publisher-4]: started with pid [2106]
process[bebop/bebop_image_view_nodelet-5]: started with pid [2107]
[ INFO] [1489164191.572465204]: Initializing nodelet with 1 worker threads.
[ INFO] [1489164193.718027960]: Using transport "raw"
[... 31 lines cut ...]
[ERROR] [1489164198.127211855]: [ARNETWORK_Sender] 18:43:18:127 | ARNETWORK_Sender_ProcessBufferToSend:401 - [0x7fe0842a82b0] Timeout waiting for ack in buffer 11
[ERROR] [1489164198.127306156]: [ARNETWORK_Sender] 18:43:18:127 | ARNETWORK_Sender_ProcessBufferToSend:417 - [0x7fe0842a82b0] Will retry sending data of buffer 11
[ERROR] [1489164198.635223575]: [ARNETWORK_Sender] 18:43:18:635 | ARNETWORK_Sender_ProcessBufferToSend:401 - [0x7fe0842a82b0] Timeout waiting for ack in buffer 11
[ERROR] [1489164198.635321752]: [ARNETWORK_Sender] 18:43:18:635 | ARNETWORK_Sender_ProcessBufferToSend:417 - [0x7fe0842a82b0] Will retry sending data of buffer 11
[ERROR] [1489164199.142855765]: [ARNETWORK_Sender] 18:43:19:142 | ARNETWORK_Sender_ProcessBufferToSend:401 - [0x7fe0842a82b0] Timeout waiting for ack in buffer 11
[ERROR] [1489164199.142950550]: [ARNETWORK_Sender] 18:43:19:142 | ARNETWORK_Sender_ProcessBufferToSend:417 - [0x7fe0842a82b0] Will retry sending data of buffer 11
[ERROR] [1489164199.650693765]: [ARNETWORK_Sender] 18:43:19:650 | ARNETWORK_Sender_ProcessBufferToSend:401 - [0x7fe0842a82b0] Timeout waiting for ack in buffer 11
[ WARN] [1489164199.650821349]: [ARNETWORK_Sender] 18:43:19:650 | ARNETWORK_Sender_TimeOutCallback:651 - [0x7fe0842a82b0] Did timeout sending command : project = 0 | class = 4 | command = 2 - 0
[ERROR] [1489164200.159466556]: [ARNETWORK_Sender] 18:43:20:159 | ARNETWORK_Sender_ProcessBufferToSend:401 - [0x7fe0842a82b0] Timeout waiting for ack in buffer 11
[ERROR] [1489164200.159564203]: [ARNETWORK_Sender] 18:43:20:159 | ARNETWORK_Sender_ProcessBufferToSend:417 - [0x7fe0842a82b0] Will retry sending data of buffer 11
[... cut ...]
[bebop/bebop_nodelet-3] process has died [pid 2105, exit code 255, cmd /opt/ros/indigo/lib/nodelet/nodelet load bebop_driver/BebopDriverNodelet bebop_nodelet_manager __name:=bebop_nodelet __log:=/home/dalas/.ros/log/a430438a-05b0-11e7-8ef4-0800271017ca/bebop-bebop_nodelet-3.log].
log file: /home/dalas/.ros/log/a430438a-05b0-11e7-8ef4-0800271017ca/bebop-bebop_nodelet-3*.log
[bebop/bebop_image_view_nodelet-5] process has finished cleanly
log file: /home/dalas/.ros/log/a430438a-05b0-11e7-8ef4-0800271017ca/bebop-bebop_image_view_nodelet-5*.log

mani-monaj, 2017-03-13:
@dalas10 Can you run the driver in a non-VM environment? I have not tried it myself, but people have already reported issues when running the driver using a VM.

dalas10, 2017-03-13:
Ok thank you..i am gonna try it!!

guozhenglong, 2017-03-22:
if you run the driver in VirtualMachine, you need to set the network model to "bridge connet".