AutonomyLab/bebop_autonomy #92: Camera control overriding autoflight

thomas-bamford, 2017-02-06:
Hello,

I have been writing my own flight plans and loading them into the Bebop 2 for autonomous flight outdoors. When I create a flight plan on the Parrot FlightPlan Pro app that includes camera control commands (such as tilting -83 degrees) and then try to start it in the ros environment, the camera control in the flight plan is overridden. Instead of following the command to tilt to -83, the camera remains at 0 with some sporadic rotating down (up to the end of the command timeframe), and then steadily remains at 0.

Is this potentially caused by the camera control topic in the ros driver?

When I check rostopic list, the topic /bebop/camera_control does not exist.

Thanks,
Thomas

thomas-bamford, 2017-02-08:
Tested commenting out ros subscriber node in bebop.cpp and recompiling but same behaviour was observed. Bebop 2 firmware version 4.0.1.

mani-monaj, 2017-02-08:
@thomas-bamford Thanks for your detailed bug report. 

Does this issue happen when you use flight plans directly within the *FreeFlight App*?

There is a slight chance that firmware 4.0.1 is not 100% compatible with SDK 3.10.1 . I am planning to upgrade the SDK to 3.11.0 and compile the driver against it.

thomas-bamford, 2017-02-08:
Thanks for getting back to me on this. This issue does not happen when the same flight plan run in ros is run through the FreeFlight App. This is what I was thinking as well, since the issue does not occur when a point of interest flight plan is run in both environments.

Thank you for keeping this up, it has been a huge help.

mani-monaj, 2017-02-10:
@thomas-bamford You are welcome. According to SDK 3.11.0 changelog, it might be a SDK issue:

> SDK Release version 3.11.0

> Changelog:

> **Fixed non-ack commands (camera orientation was always sent)**
> Updated features list in the device controllers
> Old and deprecated Unix samples have been removed
> Coverity fixes

thomas-bamford, 2017-02-22:
Hi Mani, I have updated to the most recent version of bebop-autonomy and sdk and this bug is no longer occurring.

mani-monaj, 2017-03-13:
Thanks @thomas-bamford. I released the new SDK and updated the driver to use it. This problem should now be fixed.

closed 2017-03-13 by mani-monaj