AutonomyLab/bebop_autonomy #93: Catkin build fails on Rpi 3

siddyyagno3, 2017-02-08:
Ubuntu MATE 16.04.1 and ROS kinetic on a Rpi 3 is my setup.
Ros installation is barebones.
The output of catkin_make is 
`[ 97%] Generating C++ code from bebop_msgs/Ardrone3MediaStreamingStateVideoEnableChanged.msg
[ 97%] Generating C++ code from bebop_msgs/CommonRunStateRunIdChanged.msg
[ 97%] Generating C++ code from bebop_msgs/CommonCommonStateCurrentDateChanged.msg
[ 98%] Generating C++ code from bebop_msgs/CommonCommonStateSensorsStatesListChanged.msg
[ 98%] Generating C++ code from bebop_msgs/CommonOverHeatStateOverHeatChanged.msg
[ 98%] Built target bebop_msgs_generate_messages_cpp
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed`

`[ 78%] Built target bebop_msgs_generate_messages_cpp
[ 98%] Built target bebop_msgs_generate_messages_lisp
Scanning dependencies of target bebop_msgs_generate_messages
[ 98%] Built target bebop_msgs_generate_messages
c++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-5/README.Bugs> for instructions.
bebop_autonomy/bebop_driver/CMakeFiles/bebop.dir/build.make:62: recipe for target 'bebop_autonomy/bebop_driver/CMakeFiles/bebop.dir/src/bebop.cpp.o' failed
make[2]: *** [bebop_autonomy/bebop_driver/CMakeFiles/bebop.dir/src/bebop.cpp.o] Error 4
CMakeFiles/Makefile2:4529: recipe for target 'bebop_autonomy/bebop_driver/CMakeFiles/bebop.dir/all' failed
make[1]: *** [bebop_autonomy/bebop_driver/CMakeFiles/bebop.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed`

Also errors looks like
`Generating Lisp code from bebop_msgs/Ardrone3AntiflickeringStateelectricFrequencyChanged.msg
[ 77%] Generating Lisp code from bebop_msgs/CommonAnimationsStateList.msg
[ 78%] Generating Lisp code from bebop_msgs/CommonARLibsVersionsStateSkyControllerLibARCommandsVersion.msg
In file included from /home/rbccps/bebop_ws/src/bebop_autonomy/bebop_driver/src/bebop.cpp:47:0:
/home/rbccps/bebop_ws/src/bebop_autonomy/bebop_driver/include/bebop_driver/autogenerated/ardrone3_state_callbacks.h:43:66: fatal error: bebop_msgs/Ardrone3MediaRecordStateVideoStateChanged.h: No such file or directory
compilation terminated.
[ 78%] Generating Lisp code from bebop_msgs/Ardrone3NetworkStateAllWifiScanChanged.msg
bebop_autonomy/bebop_driver/CMakeFiles/bebop.dir/build.make:62: recipe for target 'bebop_autonomy/bebop_driver/CMakeFiles/bebop.dir/src/bebop.cpp.o' failed
make[2]: *** [bebop_autonomy/bebop_driver/CMakeFiles/bebop.dir/src/bebop.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
[ 78%] Generating Lisp code from bebop_msgs/CommonCommonStateDeprecatedMassStorageContentChanged.msg
[ 78%] Generating C++ code from bebop_msgs/Ardrone3GPSStateHomeTypeChosenChanged.msg
CMakeFiles/Makefile2:4529: recipe for target 'bebop_autonomy/bebop_driver/CMakeFiles/bebop.dir/all' failed
make[1]: *** [bebop_autonomy/bebop_driver/CMakeFiles/bebop.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 79%] Generating C++ code from bebop_msgs/CommonChargerStateMaxChargeRateChanged.msg
[ 79%] Generating Lisp code from bebop_msgs/Ardrone3PilotingStateNavigateHomeStateChanged.msg`

And warning looks like
```
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Checking for module 'libavcodec'
--   Found libavcodec, version 56.60.100
-- Checking for module 'libavformat'
--   Found libavformat, version 56.40.101
-- Checking for module 'libswscale'
--   Found libswscale, version 3.1.101
CMake Warning at /opt/ros/kinetic/share/catkin/cmake/catkin_package.cmake:166 (message):
  catkin_package() DEPENDS on 'system_lib' but neither
  'system_lib_INCLUDE_DIRS' nor 'system_lib_LIBRARIES' is defined.
Call Stack (most recent call first):
  /opt/ros/kinetic/share/catkin/cmake/catkin_package.cmake:102 (_catkin_package)
  bebop_autonomy/bebop_driver/CMakeLists.txt:40 (catkin_package)

CMake Warning at /opt/ros/kinetic/share/catkin/cmake/catkin_package.cmake:166 (message):
  catkin_package() DEPENDS on 'libavcodec' but neither
  'libavcodec_INCLUDE_DIRS' nor 'libavcodec_LIBRARIES' is defined.
Call Stack (most recent call first):
  /opt/ros/kinetic/share/catkin/cmake/catkin_package.cmake:102 (_catkin_package)
  bebop_autonomy/bebop_driver/CMakeLists.txt:40 (catkin_package)
[... 115 lines cut ...]
[build] Runtime: 38 minutes and 18.9 seconds total.
`
```
And log file output is
````
[35m[1mScanning dependencies of target bebop_driver_gencfg[0m
[ 12%] [34m[1mGenerating dynamic reconfigure files from cfg/autogenerated/BebopArdrone3.cfg: /home/rbccps/bebop_ws/devel/.private/bebop_driver/include/bebop_driver/BebopArdrone3Config.h /home/rbccps/bebop_ws/devel/.private/bebop_driver/lib/python2.7/dist-packages/bebop_driver/cfg/BebopArdrone3Config.py[0m
Generating reconfiguration files for BebopArdrone3 in bebop_driver
Wrote header file in /home/rbccps/bebop_ws/devel/.private/bebop_driver/include/bebop_driver/BebopArdrone3Config.h
[ 12%] Built target bebop_driver_gencfg
[35m[1mScanning dependencies of target bebop[0m
[ 25%] [32mBuilding CXX object CMakeFiles/bebop.dir/src/bebop.cpp.o[0m
[ 37%] [32mBuilding CXX object CMakeFiles/bebop.dir/src/bebop_video_decoder.cpp.o[0m
[01m[Kc++:[m[K [01;31m[Kinternal compiler error: [m[KKilled (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-5/README.Bugs> for instructions.
make[2]: *** [CMakeFiles/bebop.dir/src/bebop.cpp.o] Error 4
CMakeFiles/bebop.dir/build.make:62: recipe for target 'CMakeFiles/bebop.dir/src/bebop.cpp.o' failed
CMakeFiles/Makefile2:1096: recipe for target 'CMakeFiles/bebop.dir/all' failed
make[1]: *** [CMakeFiles/bebop.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
```
Any help would be appreciated.

thomas-bamford, 2017-06-23:
Hello,
Sorry for the late reply. Please see issue #118, this may help with the problem you are having. I will repeat the solution here, instead of running
`catkin build -DCMAKE_BUILD_TYPE=RelWithDebInfo`
please try running
`catkin build`
Please let us know if this helps.
Thanks.

mani-monaj, 2017-06-27:
Thanks @thomas-bamford for the solution. 

The CI does not catch this error. Do you think if it is related to newer Ubuntu/Debian versions being used that the CI?

pmordel, 2018-06-06:
> bebop_driver/autogenerated/ardrone3_state_callbacks.h:43:66: fatal error: bebop_msgs/Ardrone3MediaRecordStateVideoStateChanged.h: No such file or directory

This is typically an error due to the fact that the messages were not generated before the build was made.

To make sur that all your messages are well generated in the first place, you need to edit the "bebop_autonomy/bebop_driverCMakeLists.txt" file 

Replace the line
`add_dependencies(bebop ${PROJECT_NAME}_gencfg)`
by 
`'add_dependencies(bebop ${PROJECT_NAME}_gencfg ${catkin_EXPORTED_TARGETS})`

Works for me. I tried both version in a new workspace with only this modification, the first one send me this error, the second one compile successfully.