AutonomyLab/bebop_autonomy #95: How can I disable giro stabilization?

RobertoIV, 2017-02-15:
Hello, I've been trying to disable giro stabilization but I cannot disable it completely, I would like to disable it to avoid the blur caused by the stabilization and to get the orientation of the bebop based on the camera.
Thanks in advance.

mani-monaj, 2017-02-18:
Have you tried setting the following parameter:

`~PictureSettingsVideoStabilizationModeMode`

http://bebop-autonomy.readthedocs.io/en/latest/autogenerated/ardrone3_settings_param.html#picturesettingsvideostabilizationmodemode

RobertoIV, 2017-02-22:
I have been able to disable it thay way, but I would like to disable it from the start of the node. Thanks for your atention.

mani-monaj, 2017-03-13:
@RobertoIV You can set any ROS param directly from your launch file:

http://wiki.ros.org/roslaunch/XML#Setting_parameters