AutonomyLab/bebop_autonomy #104: Problem when trying to run Bebop’s ROS driver

Brian951, 2017-04-01:
Hi! I am relatively new to ROS and bebop drones, and I am just starting to install and use bebop_autonomy to connect to my  Parrot Bebop drone.

I could ping my drone at  192.168.42.1, but then I encounter problems when I am following the documentation on "Running the Driver".

When I try to run the driver as a node or a nodelet, I get error messages as followings:

$ roslaunch bebop_driver bebop_node.launch
... logging to /home/brian/.ros/log/4d22fb4c-1711-11e7-80d1-0800279b4b72/roslaunch-brian-VirtualBox-28034.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: Traditional processing is deprecated. Switch to --inorder processing!
To check for compatibility of your document, use option --check-order.
For more infos, see http://wiki.ros.org/xacro#Processing_Order
started roslaunch server http://brian-VirtualBox:35861/

SUMMARY
========

PARAMETERS
 * /bebop/bebop_driver/bebop_ip: 192.168.42.1
 * /bebop/bebop_driver/camera_info_url: package://bebop_d...
 * /bebop/bebop_driver/cmd_vel_timeout: 0.2
 * /bebop/bebop_driver/odom_frame_id: odom
 * /bebop/bebop_driver/publish_odom_tf: True
 * /bebop/bebop_driver/reset_settings: True
 * /bebop/bebop_driver/states/enable_altitudechanged: True
 * /bebop/bebop_driver/states/enable_autotakeoffmodechanged: True
 * /bebop/bebop_driver/states/enable_camerastate_orientation: True
 * /bebop/bebop_driver/states/enable_commonstate_batterystatechanged: True
 * /bebop/bebop_driver/states/enable_commonstate_wifisignalchanged: True
 * /bebop/bebop_driver/states/enable_controllerstate_ispilotingchanged: True
 * /bebop/bebop_driver/states/enable_flightplanstate_availabilitystatechanged: True
 * /bebop/bebop_driver/states/enable_flightplanstate_componentstatelistchanged: True
 * /bebop/bebop_driver/states/enable_gpsstate_numberofsatellitechanged: True
 * /bebop/bebop_driver/states/enable_mavlinkstate_mavlinkfileplayingstatechanged: True
 * /bebop/bebop_driver/states/enable_mavlinkstate_mavlinkplayerrorstatechanged: True
 * /bebop/bebop_driver/states/enable_mediastreamingstate_videoenablechanged: True
 * /bebop/bebop_driver/states/enable_numberofsatellitechanged: True
 * /bebop/bebop_driver/states/enable_overheatstate_overheatchanged: True
 * /bebop/bebop_driver/states/enable_pilotingstate_altitudechanged: True
 * /bebop/bebop_driver/states/enable_pilotingstate_attitudechanged: True
 * /bebop/bebop_driver/states/enable_pilotingstate_flattrimchanged: True
 * /bebop/bebop_driver/states/enable_pilotingstate_flyingstatechanged: True
 * /bebop/bebop_driver/states/enable_pilotingstate_navigatehomestatechanged: True
 * /bebop/bebop_driver/states/enable_pilotingstate_positionchanged: True
 * /bebop/bebop_driver/states/enable_pilotingstate_speedchanged: True
 * /bebop/robot_description: <?xml version="1....
 * /rosdistro: kinetic
 * /rosversion: 1.12.7

NODES
  /bebop/
    bebop_driver (bebop_driver/bebop_driver_node)
    robot_state_publisher (robot_state_publisher/robot_state_publisher)

auto-starting new master
process[master]: started with pid [28048]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 4d22fb4c-1711-11e7-80d1-0800279b4b72
process[rosout-1]: started with pid [28061]
started core service [/rosout]
process[bebop/bebop_driver-2]: started with pid [28073]
process[bebop/robot_state_publisher-3]: started with pid [28079]
[ INFO] [1491074874.747679255]: Initializing nodelet with 1 worker threads.
[ INFO] [1491074874.904582440]: [BebopSDK] 03:27:54:904 | Bebop:225 - Bebop Cnstr()
[ INFO] [1491074874.905884516]: Nodelet Cstr
[ INFO] [1491074874.919409933]: Connecting to Bebop ...
[... 29 lines cut ...]
[ERROR] [1491074877.847486088]: [ARNETWORK_Sender] 03:27:57:847 | ARNETWORK_Sender_ProcessBufferToSend:421 - [0x7ff9182a7de0] Will retry sending data of buffer 11
[ERROR] [1491074878.348124995]: [ARNETWORK_Sender] 03:27:58:348 | ARNETWORK_Sender_ProcessBufferToSend:405 - [0x7ff9182a7de0] Timeout waiting for ack in buffer 11
[ERROR] [1491074878.348251012]: [ARNETWORK_Sender] 03:27:58:348 | ARNETWORK_Sender_ProcessBufferToSend:421 - [0x7ff9182a7de0] Will retry sending data of buffer 11
[ERROR] [1491074878.848439032]: [ARNETWORK_Sender] 03:27:58:848 | ARNETWORK_Sender_ProcessBufferToSend:405 - [0x7ff9182a7de0] Timeout waiting for ack in buffer 11
[ERROR] [1491074878.848561779]: [ARNETWORK_Sender] 03:27:58:848 | ARNETWORK_Sender_ProcessBufferToSend:421 - [0x7ff9182a7de0] Will retry sending data of buffer 11
[ERROR] [1491074879.349409300]: [ARNETWORK_Sender] 03:27:59:349 | ARNETWORK_Sender_ProcessBufferToSend:405 - [0x7ff9182a7de0] Timeout waiting for ack in buffer 11**
_[ WARN] [1491074879.349558890]: [ARNETWORK_Sender] 03:27:59:349 | ARNETWORK_Sender_TimeOutCallback:655 - [0x7ff9182a7de0] Did timeout sending command : project = 0 | class = 4 | command = 2 - 0_
**[ERROR] [1491074879.849932869]: [ARNETWORK_Sender] 03:27:59:849 | ARNETWORK_Sender_ProcessBufferToSend:405 - [0x7ff9182a7de0] Timeout waiting for ack in buffer 11
[ERROR] [1491074879.850073686]: [ARNETWORK_Sender] 03:27:59:850 | ARNETWORK_Sender_ProcessBufferToSend:421 - [0x7ff9182a7de0] Will retry sending data of buffer 11**
[ INFO] [1491074880.327451025]: [ARNETWORKAL_WifiNetwork] 03:28:00:327 | ARNETWORKAL_WifiNetwork_Receive:918 - [0x7ff91824dff0] connection lost (too long time without reception)
[ INFO] [1491074880.327663960]: [ARCONTROLLER_Device] 03:28:00:327 | ARCONTROLLER_Device_StartRun:4987 - Start failed or canceled.
**[ERROR] [1491074880.351238007]: [ARNETWORK_Sender] 03:28:00:351 | ARNETWORK_Sender_ProcessBufferToSend:405 - [0x7ff9182a7de0] Timeout waiting for ack in buffer 11
[ERROR] [1491074880.351367810]: [ARNETWORK_Sender] 03:28:00:351 | ARNETWORK_Sender_ProcessBufferToSend:421 - [0x7ff9182a7de0] Will retry sending data of buffer 11
[ERROR] [1491074881.325474151]: [ARCONTROLLER_Device] 03:28:01:325 | ARCONTROLLER_Device_StartRun:4995 - Start fail error :start canceled**
[ INFO] [1491074881.325635335]: [BebopSDK] 03:28:01:325 | Cleanup:326 - Bebop Cleanup()
[FATAL] [1491074881.325830937]: Init failed: Waiting for device failed: No error
[ INFO] [1491074881.325905075]: Bebop Nodelet Dstr: 0
[ INFO] [1491074881.325916417]: Killing Camera Thread ...
[ INFO] [1491074881.325929273]: Killing Aux Thread ...
^C[bebop/robot_state_publisher-3] killing on exit
[bebop/bebop_driver-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit

What could be the possible cause of this and how could I fix this?

MabeCr, 2017-04-11:
Are you trying to run this in a VM? I'm attempting to run it in a docker container on windows, and I'm having the same problem. I'm not entirely sure how to configure it so that it works correctly.

thomas-bamford, 2017-04-11:
@MabeCr I run the bebop through a virtual machine on Windows. You require a bridged network adapter on your virtual machine to get it to run. The default adapter is configured as NAT.

MabeCr, 2017-04-11:
@thomas-bamford I'm fairly certain a bridge is the default adapter for Docker. I guess I could download a VM, but the whole point of us using Docker is that any version of Ubuntu/Windows could run our environment.

thomas-bamford, 2017-04-11:
@MabeCr Sorry, the default adapter for the VM I use is NAT (VMWare).

thomas-bamford, 2017-06-23:
@Brian951 Sorry for the late follow up. Have you tried running the driver with the latest version of bebop_autonomy?

asimonov, 2018-06-29:
hi, I am running this on ubuntu 16.04/kinetic with the latest bebop_autonomy and latest bebop firmware (4.7.1) on bebop 2. can ping my drone from Ubuntu (not VM) but see the above errors