AutonomyLab/bebop_autonomy #108: Connection easy to lost after bebop ip changed

chesternimiz, 2017-05-05:
We are currently controlling 3 bebop 2 drones by 1 computer. We ssh and configure the ip of bebops to 192.168.42.1, 192.168.42.2, 192.168.42.3 respectively. We connect the three bebops using three USB WLAN card and set the route table correctly. We add <group ns="uav0/1/2"> to launch three ROS drivers and set the ip parameter for them.
Everthing just works fine.

But the problem is that during our experiments, the connections to bebops (192.168.42.2 and 192.168.42.3) are easy to lost. The ROS driver report:
[ARNETWORKAL_WifiNetwork]...connection lost (too long time without reception)...
[ARSTREAM2_StreamReceiver]...App output thread has ended
[ARSTREAM2_StreamReceiver]...Receiver thread has ended

But at this time, there's no problem by pinging 192.168.42.2 or 42.3. Restarting the ros driver could fix the problem. (But after restarting, the problem might occur again. It is difficult to track what causes the problem.)

When the problem occurs, we can no longer interact with the bebop, neither receiving sensor data nor control it. This problem occur occasionally. But the one with ip 192.168.42.1 never has this problem.

We also tested by controlling only one drone. The ip is set to 192.168.42.2. This connection problem still exists and occurs occasionally. When the ip is set to 192.168.42.1, there's no problem.

Could you please give me any suggestion on solving this problem? It's really tough to has this problem when flying with multi drones. Each time we have to restart all ROS drivers. Thanks for your attention.

chesternimiz, 2017-05-08:
Is there any way to change the time for judging connection lost? I think it may be helpful if I could have a longer time. 
I checked the code but just could not find where "[ARNETWORKAL_WifiNetwork]...connection lost (too long time without reception)..." was generated. I hope I could have a longer time to determine connection lost, any help will be appreciated.

tnaegeli, 2017-05-12:
Hi,

Not sure if this helps, but we used 4 usb network cards. Then each card connected to a Bebop2 network. Because of IP subnetwork issues, we renamed the networks in the following way:

Bebop1: 192.168.101.1
Bebop2: 192.168.102.1
Bebop3: 192.168.103.1
Bebop4: 192.168.104.1

Which this ip configurations we had no problems at all. but the ip has to be changed on the drone itself.

best,
Tobi

DengMark, 2018-06-12:
Hi,

Have you solved the connection lost problem when controlling multiple bebops. We have the same problem, but we use a router to constitute a LAN. with the IP address: 
Bebop1: 192.168.207.51
Bebop2: 192.168.207.52
Bebop3: 192.168.207.53
Bebop4: 192.168.207.54
The connection is easy to lost.