AutonomyLab/bebop_autonomy #111: prolem in ros for bebop?

kevin456, 2017-05-29:
when I Install the bebop in ros,I face the problem

# Create and initialize the workspace
$ mkdir -p ~/bebop_ws/src && cd ~/bebop_ws
$ catkin init
$ git clone https://github.com/AutonomyLab/bebop_autonomy.git src/bebop_autonomy
# Update rosdep database and install dependencies (including parrot_arsdk)
$ rosdep update
$ rosdep install --from-paths src -i

above all are successful

but when I next

# Build the workspace
$ catkin build -DCMAKE_BUILD_TYPE=RelWithDebInfo

the problems are appearing

Failed     << bebop_driver:make                [ Exited with code 2 ]          
Failed    <<< bebop_driver                     [ 37.9 seconds ]                
Abandoned <<< bebop_tools                      [ Unrelated job failed ]        
[build] Summary: 2 of 4 packages succeeded.                                    
[build]   Ignored:   1 packages were skipped or are blacklisted.               
[build]   Warnings:  None.                                                     
[build]   Abandoned: 1 packages were abandoned.                                
[build]   Failed:    1 packages failed.                                        
[build] Runtime: 43.7 seconds total.                                           
[build] Note: Workspace packages have changed, please re-source setup files to use them.

How to slove this problem?

thomas-bamford, 2017-06-23:
Hello,
Sorry for the late comment. Please see issue #118, I believe the solution to this should work for the problem you have been having. I will repeat it here, instead of running
`catkin build -DCMAKE_BUILD_TYPE=RelWithDebInfo`
Run
`catkin build`
instead. Please let us know if this solves your problem.
Thanks.

Cullen-Dubbs, 2017-10-29:
Hi there, i get the same issue after running catkin build as well, any suggestions?

thomas-bamford, 2017-10-30:
What version of ROS are you trying to install this for?

wajeehulhassanvii, 2019-04-08:
Installing on ROS Kinetic and getting the same error, can you please help. I saw different threads and followed the instructions but cound't get past this error.

araujorayza, 2019-04-20:
> Hi there, i get the same issue after running catkin build as well, any suggestions?

me too. running ros kinetic on ubuntu 16.04

Debjoy10, 2019-05-25:
Any updates? Getting same error on ROS-Kinetic ubuntu 16.04

wajeehulhassanvii, 2019-05-26:
> Any updates? Getting same error on ROS-Kinetic ubuntu 16.04

I don't remember what I did but I found bebop drone in a Rotors simulator, you can also get try Rotors simulator in another workspace. I'm working with Ardrone by the way.