AutonomyLab/bebop_autonomy #114: Update to Parrot ARSDK 3.12.4

mani-monaj, 2017-06-13:


thomas-bamford, 2017-06-22:
Finished update of autonomy lab parrot arsdk to 3.12.6 and submitted PR. I tried modifying and running the generate.py script and install.sh in bebop_driver/scripts/meta folder but some errors occur on the build of bebop_autonomy due to new names that do not fit the autogenerate script. Are there extra instructions that I may be missing (#113)?

mani-monaj, 2017-06-27:
@thomas-bamford Have you checked this part of the documentation? http://bebop-autonomy.readthedocs.io/en/latest/dev.html#upgrading-the-bebop-sdk (Step 6)?

thomas-bamford, 2017-06-27:
@mani-monaj Thanks for pointing this out. Managed to fix these names (and some others) and everything works well. Submitted PR, but still have to test on vehicles.

mani-monaj, 2017-06-28:
@thomas-bamford You are welcome! Both AutonomyLab/parrot_arsdk#5 and #120 look great. Please check my comments on both those issues.

VINICIOSTALIN, 2017-10-23:
sorry,
In the last version I tried to read the states, such as the battery changed but I get this error

ERROR: Cannot load message class for [bebop_msgs/CommonCommonStateBatteryStateChanged]. Are your messages built?

I installed everything as says in instructions, even so I could not read the information published in any of the bebop / states /.. topics

Could you help me with this?

thomas-bamford, 2017-10-30:
Hello @VINICIOSTALIN, could you please open an issue for this?
Quick guess at this: If you are updating to latest version (not clean install) you may need to clean the workspace before rebuilding, i.e.
`catkin clean`
`catkin build`
Or you may need to update the parrot-arsdk package to the latest version before building. If this does not work than this may be another issue but we should discuss it in the issues section.

VINICIOSTALIN, 2017-11-09:
Excuse me @thomas-bamford  I could not solve it, could you help me in issue #143 that I have raised?

closed 2018-07-03 by thomas-bamford