AutonomyLab/bebop_autonomy #117: No video stream when running bebop_nodelet_iv

DingdianZhang, 2017-06-20:
tried on two software versions of Bebop1 2.0.29 and 4.0.6
Ubuntu 16.04
ROS Kinetic 1.12.7
When I run `roslaunch bebop_tools bebop_nodelet_iv.launch` it shows

SUMMARY
========

PARAMETERS
 * /bebop/bebop_nodelet/bebop_ip: 192.168.42.1
 * /bebop/bebop_nodelet/camera_info_url: package://bebop_d...
 * /bebop/bebop_nodelet/cmd_vel_timeout: 0.2
 * /bebop/bebop_nodelet/odom_frame_id: odom
 * /bebop/bebop_nodelet/publish_odom_tf: True
 * /bebop/bebop_nodelet/reset_settings: True
 * /bebop/bebop_nodelet/states/enable_altitudechanged: True
 * /bebop/bebop_nodelet/states/enable_autotakeoffmodechanged: True
 * /bebop/bebop_nodelet/states/enable_camerastate_orientation: True
 * /bebop/bebop_nodelet/states/enable_commonstate_batterystatechanged: True
 * /bebop/bebop_nodelet/states/enable_commonstate_wifisignalchanged: True
 * /bebop/bebop_nodelet/states/enable_controllerstate_ispilotingchanged: True
 * /bebop/bebop_nodelet/states/enable_flightplanstate_availabilitystatechanged: True
 * /bebop/bebop_nodelet/states/enable_flightplanstate_componentstatelistchanged: True
 * /bebop/bebop_nodelet/states/enable_gpsstate_numberofsatellitechanged: True
 * /bebop/bebop_nodelet/states/enable_mavlinkstate_mavlinkfileplayingstatechanged: True
 * /bebop/bebop_nodelet/states/enable_mavlinkstate_mavlinkplayerrorstatechanged: True
 * /bebop/bebop_nodelet/states/enable_mediastreamingstate_videoenablechanged: True
 * /bebop/bebop_nodelet/states/enable_numberofsatellitechanged: True
 * /bebop/bebop_nodelet/states/enable_overheatstate_overheatchanged: True
 * /bebop/bebop_nodelet/states/enable_pilotingstate_altitudechanged: True
 * /bebop/bebop_nodelet/states/enable_pilotingstate_attitudechanged: True
 * /bebop/bebop_nodelet/states/enable_pilotingstate_flattrimchanged: True
 * /bebop/bebop_nodelet/states/enable_pilotingstate_flyingstatechanged: True
 * /bebop/bebop_nodelet/states/enable_pilotingstate_navigatehomestatechanged: True
 * /bebop/bebop_nodelet/states/enable_pilotingstate_positionchanged: True
 * /bebop/bebop_nodelet/states/enable_pilotingstate_speedchanged: True
 * /bebop/robot_description: <?xml version="1....
 * /rosdistro: kinetic
 * /rosversion: 1.12.7

NODES
  /bebop/
    bebop_image_view_nodelet (nodelet/nodelet)
    bebop_nodelet (nodelet/nodelet)
    bebop_nodelet_manager (nodelet/nodelet)
    robot_state_publisher (robot_state_publisher/robot_state_publisher)

auto-starting new master
process[master]: started with pid [28671]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 0c425e36-5516-11e7-bb49-00270e0a73dc
process[rosout-1]: started with pid [28684]
started core service [/rosout]
process[bebop/bebop_nodelet_manager-2]: started with pid [28701]
process[bebop/bebop_nodelet-3]: started with pid [28702]
process[bebop/robot_state_publisher-4]: started with pid [28703]
process[bebop/bebop_image_view_nodelet-5]: started with pid [28710]
[ INFO] [1497893885.401550782]: Initializing nodelet with 4 worker threads.
[ INFO] [1497893885.460768911]: Using transport "raw"
init done

(nodelet:28701): GLib-GObject-WARNING **: invalid uninstantiatable type '(null)' in cast to 'GtkWidget'

(nodelet:28701): GLib-GObject-WARNING **: instance of invalid non-instantiatable type '(null)'

(nodelet:28701): GLib-GObject-CRITICAL **: g_signal_connect_data: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed
[ INFO] [1497893885.727227095]: [BebopSDK] 18:38:05:727 | Bebop:225 - Bebop Cnstr()
[ INFO] [1497893885.727303841]: Nodelet Cstr
[ INFO] [1497893885.741263686]: Connecting to Bebop ...
[... 89 lines cut ...]
**[ WARN] [1497893922.068382032]: [BebopSDK] 18:38:42:068 | FrameReceivedCallback:191 - Previous frame might have been missed.
[ WARN] [1497893922.611161718]: [BebopSDK] 18:38:42:611 | FrameReceivedCallback:191 - Previous frame might have been missed.
[ WARN] [1497893922.882378332]: [BebopSDK] 18:38:42:882 | FrameReceivedCallback:191 - Previous frame might have been missed.
[ WARN] [1497893923.156929273]: [BebopSDK] 18:38:43:156 | FrameReceivedCallback:191 - Previous frame might have been missed.
[ INFO] [1497893923.298410984]: [ARSTREAM_Reader] 18:38:43:298 | ARSTREAM_Reader_RunDataThread:538 - Missed 1 frames !
[ INFO] [1497893923.419359568]: [ARSTREAM_Reader] 18:38:43:419 | ARSTREAM_Reader_RunDataThread:538 - Missed 1 frames !
[ WARN] [1497893923.709226909]: [BebopSDK] 18:38:43:709 | FrameReceivedCallback:191 - Previous frame might have been missed.
[ WARN] [1497893923.712641504]: [BebopSDK] 18:38:43:712 | FrameReceivedCallback:191 - Previous frame might have been missed.
[ WARN] [1497893923.980139307]: [BebopSDK] 18:38:43:980 | FrameReceivedCallback:191 - Previous frame might have been missed.
[ WARN] [1497893923.983467047]: [BebopSDK] 18:38:43:983 | FrameReceivedCallback:191 - Previous frame might have been missed.
[ WARN] [1497893924.252216971]: [BebopSDK] 18:38:44:252 | FrameReceivedCallback:191 - Previous frame might have been missed.
[ WARN] [1497893924.523443738]: [BebopSDK] 18:38:44:523 | FrameReceivedCallback:191 - Previous frame might have been missed.
[ WARN] [1497893924.795732628]: [BebopSDK] 18:38:44:795 | FrameReceivedCallback:191 - Previous frame might have been missed.
[ WARN] [1497893925.074378419]: [BebopSDK] 18:38:45:074 | FrameReceivedCallback:191 - Previous frame might have been missed.
[ WARN] [1497893925.341275785]: [BebopSDK] 18:38:45:341 | FrameReceivedCallback:191 - Previous frame might have been missed.
[ WARN] [1497893985.063449340]: [BebopSDK] 18:39:45:063 | FrameReceivedCallback:191 - Previous frame might have been missed.
[ WARN] [1497893985.336799046]: [BebopSDK] 18:39:45:336 | FrameReceivedCallback:191 - Previous frame might have been missed.
[ WARN] [1497893985.611102064]: [BebopSDK] 18:39:45:611 | FrameReceivedCallback:191 - Previous frame might have been missed.**

The main problem is there is no video stream received. 
The error is **[ERROR] [1497893894.048743322]: [Decoder] 18:38:14:048 | Decode:302 - Frame size changed to 640 x 368**.
I can send messages to the Bebop by `rostopic pub`, such as takeoff and land.

Please help!!!
Many thanks.

thomas-bamford, 2017-06-23:
Hello,
Please try the following, in one terminal
`roslaunch bebop_driver bebop_nodelet.launch`
and then in a new terminal
`rosrun image_view image_view image:=/bebop/image_raw`
This uses a different method of subscribing to the image topic broadcasted by the bebop. If this does not solve this and the launch you tried above continues to fail, please post more information on the vehicle hardware, vehicle firmware, and system architecture.
Thanks,
Thomas

DingdianZhang, 2017-06-23:
@thomas-bamford It works thank you so much!!!!

closed 2017-06-23 by DingdianZhang