AutonomyLab/bebop_autonomy #123: Unsuccessful installation

mnahangi, 2017-07-11:
the driver is not installed successfully. The terminal output is pasted below for your reference. 
I get the following msgs after building the catkin workspace for the bebop driver.

Profile:                     default
Extending:             [env] /home/mohammad/catkin_ws/devel:/home/mohammad/charlottetown/ros_osrf/install_isolated
Workspace:                   /home/mohammad/bebop_ws
-------------------------------------------------------------------
Source Space:       [exists] /home/mohammad/bebop_ws/src
Log Space:         [missing] /home/mohammad/bebop_ws/logs
Build Space:        [exists] /home/mohammad/bebop_ws/build
Devel Space:        [exists] /home/mohammad/bebop_ws/devel
Install Space:      [unused] /home/mohammad/bebop_ws/install
DESTDIR:            [unused] None
-------------------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
-------------------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
-------------------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
-------------------------------------------------------------------
Workspace configuration appears valid.

NOTE: Forcing CMake to run for each package.
-------------------------------------------------------------------
[build] Found '5' packages in 0.0 seconds.                        
[build] Updating package table.                                   
Starting  >>> catkin_tools_prebuild                               
Finished  <<< catkin_tools_prebuild                [ 1.7 seconds ]
Starting  >>> bebop_description                                   
Starting  >>> bebop_msgs                                          
Finished  <<< bebop_description                    [ 1.8 seconds ]
Finished  <<< bebop_msgs                           [ 16.9 seconds ]
Starting  >>> bebop_driver                                        
__________________________________________________________________
Errors     << bebop_driver:cmake /home/mohammad/bebop_ws/logs/bebop_driver/build.cmake.000.log
CMake Warning at /home/mohammad/charlottetown/ros_osrf/install_isolated/share/catkin/cmake/catkinConfig.cmake:76 (find_package):
  Could not find a package configuration file provided by "parrot_arsdk" with
  any of the following names:

    parrot_arsdkConfig.cmake
    parrot_arsdk-config.cmake

  Add the installation prefix of "parrot_arsdk" to CMAKE_PREFIX_PATH or set
  "parrot_arsdk_DIR" to a directory containing one of the above files.  If
  "parrot_arsdk" provides a separate development package or SDK, be sure it
  has been installed.
Call Stack (most recent call first):
  CMakeLists.txt:7 (find_package)

CMake Error at /home/mohammad/charlottetown/ros_osrf/install_isolated/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by "parrot_arsdk" with
  any of the following names:

    parrot_arsdkConfig.cmake
    parrot_arsdk-config.cmake

  Add the installation prefix of "parrot_arsdk" to CMAKE_PREFIX_PATH or set
  "parrot_arsdk_DIR" to a directory containing one of the above files.  If
  "parrot_arsdk" provides a separate development package or SDK, be sure it
  has been installed.
Call Stack (most recent call first):
  CMakeLists.txt:7 (find_package)

cd /home/mohammad/bebop_ws/build/bebop_driver; catkin build --get-env bebop_driver | catkin env -si  /usr/bin/cmake /home/mohammad/bebop_ws/src/bebop_autonomy/bebop_driver --no-warn-unused-cli -DCATKIN_DEVEL_PREFIX=/home/mohammad/bebop_ws/devel/.private/bebop_driver -DCMAKE_INSTALL_PREFIX=/home/mohammad/bebop_ws/install; cd -
..................................................................
Failed     << bebop_driver:cmake                   [ Exited with code 1 ]
Failed    <<< bebop_driver                         [ 1.6 seconds ]
Abandoned <<< bebop_tools                          [ Unrelated job failed ]
[build] Summary: 3 of 5 packages succeeded.                       
[build]   Ignored:   1 packages were skipped or are blacklisted.  
[build]   Warnings:  None.                                        
[build]   Abandoned: 1 packages were abandoned.                   
[build]   Failed:    1 packages failed.                           
[build] Runtime: 20.5 seconds total.                              
[build] Note: Workspace packages have changed, please re-source setup files to use them.
[mohammad@dsl6:~/bebop_ws]

thomas-bamford, 2017-07-11:
Hi Mohammad, this issue is caused by the ROS environment not being able to find the parrot_arsdk package which the bebop_driver package depends on. Once you have run the `rosdep install --from-path src --ignore-src` ensure that the parrot_arsdk package can be found by running `rospack find parrot_arsdk` in a new terminal.

Thomas

mnahangi, 2017-07-11:
Thanks Thomas for fixing this issue. The issue is now resolved.

closed 2017-07-11 by mnahangi