AutonomyLab/bebop_autonomy #124: Unable to get GPS fix on Bebop Drone

BHAARATH002, 2017-07-12:
Hello,
I'm currently working on an android app for Bebop Drone. I'm using the bebop SDK to build the app.
I'm currently having problem with the GPS fix on the drone. When I take off in the outdoor i can't get a fix on the drone but I can get a fix when is use the FreeFlight Pro app.

Could someone help out please.

Adam-JW-Casey, 2017-07-18:
Are you using this ros package or just the SDK?

BHAARATH002, 2017-07-19:
Hi i can get a fix on the gps now.

thomas-bamford, 2017-07-19:
Hello,
Could you explain this issue was solved? Also, (as @Adam-JW-Casey asked) were you using the ros package or just the SDK?
Thanks.

BHAARATH002, 2017-07-20:
Hi,
I'm just using the SDK.
I came to a conclusion that there will be few meters of precision loss.
Even after i got a gps fix the drone's Lat and Lng continues to change at a stationary position. Because of that, the value taken at the take off position losses some precision. 
So when it takes off, the drone will move to the take off position first then continues with the rest of the waypoints and back to the takeoff position specified.
Regards.