AutonomyLab/bebop_autonomy #126: Can't install bebop autonomy Ubuntu 14.04

anderdefector, 2017-07-26:
I've been following the tutorials, but i just got that error.

`/home/ander/bebop_ws/src/bebop_autonomy/bebop_driver/include/bebop_driver/autogenerated/common_state_callbacks.h: In constructor ‘bebop_driver::cb::CommonMavlinkStateMissonItemExecuted::CommonMavlinkStateMissonItemExecuted(ros::NodeHandle&, ros::NodeHandle&, const string&)’:
/home/ander/bebop_ws/src/bebop_autonomy/bebop_driver/include/bebop_driver/autogenerated/common_state_callbacks.h:1230:21: error: ‘ARCONTROLLER_DICTIONARY_KEY_COMMON_MAVLINKSTATE_MISSONITEMEXECUTED’ was not declared in this scope
     : AbstractState(ARCONTROLLER_DICTIONARY_KEY_COMMON_MAVLINKSTATE_MISSONITEMEXECUTED)
                     ^
In file included from /opt/ros/indigo/include/parrot_arsdk/libARController/ARCONTROLLER_Dictionary.h:42:0,
                 from /opt/ros/indigo/include/parrot_arsdk/libARController/ARController.h:45,
                 from /home/ander/bebop_ws/src/bebop_autonomy/bebop_driver/src/bebop.cpp:39:
/home/ander/bebop_ws/src/bebop_autonomy/bebop_driver/include/bebop_driver/autogenerated/common_state_callbacks.h: In member function ‘virtual void bebop_driver::cb::CommonMavlinkStateMissonItemExecuted::Update(const ARCONTROLLER_DICTIONARY_ARG_t*, const ros::Time&)’:
/home/ander/bebop_ws/src/bebop_autonomy/bebop_driver/include/bebop_driver/autogenerated/common_state_callbacks.h:1260:31: error: ‘ARCONTROLLER_DICTIONARY_KEY_COMMON_MAVLINKSTATE_MISSONITEMEXECUTED_IDX’ was not declared in this scope
     HASH_FIND_STR (arguments, ARCONTROLLER_DICTIONARY_KEY_COMMON_MAVLINKSTATE_MISSONITEMEXECUTED_IDX, arg);
                               ^
/opt/ros/indigo/include/parrot_arsdk/uthash/uthash.h:430:43: note: in definition of macro ‘HASH_JEN’
   unsigned char *_hj_key=(unsigned char*)(key);                                  \
                                           ^
/opt/ros/indigo/include/parrot_arsdk/uthash/uthash.h:262:5: note: in expansion of macro ‘HASH_FIND’
     HASH_FIND(hh,head,findstr,(unsigned)strlen(findstr),out)
     ^
/home/ander/bebop_ws/src/bebop_autonomy/bebop_driver/include/bebop_driver/autogenerated/common_state_callbacks.h:1260:5: note: in expansion of macro ‘HASH_FIND_STR’
     HASH_FIND_STR (arguments, ARCONTROLLER_DICTIONARY_KEY_COMMON_MAVLINKSTATE_MISSONITEMEXECUTED_IDX, arg);
     ^
/home/ander/bebop_ws/src/bebop_autonomy/bebop_driver/include/bebop_driver/autogenerated/common_state_callbacks.h:1260:31: error: ‘ARCONTROLLER_DICTIONARY_KEY_COMMON_MAVLINKSTATE_MISSONITEMEXECUTED_IDX’ was not declared in this scope
     HASH_FIND_STR (arguments, ARCONTROLLER_DICTIONARY_KEY_COMMON_MAVLINKSTATE_MISSONITEMEXECUTED_IDX, arg);
                               ^
/opt/ros/indigo/include/parrot_arsdk/uthash/uthash.h:613:29: note: in definition of macro ‘HASH_FIND_IN_BKT’
     if ((out)->hh.keylen == keylen_in) {                                           \
                             ^
/opt/ros/indigo/include/parrot_arsdk/uthash/uthash.h:262:5: note: in expansion of macro ‘HASH_FIND’
     HASH_FIND(hh,head,findstr,(unsigned)strlen(findstr),out)
     ^
/home/ander/bebop_ws/src/bebop_autonomy/bebop_driver/include/bebop_driver/autogenerated/common_state_callbacks.h:1260:5: note: in expansion of macro ‘HASH_FIND_STR’
     HASH_FIND_STR (arguments, ARCONTROLLER_DICTIONARY_KEY_COMMON_MAVLINKSTATE_MISSONITEMEXECUTED_IDX, arg);
     ^
make[2]: *** [bebop_autonomy/bebop_driver/CMakeFiles/bebop.dir/src/bebop.cpp.o] Error 1
make[1]: *** [bebop_autonomy/bebop_driver/CMakeFiles/bebop.dir/all] Error 2
make: *** [all] Error 2
Invoking "make -j4 -l4" failed
ander@ander:~/bebop_ws$ 
`

JeenQ, 2017-08-01:
Hi I had same problem with you
and finally I found it!

There are some typo in '/home/ander/bebop_ws/src/bebop_autonomy/bebop_driver/include/bebop_driver/autogenerated/common_state_callbacks.h'

‘ARCONTROLLER_DICTIONARY_KEY_COMMON_MAVLINKSTATE_MISSONITEMEXECUTED’ -> ‘ARCONTROLLER_DICTIONARY_KEY_COMMON_MAVLINKSTATE_MISSIONITEMEXECUTED’
and
‘ARCONTROLLER_DICTIONARY_KEY_COMMON_MAVLINKSTATE_MISSONITEMEXECUTED_IDX' -> ‘ARCONTROLLER_DICTIONARY_KEY_COMMON_MAVLINKSTATE_MISSIONITEMEXECUTED_IDX

Very embrassing typo... 
Everything works fine after fix this.
Hope this fixes your problem!

Bye!

thomas-bamford, 2017-08-01:
Hello,
This is an api change from the Parrot sdk 3.10 to sdk 3.12. The new sdk (sdk 3.12) has been released on the parrot_arsdk package for all platforms except xenial, but the bebeop_autonomy branch adding support for sdk 3.12 has not yet been merged (see PR #120). Once this branch is merged this issue will be solved.

anderdefector, 2017-09-03:
Thank you so much @JeenQ  !

Ok @thomas-bamford , thank you. ! :)