AutonomyLab/bebop_autonomy #127: Error in runing the packages

hosh0425, 2017-07-28:
Hi 
for both node and nodelet I faced with this error:
[ERROR] [1501247262.506974734]: [Decoder] 17:37:42:506 | Decode:302 - Frame size changed to 856 x 480
how I can solve this?

thomas-bamford, 2017-07-28:
Hi
This is an error raised if the camera calibration file `bebop_driver/data/bebop1_camera_calib.yaml` or `bebop_driver/data/bebop2_camera_calib.yaml` does not match the video streamed from the vehicle. Specifically the image height and width. The support for SDK 3.12 will update the calibration for the default configurations, however if the video stream is important for your application it is recommended that you calibrate your vehicle's camera (see steps in ros [here).](http://wiki.ros.org/camera_calibration/Tutorials/MonocularCalibration)

hosh0425, 2017-07-28:
@thomas-bamford thanks for response.
sorry, before this error I'am facing with:
[ WARN] [1501253623.596181740]: [ARSTREAM2_Rtcp] 19:23:43:596 | ARSTREAM2_RTCP_Receiver_ProcessSenderReport:289 - Unexpected sender SSRC
[ERROR] [1501253623.596291003]: [ARSTREAM2_Rtcp] 19:23:43:596 | ARSTREAM2_RTCP_Receiver_ProcessCompoundPacket:1492 - Failed to process sender report (-1)
[ WARN] [1501253623.596455953]: [ARSTREAM2_Rtcp] 19:23:43:596 | ARSTREAM2_RTCP_ProcessApplicationClockDelta:824 - Unexpected peer SSRC
[ERROR] [1501253623.596509164]: [ARSTREAM2_Rtcp] 19:23:43:596 | ARSTREAM2_RTCP_Receiver_ProcessCompoundPacket:1519 - Failed to process application clock delta (-1)
what's this?

hosh0425, 2017-07-28:
I stream the video and assured that the size of image is 856*480 and changed the calibration width and height to 856*480 but the problem not solved

thomas-bamford, 2017-08-12:
This may be an issue with the default decoder settings. I have found that this error is not critical (does not effect video streaming) and, as it states, the frame size is adjusted appropriately. Have you found that there are any issues streaming video?

Once we have released the new SDK we will look to fix this.

thomas-bamford, 2017-08-12:
The other error that you have appears to be raised by the Parrot SDK. You may be able to find more information on the Parrot developers forum.

LArayane, 2019-09-01:
I need the calibration matrix K to convert pixels to meters. I have calibrated the camera and  actually I got the same values as in the default file but even so I am still getting the same error 

> [ERROR] [1567353696.489721846]: [Decoder] 18:01:36:489 | Decode:302 - Frame size changed to 856 x 480

and even if I change the depth and height and set them to a different values than 856 x 480 I get the same error with same values!!!
How can I solve this please I think the matrix K is critical in my work.  Thanks