AutonomyLab/bebop_autonomy #128: Can't install bebop_autonomy in Ubuntu 14.04

akhkh, 2017-08-01:
The error log is,

In file included from /home/junhyuck/bebop_ws/devel/.private/bebop_driver/include/bebop_driver/BebopArdrone3Config.h:16:0,
                 from /home/junhyuck/bebop_ws/src/bebop_autonomy/bebop_driver/src/bebop.cpp:43:
/opt/ros/indigo/include/dynamic_reconfigure/ConfigDescription.h:32:11: error: macro "max" requires 2 arguments, but only 1 given
     , max()
           ^
/opt/ros/indigo/include/dynamic_reconfigure/ConfigDescription.h:33:11: error: macro "min" requires 2 arguments, but only 1 given
     , min()
           ^
/opt/ros/indigo/include/dynamic_reconfigure/ConfigDescription.h:38:17: error: macro "max" requires 2 arguments, but only 1 given
     , max(_alloc)
                 ^
/opt/ros/indigo/include/dynamic_reconfigure/ConfigDescription.h:39:17: error: macro "min" requires 2 arguments, but only 1 given
     , min(_alloc)
                 ^
In file included from /opt/ros/indigo/include/json-c/json.h:19:0,
                 from /opt/ros/indigo/include/parrot_arsdk/libARDiscovery/ARDISCOVERY_Device.h:41,
                 from /opt/ros/indigo/include/parrot_arsdk/libARDiscovery/ARDiscovery.h:38,
                 from /home/junhyuck/bebop_ws/src/bebop_autonomy/bebop_driver/src/bebop.cpp:38:
/usr/include/c++/4.8/complex: In function ‘_Tp std::__complex_abs(const std::complex<_Tp>&)’:
/opt/ros/indigo/include/json-c/bits.h:20:18: error: expected unqualified-id before ‘(’ token
 #define max(a,b) ((a) > (b) ? (a) : (b))
                  ^
/usr/include/c++/4.8/bits/locale_facets_nonio.tcc: In member function ‘_InIter std::time_get<_CharT, _InIter>::_M_extract_name(std::time_get<_CharT, _InIter>::iter_type, std::time_get<_CharT, _InIter>::iter_type, int&, const _CharT**, std::size_t, std::ios_base&, std::ios_base::iostate&) const’:
/opt/ros/indigo/include/json-c/bits.h:16:18: error: expected unqualified-id before ‘(’ token
 #define min(a,b) ((a) < (b) ? (a) : (b))
                  ^
In file included from /home/junhyuck/bebop_ws/devel/.private/bebop_driver/include/bebop_driver/BebopArdrone3Config.h:16:0,
                 from /home/junhyuck/bebop_ws/src/bebop_autonomy/bebop_driver/src/bebop.cpp:43:
/opt/ros/indigo/include/dynamic_reconfigure/ConfigDescription.h: In constructor ‘dynamic_reconfigure::ConfigDescription_<ContainerAllocator>::ConfigDescription_()’:
/opt/ros/indigo/include/dynamic_reconfigure/ConfigDescription.h:33:5: error: expected ‘(’ before ‘,’ token
     , min()
     ^
/opt/ros/indigo/include/dynamic_reconfigure/ConfigDescription.h:34:5: error: expected ‘(’ before ‘,’ token
     , dflt()  {
     ^
/opt/ros/indigo/include/dynamic_reconfigure/ConfigDescription.h: In constructor ‘dynamic_reconfigure::ConfigDescription_<ContainerAllocator>::ConfigDescription_(const ContainerAllocator&)’:
/opt/ros/indigo/include/dynamic_reconfigure/ConfigDescription.h:39:5: error: expected ‘(’ before ‘,’ token
     , min(_alloc)
     ^
/opt/ros/indigo/include/dynamic_reconfigure/ConfigDescription.h:40:5: error: expected ‘(’ before ‘,’ token
     , dflt(_alloc)  {
     ^
make[2]: *** [CMakeFiles/bebop.dir/src/bebop.cpp.o] error 1
make[1]: *** [CMakeFiles/bebop.dir/all] error 2
make: *** [all] error 2

---------------------------------------------------------------------------------------------------------------------------------

I fixed the mistypes but another error occurred.
How can I install it?

Actually It was successfully installed 1 week ago, but not now.

thomas-bamford, 2017-08-01:
Hi,
The parrot_arsdk package has been updated this week but the branch adding support has not yet been merged (see pr #120). To correct these errors temporarily, please check out the sdk-3-12 branch.

Debjoy10, 2019-05-23:
Any updates on this? Facing same problem.