AutonomyLab/bebop_autonomy #129: looks like some bugs about moving the virtual camera, include TF

danquxunhuantingge, 2017-08-03:
1. TF and joint states 
After start the bebop_nodelet.launch, use rqt_tf_tree tool, seems work fine.
use rostopic echo /bebop/joint_states seems get the correct camera Pan&Tilt states.
and I try to move the virtual camera, for example, 
```
rostopic pub /bebop/camera_control geometry_msgs/Twist "linear:
  x: 0.0
  y: 0.0
  z: 0.0
angular:
  x: 0.0
  y: -10.0
  z: 0.0"
```
The camera view is moving, but use rostopic echo /bebop/joint_states can't get any messages any more, and rqt_tf_tree tools shows bellow,

It seems that the connection is interrupted.
2. how can I set absolute tilt and pan of the camera? After read the docs, I tried to let the camera watch directly below use set the angular.y to a small negative number such as -90, but it seems doesn't move 90 degrees. How can I do that?

danquxunhuantingge, 2017-08-03:
I have tried indigo-devel and sdk-3-12 two branches of this package, it seems that this problem still exist.

Ignacio1994, 2017-10-30:
I have the same problem. Would you fix it? Is any other topic where I can get the camera angles?
Thank you.

itssme, 2018-10-20:
I forked the repository and fixed the bug for the beobp2: https://github.com/itssme/bebop_autonomy

gsilano, 2018-10-20:
@itssme, great!! Why you didn't open a pull request? It would be nice to share your results with others.

itssme, 2018-10-20:
@gsilano Because I am not sure if the package would still work with the bebop 1. I only have a bebop2 to test :/

gsilano, 2018-10-20:
@itssme, don't worry. I think the maintainer or contributors have a Bebop 1 to test the code.

itssme, 2018-10-21:
@gsilano Ok, I made the pull request: https://github.com/AutonomyLab/bebop_autonomy/pull/178

LArayane, 2019-08-22:
@itssme @gsilano  why it is called virtual since there is a camera there!!! I don't get it