AutonomyLab/bebop_autonomy #130: ERROR in roslaunch bebop_driver bebop_node.launch

dcsgman, 2017-08-03:
When the computer connects to the bebop by wifi and roslaunch bebop_driver bebop_node.launch, The following error occurred：
......
[ INFO] [1501751964.009304462]: [ARCONTROLLER_Network] 17:19:24:009 | ARCONTROLLER_Network_GetAvailableSocketPort:128 - d2c_port port: 43210
[ INFO] [1501751964.009713925]: [ARCONTROLLER_Stream2] 17:19:24:009 | ARCONTROLLER_Stream2_Open_Socket:143 - udp local port stream: 55004
[ INFO] [1501751964.009789257]: [ARCONTROLLER_Stream2] 17:19:24:009 | ARCONTROLLER_Stream2_Open_Socket:143 - udp local port control: 55005
[ERROR] [1501751964.208893973]: [ARNETWORK_Sender] 17:19:24:208 | ARNETWORK_Sender_ProcessBufferToSend:405 - [0x7fa014273260] Timeout waiting for ack in buffer 11
[ERROR] [1501751964.209086805]: [ARNETWORK_Sender] 17:19:24:209 | ARNETWORK_Sender_ProcessBufferToSend:421 - [0x7fa014273260] Will retry sending data of buffer 11
[ERROR] [1501751964.360609586]: [ARNETWORK_Sender] 17:19:24:360 | ARNETWORK_Sender_ProcessBufferToSend:405 - [0x7fa014273260] Timeout waiting for ack in buffer 11
[ERROR] [1501751964.360814939]: [ARNETWORK_Sender] 17:19:24:360 | ARNETWORK_Sender_ProcessBufferToSend:421 - [0x7fa014273260] Will retry sending data of buffer 11
[ERROR] [1501751964.512071860]: [ARNETWORK_Sender] 17:19:24:511 | 
......
Begging for your help，Thank you!

thomas-bamford, 2017-08-12:
Hi,
I have managed to run the bebop autonomy package using VMware. An additional step is to configure the network adapter to "bridged" with same physical address since the default (NAT) does not allow the driver to connect directly to the Bebop. Please let us know if there are any other issues.

dcsgman, 2017-08-12:
Thank you, I have resolved this error by using bridged.

Kornelia11101, 2018-07-16:
How did you fix it? :)

Kornelia11101, 2018-07-18:
Would it be a problem for you to write step by step how to use (in this case) "bridged" network connection? And how to connect it with Sphinx, Gazebo and Bebop2 all together? 
Thank you!