AutonomyLab/bebop_autonomy #131: Error to building bebop_driver 8/8/2017

jokky009, 2017-08-08:
hi, I have a problem when I use catkin build.I use Ubuntu 14.04LTS and ROS indigo.
thanks.
project@project-X550JX:~/bebop_ws$ catkin build
-----------------------------------------------------------
Profile:                     default
Extending:          [cached] /opt/ros/indigo
Workspace:                   /home/project/bebop_ws
-----------------------------------------------------------
Source Space:       [exists] /home/project/bebop_ws/src
Log Space:          [exists] /home/project/bebop_ws/logs
Build Space:        [exists] /home/project/bebop_ws/build
Devel Space:        [exists] /home/project/bebop_ws/devel
Install Space:      [unused] /home/project/bebop_ws/install
DESTDIR:            [unused] None
-----------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
-----------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
-----------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
-----------------------------------------------------------
Workspace configuration appears valid.
-----------------------------------------------------------
[build] Found '5' packages in 0.0 seconds.                                                                                                     
[build] Package table is up to date.                                                                                                           
Starting  >>> bebop_description                                                                                                                
Starting  >>> bebop_msgs                                                                                                                       
Finished  <<< bebop_description                [ 0.1 seconds ]                                                                                 
Finished  <<< bebop_msgs                       [ 0.7 seconds ]                                                                                 
Starting  >>> bebop_driver                                                                                                                     
_______________________________________________________________________________________________________________________________________________
Errors     << bebop_driver:make /home/project/bebop_ws/logs/bebop_driver/build.make.001.log                                                    
In file included from /home/project/bebop_ws/src/bebop_autonomy/bebop_driver/src/bebop.cpp:48:0:
/home/project/bebop_ws/src/bebop_autonomy/bebop_driver/include/bebop_driver/autogenerated/common_state_callbacks.h: In constructor ‘bebop_driver::cb::CommonMavlinkStateMissonItemExecuted::CommonMavlinkStateMissonItemExecuted(ros::NodeHandle&, ros::NodeHandle&, const string&)’:
/home/project/bebop_ws/src/bebop_autonomy/bebop_driver/include/bebop_driver/autogenerated/common_state_callbacks.h:1230:21: error: ‘ARCONTROLLER_DICTIONARY_KEY_COMMON_MAVLINKSTATE_MISSONITEMEXECUTED’ was not declared in this scope
     : AbstractState(ARCONTROLLER_DICTIONARY_KEY_COMMON_MAVLINKSTATE_MISSONITEMEXECUTED)
                     ^
In file included from /opt/ros/indigo/include/parrot_arsdk/libARController/ARCONTROLLER_Dictionary.h:42:0,
                 from /opt/ros/indigo/include/parrot_arsdk/libARController/ARController.h:45,
                 from /home/project/bebop_ws/src/bebop_autonomy/bebop_driver/src/bebop.cpp:39:
/home/project/bebop_ws/src/bebop_autonomy/bebop_driver/include/bebop_driver/autogenerated/common_state_callbacks.h: In member function ‘virtual void bebop_driver::cb::CommonMavlinkStateMissonItemExecuted::Update(const ARCONTROLLER_DICTIONARY_ARG_t*, const ros::Time&)’:
/home/project/bebop_ws/src/bebop_autonomy/bebop_driver/include/bebop_driver/autogenerated/common_state_callbacks.h:1260:31: error: ‘ARCONTROLLER_DICTIONARY_KEY_COMMON_MAVLINKSTATE_MISSONITEMEXECUTED_IDX’ was not declared in this scope
     HASH_FIND_STR (arguments, ARCONTROLLER_DICTIONARY_KEY_COMMON_MAVLINKSTATE_MISSONITEMEXECUTED_IDX, arg);
                               ^
/opt/ros/indigo/include/parrot_arsdk/uthash/uthash.h:430:43: note: in definition of macro ‘HASH_JEN’
   unsigned char *_hj_key=(unsigned char*)(key);                                  \
                                           ^
/opt/ros/indigo/include/parrot_arsdk/uthash/uthash.h:262:5: note: in expansion of macro ‘HASH_FIND’
     HASH_FIND(hh,head,findstr,(unsigned)strlen(findstr),out)
     ^
/home/project/bebop_ws/src/bebop_autonomy/bebop_driver/include/bebop_driver/autogenerated/common_state_callbacks.h:1260:5: note: in expansion of macro ‘HASH_FIND_STR’
     HASH_FIND_STR (arguments, ARCONTROLLER_DICTIONARY_KEY_COMMON_MAVLINKSTATE_MISSONITEMEXECUTED_IDX, arg);
     ^
/home/project/bebop_ws/src/bebop_autonomy/bebop_driver/include/bebop_driver/autogenerated/common_state_callbacks.h:1260:31: error: ‘ARCONTROLLER_DICTIONARY_KEY_COMMON_MAVLINKSTATE_MISSONITEMEXECUTED_IDX’ was not declared in this scope
     HASH_FIND_STR (arguments, ARCONTROLLER_DICTIONARY_KEY_COMMON_MAVLINKSTATE_MISSONITEMEXECUTED_IDX, arg);
                               ^
/opt/ros/indigo/include/parrot_arsdk/uthash/uthash.h:613:29: note: in definition of macro ‘HASH_FIND_IN_BKT’
     if ((out)->hh.keylen == keylen_in) {                                           \
                             ^
/opt/ros/indigo/include/parrot_arsdk/uthash/uthash.h:262:5: note: in expansion of macro ‘HASH_FIND’
     HASH_FIND(hh,head,findstr,(unsigned)strlen(findstr),out)
     ^
/home/project/bebop_ws/src/bebop_autonomy/bebop_driver/include/bebop_driver/autogenerated/common_state_callbacks.h:1260:5: note: in expansion of macro ‘HASH_FIND_STR’
     HASH_FIND_STR (arguments, ARCONTROLLER_DICTIONARY_KEY_COMMON_MAVLINKSTATE_MISSONITEMEXECUTED_IDX, arg);
     ^
make[2]: *** [CMakeFiles/bebop.dir/src/bebop.cpp.o] Error 1
make[1]: *** [CMakeFiles/bebop.dir/all] Error 2
make: *** [all] Error 2
cd /home/project/bebop_ws/build/bebop_driver; catkin build --get-env bebop_driver | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
...............................................................................................................................................
Failed     << bebop_driver:make                [ Exited with code 2 ]                                                                          
Failed    <<< bebop_driver                     [ 3.5 seconds ]                                                                                 
Abandoned <<< bebop_tools                      [ Unrelated job failed ]                                                                        
[build] Summary: 2 of 4 packages succeeded.                                                                                                    
[build]   Ignored:   1 packages were skipped or are blacklisted.                                                                               
[build]   Warnings:  None.                                                                                                                     
[build]   Abandoned: 1 packages were abandoned.                                                                                                
[build]   Failed:    1 packages failed.                                                                                                        
[build] Runtime: 4.3 seconds total.

thomas-bamford, 2017-08-12:
Hi,
The parrot_arsdk package was updated a few weeks ago but the branch adding support has not yet been merged (see pr #120 ). To correct these errors temporarily, please check out the sdk-3-12 branch and follow instructions for installation in [here](https://github.com/AutonomyLab/bebop_autonomy/blob/8392f055b7ef0205b357f0c2d4088389910a1151/docs/installation.rst).

jokky009, 2017-08-19:
@thomas-bamford  I tired it already and it can work. Thank you very much for your help.