AutonomyLab/bebop_autonomy #132: Ubuntu 16.04 catkin build failed

arnabsinha, 2017-08-10:
Starting  >>> bebop_driver                                                                                                                                                                                 
___________________________________________________________________________________________________________________________________________________________________________________________________________
Errors     << bebop_driver:make /home/arnab/codes/bebop_workspace/logs/bebop_driver/build.make.004.log                                                                                                     
/home/arnab/codes/bebop_workspace/src/bebop_autonomy/bebop_driver/src/bebop_video_decoder.cpp: In member function 'bool bebop_driver::VideoDecoder::ReallocateBuffers()':
/home/arnab/codes/bebop_workspace/src/bebop_autonomy/bebop_driver/src/bebop_video_decoder.cpp:136:32: warning: 'int avpicture_get_size(AVPixelFormat, int, int)' is deprecated [-Wdeprecated-declarations]
     const uint32_t num_bytes = avpicture_get_size(PIX_FMT_RGB24, codec_ctx_ptr_->width, codec_ctx_ptr_->width);
                                ^
In file included from /home/arnab/codes/bebop_workspace/src/bebop_autonomy/bebop_driver/include/bebop_driver/bebop_video_decoder.h:32:0,
                 from /home/arnab/codes/bebop_workspace/src/bebop_autonomy/bebop_driver/src/bebop_video_decoder.cpp:25:
/usr/include/x86_64-linux-gnu/libavcodec/avcodec.h:5561:5: note: declared here
 int avpicture_get_size(enum AVPixelFormat pix_fmt, int width, int height);
     ^
/home/arnab/codes/bebop_workspace/src/bebop_autonomy/bebop_driver/src/bebop_video_decoder.cpp:136:32: warning: 'int avpicture_get_size(AVPixelFormat, int, int)' is deprecated [-Wdeprecated-declarations]
     const uint32_t num_bytes = avpicture_get_size(PIX_FMT_RGB24, codec_ctx_ptr_->width, codec_ctx_ptr_->width);
                                ^
In file included from /home/arnab/codes/bebop_workspace/src/bebop_autonomy/bebop_driver/include/bebop_driver/bebop_video_decoder.h:32:0,
                 from /home/arnab/codes/bebop_workspace/src/bebop_autonomy/bebop_driver/src/bebop_video_decoder.cpp:25:
/usr/include/x86_64-linux-gnu/libavcodec/avcodec.h:5561:5: note: declared here
 int avpicture_get_size(enum AVPixelFormat pix_fmt, int width, int height);
     ^
/home/arnab/codes/bebop_workspace/src/bebop_autonomy/bebop_driver/src/bebop_video_decoder.cpp:136:51: error: 'PIX_FMT_RGB24' was not declared in this scope
     const uint32_t num_bytes = avpicture_get_size(PIX_FMT_RGB24, codec_ctx_ptr_->width, codec_ctx_ptr_->width);
                                                   ^
/home/arnab/codes/bebop_workspace/src/bebop_autonomy/bebop_driver/src/bebop_video_decoder.cpp:145:27: warning: 'int avpicture_fill(AVPicture*, const uint8_t*, AVPixelFormat, int, int)' is deprecated [-Wdeprecated-declarations]
     ThrowOnCondition(0 == avpicture_fill(
                           ^
In file included from /home/arnab/codes/bebop_workspace/src/bebop_autonomy/bebop_driver/include/bebop_driver/bebop_video_decoder.h:32:0,
                 from /home/arnab/codes/bebop_workspace/src/bebop_autonomy/bebop_driver/src/bebop_video_decoder.cpp:25:
/usr/include/x86_64-linux-gnu/libavcodec/avcodec.h:5546:5: note: declared here
 int avpicture_fill(AVPicture *picture, const uint8_t *ptr,
     ^
/home/arnab/codes/bebop_workspace/src/bebop_autonomy/bebop_driver/src/bebop_video_decoder.cpp:145:27: warning: 'int avpicture_fill(AVPicture*, const uint8_t*, AVPixelFormat, int, int)' is deprecated [-Wdeprecated-declarations]
     ThrowOnCondition(0 == avpicture_fill(
                           ^
In file included from /home/arnab/codes/bebop_workspace/src/bebop_autonomy/bebop_driver/include/bebop_driver/bebop_video_decoder.h:32:0,
                 from /home/arnab/codes/bebop_workspace/src/bebop_autonomy/bebop_driver/src/bebop_video_decoder.cpp:25:
/usr/include/x86_64-linux-gnu/libavcodec/avcodec.h:5546:5: note: declared here
 int avpicture_fill(AVPicture *picture, const uint8_t *ptr,
     ^
/home/arnab/codes/bebop_workspace/src/bebop_autonomy/bebop_driver/src/bebop_video_decoder.cpp: In member function 'bool bebop_driver::VideoDecoder::Decode(const ARCONTROLLER_Frame_t*)':
/home/arnab/codes/bebop_workspace/src/bebop_autonomy/bebop_driver/src/bebop_video_decoder.cpp:267:25: warning: 'int avcodec_decode_video2(AVCodecContext*, AVFrame*, int*, const AVPacket*)' is deprecated [-Wdeprecated-declarations]
     const int32_t len = avcodec_decode_video2(codec_ctx_ptr_, frame_ptr_, &frame_finished, &packet_);
                         ^
In file included from /home/arnab/codes/bebop_workspace/src/bebop_autonomy/bebop_driver/include/bebop_driver/bebop_video_decoder.h:32:0,
                 from /home/arnab/codes/bebop_workspace/src/bebop_autonomy/bebop_driver/src/bebop_video_decoder.cpp:25:
/usr/include/x86_64-linux-gnu/libavcodec/avcodec.h:4901:5: note: declared here
 int avcodec_decode_video2(AVCodecContext *avctx, AVFrame *picture,
     ^
/home/arnab/codes/bebop_workspace/src/bebop_autonomy/bebop_driver/src/bebop_video_decoder.cpp:267:25: warning: 'int avcodec_decode_video2(AVCodecContext*, AVFrame*, int*, const AVPacket*)' is deprecated [-Wdeprecated-declarations]
     const int32_t len = avcodec_decode_video2(codec_ctx_ptr_, frame_ptr_, &frame_finished, &packet_);
                         ^
In file included from /home/arnab/codes/bebop_workspace/src/bebop_autonomy/bebop_driver/include/bebop_driver/bebop_video_decoder.h:32:0,
                 from /home/arnab/codes/bebop_workspace/src/bebop_autonomy/bebop_driver/src/bebop_video_decoder.cpp:25:
/usr/include/x86_64-linux-gnu/libavcodec/avcodec.h:4901:5: note: declared here
 int avcodec_decode_video2(AVCodecContext *avctx, AVFrame *picture,
     ^
/home/arnab/codes/bebop_workspace/src/bebop_autonomy/bebop_driver/src/bebop_video_decoder.cpp:267:100: warning: 'int avcodec_decode_video2(AVCodecContext*, AVFrame*, int*, const AVPacket*)' is deprecated [-Wdeprecated-declarations]
     const int32_t len = avcodec_decode_video2(codec_ctx_ptr_, frame_ptr_, &frame_finished, &packet_);
                                                                                                    ^
In file included from /home/arnab/codes/bebop_workspace/src/bebop_autonomy/bebop_driver/include/bebop_driver/bebop_video_decoder.h:32:0,
                 from /home/arnab/codes/bebop_workspace/src/bebop_autonomy/bebop_driver/src/bebop_video_decoder.cpp:25:
/usr/include/x86_64-linux-gnu/libavcodec/avcodec.h:4901:5: note: declared here
 int avcodec_decode_video2(AVCodecContext *avctx, AVFrame *picture,
     ^
/home/arnab/codes/bebop_workspace/src/bebop_autonomy/bebop_driver/src/bebop_video_decoder.cpp:295:25: warning: 'int avcodec_decode_video2(AVCodecContext*, AVFrame*, int*, const AVPacket*)' is deprecated [-Wdeprecated-declarations]
     const int32_t len = avcodec_decode_video2(codec_ctx_ptr_, frame_ptr_, &frame_finished, &packet_);
                         ^
In file included from /home/arnab/codes/bebop_workspace/src/bebop_autonomy/bebop_driver/include/bebop_driver/bebop_video_decoder.h:32:0,
                 from /home/arnab/codes/bebop_workspace/src/bebop_autonomy/bebop_driver/src/bebop_video_decoder.cpp:25:
/usr/include/x86_64-linux-gnu/libavcodec/avcodec.h:4901:5: note: declared here
[... 7 lines cut ...]
/usr/include/x86_64-linux-gnu/libavcodec/avcodec.h:4901:5: note: declared here
 int avcodec_decode_video2(AVCodecContext *avctx, AVFrame *picture,
     ^
/home/arnab/codes/bebop_workspace/src/bebop_autonomy/bebop_driver/src/bebop_video_decoder.cpp:295:100: warning: 'int avcodec_decode_video2(AVCodecContext*, AVFrame*, int*, const AVPacket*)' is deprecated [-Wdeprecated-declarations]
     const int32_t len = avcodec_decode_video2(codec_ctx_ptr_, frame_ptr_, &frame_finished, &packet_);
                                                                                                    ^
In file included from /home/arnab/codes/bebop_workspace/src/bebop_autonomy/bebop_driver/include/bebop_driver/bebop_video_decoder.h:32:0,
                 from /home/arnab/codes/bebop_workspace/src/bebop_autonomy/bebop_driver/src/bebop_video_decoder.cpp:25:
/usr/include/x86_64-linux-gnu/libavcodec/avcodec.h:4901:5: note: declared here
 int avcodec_decode_video2(AVCodecContext *avctx, AVFrame *picture,
     ^
make[2]: *** [CMakeFiles/bebop.dir/src/bebop_video_decoder.cpp.o] Error 1
make[1]: *** [CMakeFiles/bebop.dir/all] Error 2
make: *** [all] Error 2
[... cut ...]
[build]   Warnings:  None.                                                                                                                                                                                 
[build]   Abandoned: 1 packages were abandoned.                                                                                                                                                            
[build]   Failed:    1 packages failed.                                                                                                                                                                    
[build] Runtime: 1.8 seconds total.

thomas-bamford, 2017-08-12:
Hi,
The parrot_arsdk package was updated a few weeks ago but the branch adding support has not yet been merged (see pr #120 ). I'm not sure entirely if this is what has caused these errors, but to correct these errors temporarily, please check out the sdk-3-12 branch and follow instructions for installation in [here](https://github.com/AutonomyLab/bebop_autonomy/blob/8392f055b7ef0205b357f0c2d4088389910a1151/docs/installation.rst). If this does not work, another issue that might cause this is not installed all dependencies according to the installation instructions.

Cullen-Dubbs, 2017-10-27:
HI there I am getting a similar issue on ubuntu 14.04 I understand i need to update or upgrade something from this link https://github.com/AutonomyLab/bebop_autonomy.git src/bebop_autonomy given on the ROS bebop_autonomy installation guide. 

I am very confused on how to do this and was wondering if there was somewhere that expalins how to do this?

thomas-bamford, 2017-10-30:
Hello,
Please try the following: First, make sure your packages are up to date
`sudo apt-get update`
Then pull the latest version of the github repository for bebop_autonomy:
`cd bebop_ws/src/bebop_autonomy`
`git pull`
Then you can rebuild the catkin workspace and it should be up to date
`cd ../..`
`catkin clean`
`catkin build`
Please let me know if this works for you.

Cullen-Dubbs, 2017-10-31:
That didnt seem to work :(

Summary of what i'm seeing after i update:

-When it is building, bebop_msg and bebop_description build just fine.W hen bebop_driver gets to 100% it exits the build. Then gives me the errors seen in the original post.

Now im not sure but this may be related to the branch im using?

I am not too familiar with using different branches. it is mentioned around the fourm to use the sdk-3-12 branch and too my knowledge the way i acquire this is 

'git clone <bebop_autonomy github link.git> -b sdk-3-12 --single-branch'

is this correct? I am also unfamiliar with git pull and how to use this as well.
 also would not having the parrot arsdk affect the catkin build of bebop driver?

Sorry for all the questions on the sub posts...

thomas-bamford, 2017-11-05:
You have to use the latest version of the parrot-arsdk package, the changes have been merged. This would be the 'indigo-devel' branch of that repository.

Not having the latest parrot-arsdk package will affect the build of the bebop driver since the versions have to match to build properly. Info on the pull function of git: https://git-scm.com/docs/git-pull

There seems to be two errors, the first is an unsuccessful build, and the second is the workspace is not sourced. This is indicated by the ros system not being able to find the node. To source the workspace after the build:
`source ~/bebop_ws/devel/setup.bash`

Cullen-Dubbs, 2017-11-14:
Hi there again, I still can't seem to get catkin to build to be a success. I am reinstalling ubuntu 14.04LTS with ROS indigo using the ISO provided [here](http://wiki.ros.org/turtlebot/Tutorials/indigo/ISO%20Installation). I know it is a Turtlebot iso but i dont think that should matter much? Just want a clean start...

I am using the [installation instructions ](https://github.com/AutonomyLab/bebop_autonomy/blob/8392f055b7ef0205b357f0c2d4088389910a1151/docs/installation.rst)mentioned earlier. 

I am making sure that the default [parrot-arsdk](https://github.com/AutonomyLab/parrot_arsdk) package and the [bebop_autonomy](https://github.com/AutonomyLab/bebop_autonomy) package are both the most up-to-date branch which both appear to be indigo-devel.

This may seem redundant but below is every command i will put in immediate after the OS is installed:

`sudo apt-get install build-essential python-rosdep python-catkin-tools`

when i run this i get an error 404 not found, failed to fetch python-catkin-tools
suggests i run apt-get update --fix-missing

`sudo apt-get update --fix-missing`

only fails to fetch 2 things that dont seem relevant

`mkdir -p ~/bebop_ws/src && cd ~/bebop_ws`

`catkin init`

got an error catkin command not found, i forgot i needed to install catkin since this is a new system

`sudo apt-get install ros-indigo-catkin
`
redoing:
`sudo apt-get install build-essential python-rosdep python-catkin-tools`
no errors this time

`sudo apt-get update`
similar failed fetching as earlier, probably still irrelavant

`cd ~/bebop_ws/`

`catkin init`
workspace appears valid
says missing log space build space and devel space, assuming thats normal

```
git clone https://github.com/AutonomyLab/bebop_autonomy.git src/bebop_autonomy
git clone https://github.com/AutonomyLab/parrot_arsdk.git src/parrot_arsdk
```
no issues there, my assumption is that both repos pulled are the indigo-devel branch

from `~/bebop_ws directory` i run

`rosdep update`

it says i need to run `sudo rosdep ini`t first

rerun `rosdep update`

`rosdep install --from-paths src -i`

all required rosdeps installed successfully

moment of truth, fingers crossed, running this from ~/bebop_ws directory,

`catkin build`

build and devl space now exist, logs still miss
this is the live build order as they finish

prebuild finishes
bebop_description finishes
bebop_msgs finishes
parrot_arsdk finishes after about 5 minutes. i get a few warnings as well

then bebop_driver starts and after about 40 seconds i see that it gets to 100% and get the erros listed in the initial issue.

bebop tools is obviously abandoned. 

phew. ok so this is probably redundant but if someone could point out anything i might have done wrong here or something extra i need please let me know.

Thank you so much!

thomas-bamford, 2017-11-19:
I installed a fresh ubuntu 16.04 lts image and successfully built and ran the bebop-autonomy package. Step-by-step instructions after installing a fresh 16.04 lts:

`sudo apt-get update`
`sudo apt-get upgrade`

Install ros-kinetic
`sudo sh -c 'echo "deb http://packages.ros.org/ros/ubuntu $(lsb_release -sc) main" > /etc/apt/sources.list.d/ros-latest.list'
`sudo apt-key adv --keyserver hkp://ha.pool.sks-keyservers.net:80 --recv-key 421C365BD9FF1F717815A3895523BAEEB01FA116`
`sudo apt-get update`
`sudo apt-get install ros-kinetic-desktop-full`
`sudo rosdep init`
`rosdep update`
`echo "source /opt/ros/kinetic/setup.bash" >> ~/.bashrc`
`source ~/.bashrc`

Follow instructions from [http://bebop-autonomy.readthedocs.io/en/latest/installation.html](url)
`sudo apt-get install build-essential python-rosdep python-catkin-tools`
`mkdir -p ~/bebop_ws/src && cd ~/bebop_ws`
`catkin init`

Install git because it is not included in the 16.04 lts image.
`sudo apt-get install git`

`git clone https://github.com/AutonomyLab/bebop_autonomy.git src/bebop_autonomy`
`rosdep update`
`rosdep install --from-paths src -i`

Make sure all dependencies are updated
`sudo apt-get update`
`sudo apt-get  upgrade`

Build
`catkin build`

Source packages and make sure they will be sourced after new terminal is open
`cd`
`source ~/bebop_ws/devel/setup.bash`
`echo "source ~/bebop_ws/devel/setup.bash" >> ~/.bashrc`

Test running the image view node, for some reason the image view node is not working well on 16.04 (unrelated to bebop-autonomy, can view stream using rqt)
`roslaunch bebop_tools bebop_nodelet_iv.launch`

Test echo topic in new terminal (Ctrl + Alt + T)
`rostopic echo /bebop/states/ardrone3/PilotingState/AttitudeChanged`

arnabsinha, 2017-12-01:
I found the solution at the following link
https://stackoverflow.com/questions/38846405/ffmpeg-pix-fmt-bgr24-was-not-declared-in-this-scope

within the "bebop_autonomy/bebop_driver/src/bebop_video_decoder.cpp" code 
change "PIX_FMT_BGR24" to "AV_PIX_FMT_BGR24" .... and it builds successfully ....

closed 2017-12-01 by arnabsinha