AutonomyLab/bebop_autonomy #135: Errors with launching driver

IanMin1022, 2017-08-30:
When I launched roslaunch bebop_driver bebop_node.launch I got following errors. I guess I can just ignore the camera frame part, but I'm not sure about the upper ones. Plus I am not sure it's because of the frame but it gives me warning saying "[ WARN] [1504055502.695132811]: [BebopSDK] 10:11:42:695 | FrameReceivedCallback:191 - Previous frame might have been missed."

(    ... logging to /home/nsm/.ros/log/66d5e7e0-8d25-11e7-9916-8c882b0019c6/roslaunch-Samsung-4567.log
    Checking log directory for disk usage. This may take awhile.
    Press Ctrl-C to interrupt
    Done checking log file disk usage. Usage is <1GB.
    
    xacro: Traditional processing is deprecated. Switch to --inorder processing!
    To check for compatibility of your document, use option --check-order.
    For more infos, see http://wiki.ros.org/xacro#Processing_Order
    started roslaunch server http://localhost:33991/
    
    SUMMARY
    ========
    
    PARAMETERS
     * /bebop/bebop_driver/bebop_ip: 192.168.42.1
     * /bebop/bebop_driver/camera_info_url: package://bebop_d...
     * /bebop/bebop_driver/cmd_vel_timeout: 0.2
     * /bebop/bebop_driver/odom_frame_id: odom
     * /bebop/bebop_driver/publish_odom_tf: True
     * /bebop/bebop_driver/reset_settings: True
     * /bebop/bebop_driver/states/enable_altitudechanged: True
     * /bebop/bebop_driver/states/enable_autotakeoffmodechanged: True
     * /bebop/bebop_driver/states/enable_camerastate_orientation: True
     * /bebop/bebop_driver/states/enable_commonstate_batterystatechanged: True
     * /bebop/bebop_driver/states/enable_commonstate_wifisignalchanged: True
     * /bebop/bebop_driver/states/enable_controllerstate_ispilotingchanged: True
     * /bebop/bebop_driver/states/enable_flightplanstate_availabilitystatechanged: True
     * /bebop/bebop_driver/states/enable_flightplanstate_componentstatelistchanged: True
     * /bebop/bebop_driver/states/enable_gpsstate_numberofsatellitechanged: True
     * /bebop/bebop_driver/states/enable_mavlinkstate_mavlinkfileplayingstatechanged: True
     * /bebop/bebop_driver/states/enable_mavlinkstate_mavlinkplayerrorstatechanged: True
     * /bebop/bebop_driver/states/enable_mediastreamingstate_videoenablechanged: True
     * /bebop/bebop_driver/states/enable_numberofsatellitechanged: True
     * /bebop/bebop_driver/states/enable_overheatstate_overheatchanged: True
     * /bebop/bebop_driver/states/enable_pilotingstate_altitudechanged: True
     * /bebop/bebop_driver/states/enable_pilotingstate_attitudechanged: True
     * /bebop/bebop_driver/states/enable_pilotingstate_flattrimchanged: True
     * /bebop/bebop_driver/states/enable_pilotingstate_flyingstatechanged: True
     * /bebop/bebop_driver/states/enable_pilotingstate_navigatehomestatechanged: True
     * /bebop/bebop_driver/states/enable_pilotingstate_positionchanged: True
     * /bebop/bebop_driver/states/enable_pilotingstate_speedchanged: True
     * /bebop/robot_description: <?xml version="1....
     * /rosdistro: kinetic
     * /rosversion: 1.12.7
    
    NODES
      /bebop/
        bebop_driver (bebop_driver/bebop_driver_node)
        robot_state_publisher (robot_state_publisher/robot_state_publisher)
    
    auto-starting new master
    process[master]: started with pid [4581]
    ROS_MASTER_URI=http://localhost:11311
    
    setting /run_id to 66d5e7e0-8d25-11e7-9916-8c882b0019c6
    process[rosout-1]: started with pid [4594]
    started core service [/rosout]
    process[bebop/bebop_driver-2]: started with pid [4611]
    process[bebop/robot_state_publisher-3]: started with pid [4612]
    [ INFO] [1504057744.255694968]: Initializing nodelet with 4 worker threads.
    [ INFO] [1504057744.304123471]: [BebopSDK] 10:49:04:304 | Bebop:225 - Bebop Cnstr()
    [ INFO] [1504057744.304195070]: Nodelet Cstr
    [ INFO] [1504057744.309224942]: Connecting to Bebop ...
    [ INFO] [1504057744.311654596]: [CB] 10:49:04:311 | Ardrone3PilotingStateFlatTrimChanged:388 - [STATES] Enabling states/ardrone3/PilotingState/FlatTrimChanged
    [ INFO] [1504057744.312568331]: [CB] 10:49:04:312 | Ardrone3PilotingStateFlyingStateChanged:432 - [STATES] Enabling states/ardrone3/PilotingState/FlyingStateChanged
    [ INFO] [1504057744.313645615]: [CB] 10:49:04:313 | Ardrone3PilotingStateNavigateHomeStateChanged:534 - [STATES] Enabling states/ardrone3/PilotingState/NavigateHomeStateChanged
    [ INFO] [1504057744.314411774]: [CB] 10:49:04:314 | Ardrone3PilotingStatePositionChanged:592 - [STATES] Enabling states/ardrone3/PilotingState/PositionChanged
    [ INFO] [1504057744.315216680]: [CB] 10:49:04:315 | Ardrone3PilotingStateSpeedChanged:657 - [STATES] Enabling states/ardrone3/PilotingState/SpeedChanged
[... 130 lines cut ...]
    [ WARN] [1504057751.058787037]: [BebopSDK] 10:49:11:058 | StartStreaming:359 - Video streaming started ...
    [ INFO] [1504057751.059132093]: Nodelet lwp_id: 4611
    [ INFO] [1504057751.059283008]: bebop_driver nodelet loaded.
    [ INFO] [1504057751.059609413]: [CameraThread] thread lwp_id: 5051
    [ INFO] [1504057751.059879061]: [AuxThread] thread lwp_id: 5052
    [ INFO] [1504057751.074266301]: [ARSTREAM2_StreamReceiver] 10:49:11:074 | ARSTREAM2_StreamReceiver_StartAppOutput:1851 - App output is running
    [ INFO] [1504057751.074346998]: [ARSTREAM2_StreamReceiver] 10:49:11:074 | ARSTREAM2_StreamReceiver_StartAppOutput:1851 - App output is running
    [ INFO] [1504057751.074394093]: [ARSTREAM2_StreamReceiver] 10:49:11:074 | ARSTREAM2_StreamReceiver_StartAppOutput:1851 - App output is running
    [ WARN] [1504057751.091418305]: [ARSTREAM2_Rtcp] 10:49:11:091 | ARSTREAM2_RTCP_Receiver_ProcessSenderReport:290 - Unexpected sender SSRC
    [ERROR] [1504057751.091462340]: [ARSTREAM2_Rtcp] 10:49:11:091 | ARSTREAM2_RTCP_Receiver_ProcessCompoundPacket:1516 - Failed to process sender report (-1)
    [ WARN] [1504057751.091516979]: [ARSTREAM2_Rtcp] 10:49:11:091 | ARSTREAM2_RTCP_ProcessApplicationClockDelta:848 - Unexpected peer SSRC
    [ERROR] [1504057751.091536352]: [ARSTREAM2_Rtcp] 10:49:11:091 | ARSTREAM2_RTCP_Receiver_ProcessCompoundPacket:1543 - Failed to process application clock delta (-1)
    [ INFO] [1504057751.093510202]: [ARSTREAM2_H264Filter] 10:49:11:093 | ARSTREAM2_H264Filter_ProcessAu:577 - AU output cancelled (waitForSync)
    [ INFO] [1504057751.125437521]: [ARSTREAM2_H264Filter] 10:49:11:125 | ARSTREAM2_H264Filter_ProcessAu:577 - AU output cancelled (waitForSync)
    [ INFO] [1504057751.165005612]: [ARSTREAM2_H264Filter] 10:49:11:164 | ARSTREAM2_H264Filter_ProcessAu:577 - AU output cancelled (waitForSync)
    [ INFO] [1504057751.195123817]: [ARSTREAM2_H264Filter] 10:49:11:195 | ARSTREAM2_H264Filter_ProcessAu:577 - AU output cancelled (waitForSync)
    [ INFO] [1504057751.225540900]: [ARSTREAM2_H264Filter] 10:49:11:225 | ARSTREAM2_H264Filter_Sync:65 - SPS/PPS sync OK
    [ INFO] [1504057751.225575099]: [BebopSDK] 10:49:11:225 | DecoderConfigCallback:147 - H264 configuration packet received: #SPS: 27 #PPS: 9 (MP4? 0)
    [ INFO] [1504057751.225731137]: [ARSTREAM2_H264FilterError] 10:49:11:225 | ARSTREAM2_H264FilterError_GenerateGrayIdrFrame:101 - Gray I slice NALU output size: 1629
    [ INFO] [1504057751.225820319]: [BebopSDK] 10:49:11:225 | FrameReceivedCallback:174 - Frame Recv & Decode LWP id: 5055
    [ INFO] [1504057751.226525756]: [Decoder] 10:49:11:226 | InitCodec:117 - H264 Codec is partially initialized!
    [ INFO] [1504057751.226563390]: [Decoder] 10:49:11:226 | Decode:263 - Updating H264 codec parameters (Buffer Size: 36) ...
    [ERROR] [1504057751.227745899]: [Decoder] 10:49:11:227 | Decode:302 - Frame size changed to 856 x 480
    [ INFO] [1504057751.227773816]: [Decoder] 10:49:11:227 | ReallocateBuffers:123 - Buffer reallocation request
    [ INFO] [1504057751.229286624]: camera calibration URL: package://bebop_driver/data/bebop1_camera_calib.yaml)

hosh0425, 2017-09-04:
I have same issue. did you solve it?

IanMin1022, 2017-09-05:
@hosh0425  I guess those are ignorable errors. If you enter [ rospublic pub --once /bebop/takeoff std_msgs/Empty "{}" ](url), you will see your bebop flying

dnllp, 2017-10-07:
I have a similar problem, the dron works, but I can not see the camera. Where I can find firmware 3.3.0, my bebop1 has 4.0.6 and I can not see the camera image. On the site it says it only supports up to 3.3.0. Can you help me with that

thomas-bamford, 2017-10-30:
Hi @dnllp, the Bebop 1 has not been tested past 3.3.0 so the supported version hasn't been updated yet. Our lab just got access to the Bebop 1 so we will be able to test it and see if we can update the supported version. I am not sure why the image is not available, but I will try to replicate this issue on our hardware in the coming weeks to see if I can help.