AutonomyLab/bebop_autonomy #136: Running Algorithms Locally on Bebop Drone?

AJE-WSU, 2017-08-30:
Is it possible to upload algorithms to a Bebop Parrot 2 drone and have the drone operate independently of a central PC? My goal is to have one drone track a lead drone using visual object tracking without having to communicate with the lead drone or whatever is controlling the lead drone. I am working with ROS.

Thanks,
AJE

thomas-bamford, 2017-10-30:
I am not aware of any way of doing this right now on only the Bebop 2 without figuring out how to develop your own version of their firmware. However, the [Parrot SLAM dunk](https://www.parrot.com/uk/business-solutions/parrot-slamdunk#parrot-slamdunk) kit provides onboard ROS computing, so this may work.

NitinJSanket, 2019-06-24:
You can do it though [PaparazziUAV](https://wiki.paparazziuav.org/wiki/Main_Page)