AutonomyLab/bebop_autonomy #138: Bebop driver launch fails with "ARSTREAM2_RtpReceiver" error

nmarchais, 2017-09-05:
When i try to launch the driver it fails with this error :
- ARSTREAM2_RtpReceiver_ProcessRtp:1293 - ARSTREAM2_RTP_Receiver_PacketFifoAddFromMsgVec() failed (-3)

Here is the launch command :
- roslaunch bebop_driver bebop_node.launch

My environment is Ubuntu 16.04 and ros kinetic.
I did the installation following theses instructions : http://bebop-autonomy.readthedocs.io/en/latest/installation.html
Compilation succeed but driver package had some warning. There is already a ticket about it : https://github.com/AutonomyLab/bebop_autonomy/issues/72#issuecomment-259063473

My device is Bebop 1 with firmware 4.0.6. I also downgrade the firmware to 3.3.0 with no improvement and the error is the same.

I would be great if someone could help with this issue.

Here is the console output :

`Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: Traditional processing is deprecated. Switch to --inorder processing!
To check for compatibility of your document, use option --check-order.
For more infos, see http://wiki.ros.org/xacro#Processing_Order
started roslaunch server http://nicolas-VirtualBox:42215/

SUMMARY
========

PARAMETERS
 * /bebop/bebop_driver/bebop_ip: 192.168.42.1
 * /bebop/bebop_driver/camera_info_url: package://bebop_d...
 * /bebop/bebop_driver/cmd_vel_timeout: 0.2
 * /bebop/bebop_driver/odom_frame_id: odom
 * /bebop/bebop_driver/publish_odom_tf: True
 * /bebop/bebop_driver/reset_settings: True
 * /bebop/bebop_driver/states/enable_altitudechanged: True
 * /bebop/bebop_driver/states/enable_autotakeoffmodechanged: True
 * /bebop/bebop_driver/states/enable_camerastate_orientation: True
 * /bebop/bebop_driver/states/enable_commonstate_batterystatechanged: True
 * /bebop/bebop_driver/states/enable_commonstate_wifisignalchanged: True
 * /bebop/bebop_driver/states/enable_controllerstate_ispilotingchanged: True
 * /bebop/bebop_driver/states/enable_flightplanstate_availabilitystatechanged: True
 * /bebop/bebop_driver/states/enable_flightplanstate_componentstatelistchanged: True
 * /bebop/bebop_driver/states/enable_gpsstate_numberofsatellitechanged: True
 * /bebop/bebop_driver/states/enable_mavlinkstate_mavlinkfileplayingstatechanged: True
 * /bebop/bebop_driver/states/enable_mavlinkstate_mavlinkplayerrorstatechanged: True
 * /bebop/bebop_driver/states/enable_mediastreamingstate_videoenablechanged: True
 * /bebop/bebop_driver/states/enable_numberofsatellitechanged: True
 * /bebop/bebop_driver/states/enable_overheatstate_overheatchanged: True
 * /bebop/bebop_driver/states/enable_pilotingstate_altitudechanged: True
 * /bebop/bebop_driver/states/enable_pilotingstate_attitudechanged: True
 * /bebop/bebop_driver/states/enable_pilotingstate_flattrimchanged: True
 * /bebop/bebop_driver/states/enable_pilotingstate_flyingstatechanged: True
 * /bebop/bebop_driver/states/enable_pilotingstate_navigatehomestatechanged: True
 * /bebop/bebop_driver/states/enable_pilotingstate_positionchanged: True
 * /bebop/bebop_driver/states/enable_pilotingstate_speedchanged: True
 * /bebop/robot_description: <?xml version="1....
 * /rosdistro: kinetic
 * /rosversion: 1.12.7

NODES
  /bebop/
    bebop_driver (bebop_driver/bebop_driver_node)
    robot_state_publisher (robot_state_publisher/robot_state_publisher)

auto-starting new master
process[master]: started with pid [6203]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 43525e42-91f4-11e7-98ba-080027f630a0
[... 211 lines cut ...]
[ERROR] [1504586402.833387232]: [ARSTREAM2_RtpReceiver] 15:40:02:833 | ARSTREAM2_RtpReceiver_ProcessRtp:1293 - ARSTREAM2_RTP_Receiver_PacketFifoAddFromMsgVec() failed (-3)
[ INFO] [1504586402.833673679]: [ARSTREAM2_H264Filter] 15:40:02:833 | ARSTREAM2_H264Filter_ProcessAu:577 - AU output cancelled (waitForSync)
[ERROR] [1504586402.855844877]: [ARSTREAM2_RtpReceiver] 15:40:02:855 | ARSTREAM2_RtpReceiver_ProcessRtp:1293 - ARSTREAM2_RTP_Receiver_PacketFifoAddFromMsgVec() failed (-3)
[ INFO] [1504586402.856181090]: [ARSTREAM2_H264Filter] 15:40:02:856 | ARSTREAM2_H264Filter_ProcessAu:577 - AU output cancelled (waitForSync)
[ERROR] [1504586402.899557022]: [ARSTREAM2_RtpReceiver] 15:40:02:899 | ARSTREAM2_RtpReceiver_ProcessRtp:1293 - ARSTREAM2_RTP_Receiver_PacketFifoAddFromMsgVec() failed (-3)
[ INFO] [1504586402.899800210]: [ARSTREAM2_H264Filter] 15:40:02:899 | ARSTREAM2_H264Filter_ProcessAu:577 - AU output cancelled (waitForSync)
[ERROR] [1504586402.923844698]: [ARSTREAM2_RtpReceiver] 15:40:02:923 | ARSTREAM2_RtpReceiver_ProcessRtp:1293 - ARSTREAM2_RTP_Receiver_PacketFifoAddFromMsgVec() failed (-3)
[ERROR] [1504586402.924265172]: [ARSTREAM2_RtpReceiver] 15:40:02:924 | ARSTREAM2_RtpReceiver_ProcessRtp:1293 - ARSTREAM2_RTP_Receiver_PacketFifoAddFromMsgVec() failed (-3)
[ INFO] [1504586402.924438564]: [ARSTREAM2_H264Filter] 15:40:02:924 | ARSTREAM2_H264Filter_ProcessAu:577 - AU output cancelled (waitForSync)
[ERROR] [1504586402.954564493]: [ARSTREAM2_RtpReceiver] 15:40:02:954 | ARSTREAM2_RtpReceiver_ProcessRtp:1293 - ARSTREAM2_RTP_Receiver_PacketFifoAddFromMsgVec() failed (-3)
[ INFO] [1504586402.954702828]: [ARSTREAM2_H264Filter] 15:40:02:954 | ARSTREAM2_H264Filter_ProcessAu:577 - AU output cancelled (waitForSync)
[ERROR] [1504586403.004137688]: [ARSTREAM2_RtpReceiver] 15:40:03:004 | ARSTREAM2_RtpReceiver_ProcessRtp:1293 - ARSTREAM2_RTP_Receiver_PacketFifoAddFromMsgVec() failed (-3)
[ INFO] [1504586403.004316544]: [ARSTREAM2_H264Filter] 15:40:03:004 | ARSTREAM2_H264Filter_Sync:65 - SPS/PPS sync OK
[ INFO] [1504586403.004410839]: [BebopSDK] 15:40:03:004 | DecoderConfigCallback:147 - H264 configuration packet received: #SPS: 26 #PPS: 9 (MP4? 0)
[ INFO] [1504586403.004621983]: [ARSTREAM2_H264FilterError] 15:40:03:004 | ARSTREAM2_H264FilterError_GenerateGrayIdrFrame:101 - Gray I slice NALU output size: 929
[ INFO] [1504586403.004738547]: [BebopSDK] 15:40:03:004 | FrameReceivedCallback:174 - Frame Recv & Decode LWP id: 6670
[ INFO] [1504586403.005467955]: [Decoder] 15:40:03:005 | InitCodec:117 - H264 Codec is partially initialized!
[ INFO] [1504586403.005589865]: [Decoder] 15:40:03:005 | Decode:263 - Updating H264 codec parameters (Buffer Size: 35) ...
[ERROR] [1504586403.006589972]: [Decoder] 15:40:03:006 | Decode:302 - Frame size changed to 640 x 368
[ INFO] [1504586403.006719750]: [Decoder] 15:40:03:006 | ReallocateBuffers:123 - Buffer reallocation request
[ INFO] [1504586403.007912155]: camera calibration URL: package://bebop_driver/data/bebop1_camera_calib.yaml
[ERROR] [1504586403.038491980]: [ARSTREAM2_RtpReceiver] 15:40:03:038 | ARSTREAM2_RtpReceiver_ProcessRtp:1293 - ARSTREAM2_RTP_Receiver_PacketFifoAddFromMsgVec() failed (-3)
[ERROR] [1504586403.059148425]: [ARSTREAM2_RtpReceiver] 15:40:03:059 | ARSTREAM2_RtpReceiver_ProcessRtp:1293 - ARSTREAM2_RTP_Receiver_PacketFifoAddFromMsgVec() failed (-3)
[ERROR] [1504586403.059279443]: [ARSTREAM2_RtpReceiver] 15:40:03:059 | ARSTREAM2_RtpReceiver_ProcessRtp:1293 - ARSTREAM2_RTP_Receiver_PacketFifoAddFromMsgVec() failed (-3)
`

thomas-bamford, 2017-10-30:
This looks like an error in the Parrot SDK regarding network connection and video streaming. Our lab just got access to a Bebop 1, so I will see if I can replicate this issue and look for solutions.

cmdraghici, 2018-04-20:
Hi, I have exactly the same issue with the same environments. There are any solution for this today?

cmdraghici, 2018-09-06:
Hi, I fixed this issue, in my case the problem was that the firmware version of my drone was 4.4.3 and the latest tested for the package is 4.0.6, so I had to downgrade it. I used the version 4.0.3, downloaded from parrot community, because the Parrot web didn't have a version history and I didn't got that problem any more.

arnaldojr, 2020-11-13:
Hello, The same issue here. I fixed it only allow the ports 43210, 55005, 55004 of my firewall.

```
sudo ufw allow 43210
sudo ufw allow 55005
sudo ufw allow 55004
```