AutonomyLab/bebop_autonomy #142: Cannot Run Bebop_Driver (Code -11)

TOTON95, 2017-11-01:
I could run it without problems earlier, but when I got to continue the test, I got this sad result, I'm attaching my output. Does anyone knows how to solve it ? I am using a Bebop 1 

`totonzx@totonzx:~/bebop_ws$ roslaunch bebop_driver bebop_node.launch
... logging to /home/totonzx/.ros/log/5ff6f482-beda-11e7-ad5d-20689d850533/roslaunch-totonzx-18182.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: Traditional processing is deprecated. Switch to --inorder processing!
To check for compatibility of your document, use option --check-order.
For more infos, see http://wiki.ros.org/xacro#Processing_Order
started roslaunch server http://totonzx:41111/

SUMMARY
========

PARAMETERS
 * /bebop/bebop_driver/bebop_ip: 192.168.42.1
 * /bebop/bebop_driver/camera_info_url: package://bebop_d...
 * /bebop/bebop_driver/cmd_vel_timeout: 0.2
 * /bebop/bebop_driver/odom_frame_id: odom
 * /bebop/bebop_driver/publish_odom_tf: True
 * /bebop/bebop_driver/reset_settings: True
 * /bebop/bebop_driver/states/enable_altitudechanged: True
 * /bebop/bebop_driver/states/enable_autotakeoffmodechanged: True
 * /bebop/bebop_driver/states/enable_camerastate_orientation: True
 * /bebop/bebop_driver/states/enable_commonstate_batterystatechanged: True
 * /bebop/bebop_driver/states/enable_commonstate_wifisignalchanged: True
 * /bebop/bebop_driver/states/enable_controllerstate_ispilotingchanged: True
 * /bebop/bebop_driver/states/enable_flightplanstate_availabilitystatechanged: True
 * /bebop/bebop_driver/states/enable_flightplanstate_componentstatelistchanged: True
 * /bebop/bebop_driver/states/enable_gpsstate_numberofsatellitechanged: True
 * /bebop/bebop_driver/states/enable_mavlinkstate_mavlinkfileplayingstatechanged: True
 * /bebop/bebop_driver/states/enable_mavlinkstate_mavlinkplayerrorstatechanged: True
 * /bebop/bebop_driver/states/enable_mediastreamingstate_videoenablechanged: True
 * /bebop/bebop_driver/states/enable_numberofsatellitechanged: True
 * /bebop/bebop_driver/states/enable_overheatstate_overheatchanged: True
 * /bebop/bebop_driver/states/enable_pilotingstate_altitudechanged: True
 * /bebop/bebop_driver/states/enable_pilotingstate_attitudechanged: True
 * /bebop/bebop_driver/states/enable_pilotingstate_flattrimchanged: True
 * /bebop/bebop_driver/states/enable_pilotingstate_flyingstatechanged: True
 * /bebop/bebop_driver/states/enable_pilotingstate_navigatehomestatechanged: True
 * /bebop/bebop_driver/states/enable_pilotingstate_positionchanged: True
 * /bebop/bebop_driver/states/enable_pilotingstate_speedchanged: True
 * /bebop/robot_description: <?xml version="1....
 * /rosdistro: kinetic
 * /rosversion: 1.12.7

NODES
  /bebop/
    bebop_driver (bebop_driver/bebop_driver_node)
    robot_state_publisher (robot_state_publisher/robot_state_publisher)

auto-starting new master
process[master]: started with pid [18196]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 5ff6f482-beda-11e7-ad5d-20689d850533
process[rosout-1]: started with pid [18209]
started core service [/rosout]
process[bebop/bebop_driver-2]: started with pid [18226]
process[bebop/robot_state_publisher-3]: started with pid [18227]
[ INFO] [1509523079.104606977]: Initializing nodelet with 4 worker threads.
[ INFO] [1509523079.209727821]: [BebopSDK] 01:57:59:209 | Bebop:225 - Bebop Cnstr()
[ INFO] [1509523079.209786941]: Nodelet Cstr
[ INFO] [1509523079.229527584]: Connecting to Bebop ...
[ INFO] [1509523079.238822254]: [CB] 01:57:59:238 | Ardrone3PilotingStateFlatTrimChanged:388 - [STATES] Enabling states/ardrone3/PilotingState/FlatTrimChanged
[ INFO] [1509523079.243617567]: [CB] 01:57:59:243 | Ardrone3PilotingStateFlyingStateChanged:432 - [STATES] Enabling states/ardrone3/PilotingState/FlyingStateChanged
[ INFO] [1509523079.248799111]: [CB] 01:57:59:248 | Ardrone3PilotingStateNavigateHomeStateChanged:534 - [STATES] Enabling states/ardrone3/PilotingState/NavigateHomeStateChanged
[ INFO] [1509523079.251933856]: [CB] 01:57:59:251 | Ardrone3PilotingStatePositionChanged:592 - [STATES] Enabling states/ardrone3/PilotingState/PositionChanged
[ INFO] [1509523079.255235457]: [CB] 01:57:59:255 | Ardrone3PilotingStateSpeedChanged:657 - [STATES] Enabling states/ardrone3/PilotingState/SpeedChanged
[ INFO] [1509523079.257991711]: [CB] 01:57:59:257 | Ardrone3PilotingStateAttitudeChanged:722 - [STATES] Enabling states/ardrone3/PilotingState/AttitudeChanged
[ INFO] [1509523079.262638454]: [CB] 01:57:59:262 | Ardrone3PilotingStateAltitudeChanged:838 - [STATES] Enabling states/ardrone3/PilotingState/AltitudeChanged
[ INFO] [1509523079.279839823]: [CB] 01:57:59:279 | Ardrone3MediaStreamingStateVideoEnableChanged:1388 - [STATES] Enabling states/ardrone3/MediaStreamingState/VideoEnableChanged
[ INFO] [1509523079.283569309]: [CB] 01:57:59:283 | Ardrone3CameraStateOrientation:1490 - [STATES] Enabling states/ardrone3/CameraState/Orientation
[ INFO] [1509523079.294212442]: [CB] 01:57:59:294 | Ardrone3GPSStateNumberOfSatelliteChanged:1882 - [STATES] Enabling states/ardrone3/GPSState/NumberOfSatelliteChanged
[ INFO] [1509523079.305941224]: [CB] 01:57:59:305 | CommonCommonStateBatteryStateChanged:148 - [STATES] Enabling states/common/CommonState/BatteryStateChanged
[ INFO] [1509523079.319033361]: [CB] 01:57:59:318 | CommonCommonStateWifiSignalChanged:510 - [STATES] Enabling states/common/CommonState/WifiSignalChanged
[ INFO] [1509523079.328689962]: [CB] 01:57:59:328 | CommonOverHeatStateOverHeatChanged:1023 - [STATES] Enabling states/common/OverHeatState/OverHeatChanged
[ INFO] [1509523079.332624283]: [CB] 01:57:59:332 | CommonMavlinkStateMavlinkFilePlayingStateChanged:1118 - [STATES] Enabling states/common/MavlinkState/MavlinkFilePlayingStateChanged
[ INFO] [1509523079.336220417]: [CB] 01:57:59:336 | CommonMavlinkStateMavlinkPlayErrorStateChanged:1183 - [STATES] Enabling states/common/MavlinkState/MavlinkPlayErrorStateChanged
[ INFO] [1509523079.350745116]: [CB] 01:57:59:350 | CommonFlightPlanStateAvailabilityStateChanged:1568 - [STATES] Enabling states/common/FlightPlanState/AvailabilityStateChanged
[ INFO] [1509523079.352828330]: [CB] 01:57:59:352 | CommonFlightPlanStateComponentStateListChanged:1619 - [STATES] Enabling states/common/FlightPlanState/ComponentStateListChanged
[ INFO] [1509523079.370517112]: [ARCONTROLLER_Network] 01:57:59:370 | ARCONTROLLER_Network_GetAvailableSocketPort:128 - d2c_port port: 43210
[ INFO] [1509523079.370965483]: [ARCONTROLLER_Stream2] 01:57:59:370 | ARCONTROLLER_Stream2_Open_Socket:143 - udp local port stream: 55004
[ INFO] [1509523079.371069046]: [ARCONTROLLER_Stream2] 01:57:59:371 | ARCONTROLLER_Stream2_Open_Socket:143 - udp local port control: 55005
[ INFO] [1509523079.419875638]: [BebopSDK] 01:57:59:419 | CommandReceivedCallback:113 - Command Received Callback LWP id is: 18387
[ INFO] [1509523080.059378381]: [BebopSDK] 01:58:00:059 | Connect:321 - BebopSDK inited, lwp_id: 18226
[ WARN] [1509523080.059553515]: Resetting all settings ...
[ INFO] [1509523080.059741884]: [BebopSDK] 01:58:00:059 | ResetAllSettings:417 - All settings of the drone have been reset to default values.
[bebop/bebop_driver-2] process has died [pid 18226, exit code -11, cmd /home/totonzx/bebop_ws/devel/lib/bebop_driver/bebop_driver_node __name:=bebop_driver __log:=/home/totonzx/.ros/log/5ff6f482-beda-11e7-ad5d-20689d850533/bebop-bebop_driver-2.log].
log file: /home/totonzx/.ros/log/5ff6f482-beda-11e7-ad5d-20689d850533/bebop-bebop_driver-2*.log
`

thomas-bamford, 2017-11-05:
From what I can tell from the error log, it seems that the drone was not able to connect. The latest version of bebop autonomy has not been tested on the Bebop 1, so the issue may be with the upgrade. We just got access to a Bebop 1 in our lab, so we will be able to test this soon.

TOTON95, 2017-11-07:
@thomas-bamford 
Thanks for replying.
Indeed, it couldn't connect with the Bebop, but even with the Computer connected to the Bebop AP and I made a ping to it successfully, it appeared the same code, I got a friend's computer which it had not been updated any ROS packages and  I could run it without a problem, **please tell me if you need more information or help**, I will try to join you to fix it once I can finish my scholar duties.

TOTON95, 2018-03-22:
After updating the ROS and Ubuntu dependencies, the driver could run again as expected, I could not find any specific reason. Closing issue...

closed 2018-03-22 by TOTON95

Ceepert, 2020-02-27:
> After updating the ROS and Ubuntu dependencies, the driver could run again as expected, I could not find any specific reason to malfunction. Closing issue...

what commands do I have to use to do that?