AutonomyLab/bebop_autonomy #143: In the last version of Autonomy_bebop. I tried to read the states, such as the battery changed but I get error

VINICIOSTALIN, 2017-11-07:
sorry,
In the last version of bebop autonomy I tried to read the states, such as the battery changed but I get this error

ERROR: Cannot load message class for [bebop_msgs/CommonCommonStateBatteryStateChanged]. Are your messages built?

I installed everything as says in instructions, even so I could not read the information published in any of the bebop / states /.. topics

Could you help me with this  @thomas-bamford ?

also when I type roslaunch bebop_driver bebop_node.launch
Get this error:

ERROR: cannot launch node of type [robot_state_publisher/robot_state_publisher]: robot_state_publisher
ROS path [0]=/opt/ros/indigo/share/ros
ROS path [1]=/home/tesis/bebop_ws/src/bebop_autonomy/bebop_autonomy
ROS path [2]=/home/tesis/bebop_ws/src/bebop_autonomy/bebop_description
ROS path [3]=/home/tesis/bebop_ws/src/bebop_autonomy/bebop_msgs
ROS path [4]=/home/tesis/bebop_ws/src/bebop_autonomy/bebop_driver
ROS path [5]=/home/tesis/bebop_ws/src/bebop_autonomy/bebop_tools
ROS path [6]=/home/tesis/catkin_cv/src
ROS path [7]=/opt/ros/indigo/share
ROS path [8]=/opt/ros/indigo/stacks

thomas-bamford, 2017-11-09:
Could you please provide the terminal output prior to and after the error message:
"ERROR: Cannot load message class for [bebop_msgs/CommonCommonStateBatteryStateChanged]. Are your messages built?"

thomas-bamford, 2017-11-09:
Also have you made sure to source your catkin workspace every time you open a terminal? i.e.
`source ~/bebop_ws/devel/setup.bash'

VINICIOSTALIN, 2017-11-10:
What happens is that every time I install the autonomy_bebop driver in a workspace, at the end of the last step (catkin build) it shows this:

[build] Summary: All 5 packages succeeded!
**[build] Ignored: 1 packages were skipped or are blacklisted.**
[build] Warnings: None.
[build] Abandoned: None.
[build] Failed: None.
[build] Runtime: 37.6 seconds total.
**[build] Note: Workspace packages have changed, please re-source setup files to use them**

later when I try to run the driver with this instruction in the terminal:
source bebop_ws/devel/setup.bash
roslaunch bebop_driver bebop_node.launch
Show this error:
... logging to /home/tesis/.ros/log/53d0c8c8-c62c-11e7-93d5-4c80930f288c/roslaunch-tesis-pc-6264.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://tesis-pc:41055/

SUMMARY
========

PARAMETERS
 * /bebop/bebop_driver/bebop_ip: 192.168.42.1
 * /bebop/bebop_driver/camera_info_url: package://bebop_d...
 * /bebop/bebop_driver/cmd_vel_timeout: 0.2
 * /bebop/bebop_driver/odom_frame_id: odom
 * /bebop/bebop_driver/publish_odom_tf: True
 * /bebop/bebop_driver/reset_settings: True
 * /bebop/bebop_driver/states/enable_altitudechanged: True
 * /bebop/bebop_driver/states/enable_autotakeoffmodechanged: True
 * /bebop/bebop_driver/states/enable_camerastate_orientation: True
 * /bebop/bebop_driver/states/enable_commonstate_batterystatechanged: True
 * /bebop/bebop_driver/states/enable_commonstate_wifisignalchanged: True
 * /bebop/bebop_driver/states/enable_controllerstate_ispilotingchanged: True
 * /bebop/bebop_driver/states/enable_flightplanstate_availabilitystatechanged: True
 * /bebop/bebop_driver/states/enable_flightplanstate_componentstatelistchanged: True
 * /bebop/bebop_driver/states/enable_gpsstate_numberofsatellitechanged: True
 * /bebop/bebop_driver/states/enable_mavlinkstate_mavlinkfileplayingstatechanged: True
 * /bebop/bebop_driver/states/enable_mavlinkstate_mavlinkplayerrorstatechanged: True
 * /bebop/bebop_driver/states/enable_mediastreamingstate_videoenablechanged: True
 * /bebop/bebop_driver/states/enable_numberofsatellitechanged: True
 * /bebop/bebop_driver/states/enable_overheatstate_overheatchanged: True
 * /bebop/bebop_driver/states/enable_pilotingstate_altitudechanged: True
 * /bebop/bebop_driver/states/enable_pilotingstate_attitudechanged: True
 * /bebop/bebop_driver/states/enable_pilotingstate_flattrimchanged: True
 * /bebop/bebop_driver/states/enable_pilotingstate_flyingstatechanged: True
 * /bebop/bebop_driver/states/enable_pilotingstate_navigatehomestatechanged: True
 * /bebop/bebop_driver/states/enable_pilotingstate_positionchanged: True
 * /bebop/bebop_driver/states/enable_pilotingstate_speedchanged: True
 * /bebop/robot_description: <?xml version="1....
 * /rosdistro: indigo
 * /rosversion: 1.11.21

NODES
  /bebop/
    bebop_driver (bebop_driver/bebop_driver_node)
    robot_state_publisher (robot_state_publisher/robot_state_publisher)

auto-starting new master
process[master]: started with pid [6279]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 53d0c8c8-c62c-11e7-93d5-4c80930f288c
process[rosout-1]: started with pid [6292]
started core service [/rosout]
process[bebop/bebop_driver-2]: started with pid [6295]
**ERROR: cannot launch node of type [robot_state_publisher/robot_state_publisher]: robot_state_publisher
[... 167 lines cut ...]
[ INFO] [1510327944.339326609]: [ARSTREAM2_H264Filter] 10:32:24:339 | ARSTREAM2_H264Filter_ProcessAu:577 - AU output cancelled (waitForSync)
[ INFO] [1510327944.376131222]: [ARSTREAM2_H264Filter] 10:32:24:375 | ARSTREAM2_H264Filter_ProcessAu:577 - AU output cancelled (waitForSync)
[ INFO] [1510327944.397566742]: Value for NetworkSettingsWifiSelectionType recved: 3
[ INFO] [1510327944.397758328]: [CB] 10:32:24:397 | Update:1417 - Checking if NetworkSettingsWifiSelectionType exists in params ...
[ INFO] [1510327944.400309846]: [CB] 10:32:24:400 | Update:1425 -   Yes
[ INFO] [1510327944.400421612]: Value for NetworkSettingsWifiSelectionBand recved: 0
[ INFO] [1510327944.400539052]: [CB] 10:32:24:400 | Update:1438 - Checking if NetworkSettingsWifiSelectionBand exists in params ...
[ INFO] [1510327944.402371523]: [CB] 10:32:24:402 | Update:1446 -   Yes
[ INFO] [1510327944.402481850]: Value for NetworkSettingsWifiSelectionChannel recved: 6
[ INFO] [1510327944.402577086]: [CB] 10:32:24:402 | Update:1459 - Checking if NetworkSettingsWifiSelectionChannel exists in params ...
[ INFO] [1510327944.403956202]: [CB] 10:32:24:403 | Update:1467 -   Yes
[ INFO] [1510327944.406304439]: [ARSTREAM2_H264Filter] 10:32:24:406 | ARSTREAM2_H264Filter_ProcessAu:577 - AU output cancelled (waitForSync)
[ INFO] [1510327944.438715821]: [ARSTREAM2_H264Filter] 10:32:24:438 | ARSTREAM2_H264Filter_ProcessAu:577 - AU output cancelled (waitForSync)
[ INFO] [1510327944.474297613]: [ARSTREAM2_H264Filter] 10:32:24:474 | ARSTREAM2_H264Filter_ProcessAu:577 - AU output cancelled (waitForSync)
[ INFO] [1510327944.506746550]: [ARSTREAM2_H264Filter] 10:32:24:506 | ARSTREAM2_H264Filter_ProcessAu:577 - AU output cancelled (waitForSync)
[ INFO] [1510327944.539398109]: [ARSTREAM2_H264Filter] 10:32:24:539 | ARSTREAM2_H264Filter_ProcessAu:577 - AU output cancelled (waitForSync)
[ INFO] [1510327944.580997263]: [ARSTREAM2_H264Filter] 10:32:24:580 | ARSTREAM2_H264Filter_ProcessAu:577 - AU output cancelled (waitForSync)
[... cut ...]
[ INFO] [1510327944.642943063]: [ARSTREAM2_H264FilterError] 10:32:24:642 | ARSTREAM2_H264FilterError_GenerateGrayIdrFrame:101 - Gray I slice NALU output size: 1629
[ INFO] [1510327944.643109165]: [BebopSDK] 10:32:24:643 | FrameReceivedCallback:174 - Frame Recv & Decode LWP id: 6676
[ INFO] [1510327944.692758655]: [Decoder] 10:32:24:692 | InitCodec:117 - H264 Codec is partially initialized!
[ INFO] [1510327944.692859533]: [Decoder] 10:32:24:692 | Decode:263 - Updating H264 codec parameters (Buffer Size: 36) ...
[ERROR] [1510327944.744920677]: [Decoder] 10:32:24:744 | Decode:302 - Frame size changed to 856 x 480

VINICIOSTALIN, 2017-11-16:
Excuse me @thomas-bamford could you help me with this?
I still can not solve it

danendrasingh, 2017-11-19:
@VINICIOSTALIN  I am also not able to access bebop_msgs while subscribing to the AttitudeChanged topic. Were you able to find a solution? @thomas-bamford @mani-monaj Can you guys help?

thomas-bamford, 2017-11-19:
See recent reply to #145. I was able to successfully build in an ISO of ubuntu 14.04 and get state data. In the reply I posted step-by-step process I used to successfully build. The ISO provided a pre-built version of ros indigo, so I will test on a fresh install on ubuntu 14.04 as well and post solution (sometime soon, but I am taking courses this semester and my free time is limited). Please try the steps in #145 to install and let me know if this works.

thomas-bamford, 2017-11-19:
Step by step installation instructions for 16.04 LTS included in #132. Installed ros kinetic and bebop-autonomy on a fresh image using those instructions and tested successfully on Parrot Disco. Was able to echo AttitudeChanged, PositionChanged and BatteryStateChanged successfully using those instructions. You may consider updating or reinitialise rosdep if you can't install a fresh copy.

VINICIOSTALIN, 2017-12-04:
Thank you very much @thomas-bamford #145  worked for me