AutonomyLab/bebop_autonomy #145: Bebop_Drivers failing during Catkin build

Cullen-Dubbs, 2017-11-16:
I am creating this issue due to the realization that my error is slightly different than an error i was previously commenting on : [issue 132](https://github.com/AutonomyLab/bebop_autonomy/issues/132)

I am unable to build the bebop_driver during the catkin build step of the installation process.

I follow the installation steps line for line. located [here](https://github.com/AutonomyLab/bebop_autonomy/blob/8392f055b7ef0205b357f0c2d4088389910a1151/docs/installation.rst)

I am trying this on a fresh install of a ROS iso I got from [here](http://wiki.ros.org/turtlebot/Tutorials/indigo/ISO%20Installation)

Here is the short version of the error i get

make[2]: *** [CMakeFiles/bebop_driver_nodelet.dir/src/bebop_driver_nodelet.cpp.o] Error 1
make[1]: *** [CMakeFiles/bebop_driver_nodelet.dir/all] Error 2
make: *** [all] Error 2
cd /home/parrot/bebop_ws/build/bebop_driver; catkin build --get-env bebop_driver | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
.......................................................................
Failed     << bebop_driver:make                    [ Exited with code 2 ]
Failed    <<< bebop_driver                         [ 1 minute and 16.0 seconds ]
Abandoned <<< bebop_tools                          [ Unrelated job failed ]

I am completely clueless as to where the error is... everything before this builds fine:
bebop_msgs
bebop_description
parrot_arsdk

I am on ubuntu 14.04- ros indigo

willing to try anything for this to work. 

i have tried using the catkin build with the build flag and i still get the same error if that helps at all

Please help!

thomas-bamford, 2017-11-19:
Hi @Cullen-Dubbs,

I downloaded the ISO you mentioned and installed bebop-autonomy successfully-ish (see below for command line instructions). The two main issues I found was that all packages had to be updated and you have been using old (temporary-fix) installation instructions. The updated installation instructions that work have been located [http://bebop-autonomy.readthedocs.io/en/latest/installation.html](here). I will also provide step-by-step solution for your other issue #132 this afternoon.

The '-ish' in successful is that the robot-state-publisher node launched by the bebop_driver launch files seems to crash, but after connecting to my Parrot Disco I am still able to subscribe to the state topics (see below). I am not familiar yet with the purpose of the robot-state-publisher but will look at that when I have time (taking course this semester so spare time has been limited).

# Initialize rosdep because the ISO has not done this yet
`sudo rosdep init`
`rosdep update`

# Update all packages
`sudo apt-get update`
`sudo apt-get upgrade`

# Remove out-of-date packages
`sudo apt-get autoremove`

# Follow instructions [http://bebop-autonomy.readthedocs.io/en/latest/installation.html](here)
`sudo apt-get install build-essential python-rosdep python-catkin-tools`
`mkdir -p ~/bebop_ws/src && cd ~/bebop_ws`
`catkin init`
`git clone https://github.com/AutonomyLab/bebop_autonomy.git src/bebop_autonomy`
`rosdep update`
`rosdep install --from-paths src -i`

# Make sure dependencies are updated
`sudo apt-get update`
`sudo apt-get upgrade`

# Continue with instructions
`catkin build`

# Add bebop_ws catkin workspace to bashrc so that new terminals will have access to it
`cd`
`echo "source ~/bebop_ws/devel/setup.bash" >> ~/.bashrc`

# Source in terminal that is already open
`source ~/.bashrc`

# Launch image view nodelet after drone is connected, you may find that robot-state-publisher crashes
`roslaunch bebop_tools bebop_nodelet_iv.launch`

# In separate terminal (Ctrl + Alt + T) echo topics to check if working
`rostopic echo /bebop/states/ardrone3/PilotingState/AttitudeChanged 
header: 
  seq: 222
  stamp: 
    secs: 1511114935
    nsecs: 343502303
  frame_id: base_link
roll: -2.86580452666e-05
pitch: -0.0303278341889
yaw: 0.00553899724036
---`
...

Please let me know if this works for you as this may help close some other issues as well.
Thomas

Cullen-Dubbs, 2017-11-21:
HOLY GUACAMOLE! 

This did indeed work! I got ZERO failures, errors, or warnings! 

Thank you so much. This is truly above and beyond the call of duty! 

 I never knew about upgrade our autoremove so i got to learn some more tools in the process thank you! I unfortunately wont have access to the bebop for another week but the drivers built correctly so everything should work-ish as you said. 

Thank you so much again for the help @thomas-bamford

Cullen-Dubbs, 2017-11-28:
Hey Thomas,

Just tested on the real thing and it took off and landed BEAUTIFULY.

I did try launching the image view nodelet and got a whole mess of errors. and then it crashed.

Echoing the pitch and ya and such works fine.

dalas10, 2018-01-15:
Hello,
happy new year and best of wishes,
i would like to ask,is it possible to control parrot Disco throught bebop_autonomy. i need it to do a circle motion and use the video stream, i am thinking of using it through APM, but i am more experienced with bebop_autonomy..Thank you in advance

fabrizioschiano, 2018-03-28:
@dalas10 in the documentation [here](http://bebop-autonomy.readthedocs.io/en/latest/) there is written:

Support for Parrot Disco FPV | No | Not tested (help wanted)

So I don't think it is supported yet.

huiyinlee, 2019-03-03:
Hi, I face the error during executing  catkin build as below
-------------------------------------------------------
Profile:                     default
Extending:          [cached] /opt/ros/kinetic
Workspace:                   /home/lee/bebop_ws
-------------------------------------------------------
Source Space:       [exists] /home/lee/bebop_ws/src
Log Space:          [exists] /home/lee/bebop_ws/logs
Build Space:        [exists] /home/lee/bebop_ws/build
Devel Space:        [exists] /home/lee/bebop_ws/devel
Install Space:      [unused] /home/lee/bebop_ws/install
DESTDIR:            [unused] None
-------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
-------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
-------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
-------------------------------------------------------
Workspace configuration appears valid.
-------------------------------------------------------
[build] Found '5' packages in 0.0 seconds.                                     
[build] Package table is up to date.                                           
Starting  >>> bebop_description                                                
Starting  >>> bebop_msgs                                                       
Finished  <<< bebop_description                [ 0.3 seconds ]                 
_______________________________________________________________________________
Errors     << bebop_msgs:make /home/lee/bebop_ws/logs/bebop_msgs/build.make.005.log
Traceback (most recent call last):
  File "/opt/ros/kinetic/share/gencpp/cmake/../../../lib/gencpp/gen_cpp.py", line 41, in <module>
    import genmsg.template_tools
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/genmsg/template_tools.py", line 39, in <module>
    import em
ModuleNotFoundError: No module named 'em'
make[2]: *** [/home/lee/bebop_ws/devel/.private/bebop_msgs/include/bebop_msgs/CommonChargerStateLastChargeRateChanged.h] Error 1
make[2]: *** Waiting for unfinished jobs....
Traceback (most recent call last):
  File "/opt/ros/kinetic/share/gencpp/cmake/../../../lib/gencpp/gen_cpp.py", line 41, in <module>
    import genmsg.template_tools
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/genmsg/template_tools.py", line 39, in <module>
    import em
ModuleNotFoundError: No module named 'em'
make[2]: *** [/home/lee/bebop_ws/devel/.private/bebop_msgs/include/bebop_msgs/CommonCommonStateVideoRecordingTimestamp.h] Error 1
Traceback (most recent call last):
  File "/opt/ros/kinetic/share/gencpp/cmake/../../../lib/gencpp/gen_cpp.py", line 41, in <module>
    import genmsg.template_tools
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/genmsg/template_tools.py", line 39, in <module>
    import em
ModuleNotFoundError: No module named 'em'
make[2]: *** [/home/lee/bebop_ws/devel/.private/bebop_msgs/include/bebop_msgs/Ardrone3MediaRecordStatePictureStateChanged.h] Error 1
Traceback (most recent call last):
  File "/opt/ros/kinetic/share/gencpp/cmake/../../../lib/gencpp/gen_cpp.py", line 41, in <module>
    import genmsg.template_tools
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/genmsg/template_tools.py", line 39, in <module>
    import em
ModuleNotFoundError: No module named 'em'
make[2]: *** [/home/lee/bebop_ws/devel/.private/bebop_msgs/include/bebop_msgs/CommonCommonStateMassStorageInfoRemainingListChanged.h] Error 1
make[1]: *** [CMakeFiles/bebop_msgs_generate_messages_cpp.dir/all] Error 2
make: *** [all] Error 2
cd /home/lee/bebop_ws/build/bebop_msgs; catkin build --get-env bebop_msgs | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
...............................................................................
Failed     << bebop_msgs:make                  [ Exited with code 2 ]          
Failed    <<< bebop_msgs                       [ 3.5 seconds ]                 
Abandoned <<< bebop_driver                     [ Unrelated job failed ]        
Abandoned <<< bebop_tools                      [ Unrelated job failed ]        
[build] Summary: 1 of 4 packages succeeded.                                    
[build]   Ignored:   1 packages were skipped or are blacklisted.               
[build]   Warnings:  None.                                                     
[build]   Abandoned: 2 packages were abandoned.                                
[build]   Failed:    1 packages failed.                                        
[build] Runtime: 3.5 seconds total. 

I tried with sudo pip install empy yet still cannot solve the problem. Hope can get some help here.

wajeehulhassanvii, 2019-04-05:
Hi! @Cullen-Dubbs I followed the steps you mentioned but it didn't work for me.
Here is the error log

```
wajeeh-ros@wajeeh-ros:~/bebop_ws$ catkin build
--------------------------------------------------------------------------------
Profile:                     default
Extending:          [cached] /home/wajeeh-ros/simulators/turtlebot2/devel:/home/wajeeh-ros/ardrone_simulator/devel:/home/wajeeh-ros/catkin_ws/devel:/opt/ros/kinetic
Workspace:                   /home/wajeeh-ros/bebop_ws
--------------------------------------------------------------------------------
Build Space:        [exists] /home/wajeeh-ros/bebop_ws/build
Devel Space:        [exists] /home/wajeeh-ros/bebop_ws/devel
Install Space:      [unused] /home/wajeeh-ros/bebop_ws/install
Log Space:          [exists] /home/wajeeh-ros/bebop_ws/logs
Source Space:       [exists] /home/wajeeh-ros/bebop_ws/src
DESTDIR:            [unused] None
--------------------------------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
--------------------------------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
--------------------------------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
--------------------------------------------------------------------------------
Workspace configuration appears valid.
--------------------------------------------------------------------------------
[build] Found '5' packages in 0.0 seconds.                                     
[build] Package table is up to date.                                           
Starting  >>> bebop_description                                                
Starting  >>> bebop_msgs                                                       
Finished  <<< bebop_description                [ 0.1 seconds ]                 
Finished  <<< bebop_msgs                       [ 0.8 seconds ]                 
Starting  >>> bebop_driver                                                     
_______________________________________________________________________________
Errors     << bebop_driver:make /home/wajeeh-ros/bebop_ws/logs/bebop_driver/build.make.001.log
/home/wajeeh-ros/bebop_ws/src/bebop_autonomy/bebop_driver/src/bebop_video_decoder.cpp: In member function ‘bool bebop_driver::VideoDecoder::InitCodec()’:
/home/wajeeh-ros/bebop_ws/src/bebop_autonomy/bebop_driver/src/bebop_video_decoder.cpp:74:5: warning: ‘void avcodec_register_all()’ is deprecated [-Wdeprecated-declarations]
     avcodec_register_all();
     ^
In file included from /home/wajeeh-ros/bebop_ws/src/bebop_autonomy/bebop_driver/include/bebop_driver/bebop_video_decoder.h:32:0,
                 from /home/wajeeh-ros/bebop_ws/src/bebop_autonomy/bebop_driver/src/bebop_video_decoder.cpp:25:
/usr/include/x86_64-linux-gnu/libavcodec/avcodec.h:4102:6: note: declared here
 void avcodec_register_all(void);
      ^
/home/wajeeh-ros/bebop_ws/src/bebop_autonomy/bebop_driver/src/bebop_video_decoder.cpp:74:5: warning: ‘void avcodec_register_all()’ is deprecated [-Wdeprecated-declarations]
     avcodec_register_all();
     ^
In file included from /home/wajeeh-ros/bebop_ws/src/bebop_autonomy/bebop_driver/include/bebop_driver/bebop_video_decoder.h:32:0,
                 from /home/wajeeh-ros/bebop_ws/src/bebop_autonomy/bebop_driver/src/bebop_video_decoder.cpp:25:
/usr/include/x86_64-linux-gnu/libavcodec/avcodec.h:4102:6: note: declared here
 void avcodec_register_all(void);
      ^
/home/wajeeh-ros/bebop_ws/src/bebop_autonomy/bebop_driver/src/bebop_video_decoder.cpp:74:26: warning: ‘void avcodec_register_all()’ is deprecated [-Wdeprecated-declarations]
     avcodec_register_all();
                          ^
In file included from /home/wajeeh-ros/bebop_ws/src/bebop_autonomy/bebop_driver/include/bebop_driver/bebop_video_decoder.h:32:0,
                 from /home/wajeeh-ros/bebop_ws/src/bebop_autonomy/bebop_driver/src/bebop_video_decoder.cpp:25:
/usr/include/x86_64-linux-gnu/libavcodec/avcodec.h:4102:6: note: declared here
 void avcodec_register_all(void);
      ^
/home/wajeeh-ros/bebop_ws/src/bebop_autonomy/bebop_driver/src/bebop_video_decoder.cpp:75:5: warning: ‘void av_register_all()’ is deprecated [-Wdeprecated-declarations]
     av_register_all();
     ^
In file included from /home/wajeeh-ros/bebop_ws/src/bebop_autonomy/bebop_driver/include/bebop_driver/bebop_video_decoder.h:33:0,
                 from /home/wajeeh-ros/bebop_ws/src/bebop_autonomy/bebop_driver/src/bebop_video_decoder.cpp:25:
[... 116 lines cut ...]
 int avcodec_decode_video2(AVCodecContext *avctx, AVFrame *picture,
     ^
/home/wajeeh-ros/bebop_ws/src/bebop_autonomy/bebop_driver/src/bebop_video_decoder.cpp:295:100: warning: ‘int avcodec_decode_video2(AVCodecContext*, AVFrame*, int*, const AVPacket*)’ is deprecated [-Wdeprecated-declarations]
     const int32_t len = avcodec_decode_video2(codec_ctx_ptr_, frame_ptr_, &frame_finished, &packet_);
                                                                                                    ^
In file included from /home/wajeeh-ros/bebop_ws/src/bebop_autonomy/bebop_driver/include/bebop_driver/bebop_video_decoder.h:32:0,
                 from /home/wajeeh-ros/bebop_ws/src/bebop_autonomy/bebop_driver/src/bebop_video_decoder.cpp:25:
[... cut ...]
Abandoned <<< bebop_tools                      [ Unrelated job failed ]        
[build] Summary: 2 of 4 packages succeeded.                                    
[build]   Ignored:   1 packages were skipped or are blacklisted.               
[build]   Warnings:  None.                                                     
[build]   Abandoned: 1 packages were abandoned.                                
[build]   Failed:    1 packages failed.                                        
[build] Runtime: 1.7 seconds total.
```

wajeehulhassanvii, 2019-04-08:
> Hi! @Cullen-Dubbs I followed the steps you mentioned but it didn't work for me.
> Here is the error log
>
> ```
> wajeeh-ros@wajeeh-ros:~/bebop_ws$ catkin build [...]

@thomas-bamford can you please help us in getting through this error?

mei123hao, 2020-07-17:
> @thomas-bamford can you please help us in getting through this error?

Have you solved this problem？ I tried to catkin build `bebop_autonomy`package on Ubuntu18.04 melodic ,and Now I face the same error！